esphome/issues #3777: ESP32 platformIO upgrade error

mikosoft83, 2022-11-11:
### The problem

When trying to compile an ESP32 project I get error:

Platform Manager: Installing platformio/espressif32 @ 3.5.0
Error: Please upgrade to the PlatformIO Core 6

When trying to compile an ESP8266 project it compiles fine

### Which version of ESPHome has the issue?

2022.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

### Board

NSpanel

### Component causing the issue

n/a

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
```

randybb, 2022-11-11:
2022.3.1 is a bit obsolete. Only supported scenario is to remove the deprecated ESPHome Community Addon and install the official one https://esphome.io/changelog/2022.2.0.html#home-assistant-addon-repo Sure, you have to deal with some breaking changes, but at least you will have the latest supported version.

closed 2022-11-11 by randybb

mikosoft83, 2022-11-11:
Thanks for the heads up Roman. I had my addon on autoupdate so I had literally zero idea that repo has changed.
Installing and compiling now.