esphome/issues #3778: Python SyntaxError: invalid decimal literal

nebhead, 2022-11-12:
### The problem

I'm trying to simply run an ESPHome with a YAML file and I get the following error.  This has been reproduced on two different platforms (Both running different versions of Ubuntu, but both running Python 3.10.x). 

```
ben@framework:~/syncthing/Tech-Projects/WyzeOutdoor$ esphome run wyzeoutdoor.yaml 
INFO Reading configuration wyzeoutdoor.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing wyzeoutdoor (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Library Manager: Installing Hash
INFO Installing Hash
Library Manager: Installing ESP8266WiFi
INFO Installing ESP8266WiFi
Traceback (most recent call last):
  File "/root/.platformio/packages/tool-scons/scons.py", line 89, in <module>
    import SCons.Script
  File "/root/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/__init__.py", line 72, in <module>
    import SCons.Builder
  File "/root/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Builder.py", line 106, in <module>
    import SCons.Executor
  File "/root/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Executor.py", line 563, in <module>
    class NullEnvironment(SCons.Util.Null):
  File "/root/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Executor.py", line 564, in NullEnvironment
    import SCons.CacheDir
  File "/root/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/CacheDir.py", line 32, in <module>
    import uuid
  File "/usr/local/lib/python3.10/dist-packages/uuid.py", line 138
    if not 0 <= time_low < 1<<32L:
                               ^
SyntaxError: invalid decimal literal
========================== [FAILED] Took 0.59 seconds ==========================
```

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

NA

### What platform are you using?

ESP32

### Board

Wyze Outdoor Plug

### Component causing the issue

Appears to be PlatformIO / UUID Bug

### Example YAML snippet

```yaml
substitutions:
  display_name: WyzeOutdoor
  # Higher value gives lower watt readout
  current_res: "0.001"
  # Lower value gives lower voltage readout
  voltage_div: "770"  
  
esphome:
  name: wyzeoutdoor
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "WyzePlug Fallback Hotspot"
    password: !secret wifi_pass

captive_portal:

logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

web_server:
  port: 80
  local: true

#esp32_ble_tracker:
#   scan_parameters:
#     active: false

switch:
  - platform: gpio
    name: ${display_name} Relay1
    pin:
      number: GPIO15
      inverted: false
    id: relay1
    on_turn_on:
      - light.turn_on: relay1_led
    on_turn_off:
      - light.turn_off: relay1_led
  - platform: gpio
    name: ${display_name} Relay2
    pin:
      number: GPIO32
      inverted: false
    id: relay2
    on_turn_on:
      - light.turn_on: relay2_led
    on_turn_off:
      - light.turn_off: relay2_led
  - platform: restart
    name: ${display_name} Restart

output:
  - platform: gpio
    pin: GPIO19
    inverted: True
    id: relay1_led_gpio
  - platform: gpio
    pin: GPIO16
    inverted: True
    id: relay2_led_gpio

light:
  - platform: binary
    name: "Relay1 LED"
    id: relay1_led
    internal: true    
    output: relay1_led_gpio
  - platform: binary
    name: "Relay2 LED"
    id: relay2_led
    internal: true
    output: relay2_led_gpio

sensor:
  - platform: adc
    pin: GPIO34
    name: "${display_name} LUX"
    update_interval: 10s
    attenuation: 11db
  - platform: hlw8012
    sel_pin:
      number: GPIO25
      inverted: true
    cf_pin: GPIO27
    cf1_pin: GPIO26
    current_resistor: ${current_res}
    voltage_divider: ${voltage_div}
    change_mode_every: 3
    update_interval: 3s    
    current:
      name: "${display_name} Amps"
      unit_of_measurement: A
      accuracy_decimals: 2
    voltage:
      name: "${display_name} Volts"
      unit_of_measurement: V
      accuracy_decimals: 1
    power:
      name: "${display_name} Watts"
      unit_of_measurement: W
      accuracy_decimals: 0    
      filters:
        - calibrate_linear:
            - 0.0 -> 0.0
            - 134 -> 58 

binary_sensor:
  - platform: gpio
    internal: true
    pin:
      number: GPIO18
      mode: INPUT_PULLDOWN
      inverted: True
    name: ${display_name} Button1
    on_press:
      - switch.toggle: relay1
  - platform: gpio
    internal: true
    pin:
      number: GPIO17
      mode: INPUT_PULLDOWN
      inverted: True
    name: ${display_name} Button2
    on_press:
      - switch.toggle: relay2

status_led:
  pin:
    number: GPIO5
    inverted: true
```

ssieb, 2022-11-12:
That's not esphome and it's not platformio.  It's an external library and that line isn't in the version I have.  Try cleaning your python environment and try again.

closed 2022-11-12 by ssieb

nebhead, 2022-11-12:
Thanks for taking a look.  I've tried on two different machines and I've had the same results.  How might I go about 'cleaning' my python environment?

ssieb, 2022-11-12:
Try `pip uninstall uuid` to start with.  Maybe join the esphome discord server as that's a lot more interactive for troubleshooting.

nebhead, 2022-11-12:
Got it, thank you very much, I will give it a shot.

nebhead, 2022-11-12:
For anyone that comes back here in the future with a similar issue, this is the fix that eventually worked for me is to completely purge all pip modules and start from scratch:

1. Type "pip freeze" to get list of all installed packages.
2. Copy all the names into a text file name it pkg. txt.
3. Then type the following "sudo pip uninstall -r pkg.txt"
4. You will be asked to proceed type "Y" and enter after each package

You'll then need to 'pip install --user esphome' again.  After which, it should work.  Keep in mind you'll need to go re-install any packages you might have been using previously, as they will all be gone.