esphome/issues #3779: after the update, the am 43 components stopped working

belenkiy-lab, 2022-11-12:
### The problem

After updating to ESPHome version 2022.10.2 from 2022.6.2, the am43 component stopped working.

### Which version of ESPHome has the issue?

 2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

am43

### Example YAML snippet

```yaml
esphome:
  name: curtain-bedroom
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "SSID"
  password: "PASSWORD"

   ap:
    ssid: "Curtain bedroom Fallback Hotspot"
    password: "PASWORD"

captive_portal:

logger:

api:
  password: "00000000"

ota:
  password: "00000000"

web_server:
  port: 80

esp32_ble_tracker:

ble_client:
  - mac_address: 02:27:89:39:A4:85
    id: bedroom_curtain_ble_client_left
  - mac_address: 02:6D:78:1E:CF:CC
    id: bedroom_curtain_ble_client_right

cover:
  - platform: am43
    name: "Bedroom Curtain Left Cover"
    id: bedroom_curtain_left_cover
    ble_client_id: bedroom_curtain_ble_client_left
[... 375 lines cut ...]
[11:16:40][D][esp32_ble_tracker:832]:   Name: 'Flower care'
[11:16:45][D][esp32_ble_client:036]: Found device at MAC address [02:E1:A9:7A:0B:96]
[11:16:45][I][esp32_ble_client:058]: Attempting BLE connection to 02:e1:a9:7a:0b:96
[11:16:46][D][esp32_ble_client:099]: [02:31:6c:30:2d:5a] Unexpected conn_id in CONNECT_EVT: param conn=1, open conn=0
[11:16:46][D][esp32_ble_client:099]: [02:e1:a9:7a:0b:96] Unexpected conn_id in CONNECT_EVT: param conn=1, open conn=0
[11:16:47][E][am43_cover:089]: [Kitchen Curtain Left Cover] No control service found at device, not an AM43..?
[11:16:47][E][am43:049]: [02:31:6c:30:2d:5a] No control service found at device, not an AM43..?
[11:16:47][I][esp32_ble_client:142]: Service UUID: 0x1801
[11:16:47][I][esp32_ble_client:143]:   start_handle: 0x1  end_handle: 0x4
[11:16:47][I][esp32_ble_client.service:057]:  characteristic 0x2A05, handle 0x3, properties 0x20
[11:16:47][I][esp32_ble_client:142]: Service UUID: 0x1800
[11:16:47][I][esp32_ble_client:143]:   start_handle: 0x5  end_handle: 0xb
[11:16:47][I][esp32_ble_client.service:057]:  characteristic 0x2A00, handle 0x7, properties 0x4e
[11:16:47][I][esp32_ble_client.service:057]:  characteristic 0x2A01, handle 0x9, properties 0x4e
[11:16:47][I][esp32_ble_client.service:057]:  characteristic 0x2A04, handle 0xb, properties 0x2
[11:16:47][I][esp32_ble_client:142]: Service UUID: 0xFE50
INFO curtain-kitchen.local: Ping timed out!
INFO Disconnected from ESPHome API for curtain-kitchen.local
WARNING Disconnected from API
WARNING Can't connect to ESPHome API for curtain-kitchen.local: Error resolving IP address: [Errno -5] No address associated with hostname
INFO Trying to reconnect to curtain-kitchen.local in the background
INFO Successfully connected to curtain-kitchen.local
[11:17:10][W][api.connection:081]: Home Assistant 2022.10.5 (::FFFF:C0A8:50AF): Connection reset
[11:17:16][D][sensor:127]: 'Kitchen Curtain Controler Uptime Sensor': Sending state 44.75800 s with 0 decimals of accuracy
```

akrigator, 2022-11-14:
Looks as a duplicate https://github.com/esphome/issues/issues/3776

akrigator, 2022-11-14:
[alternative workaround] (https://github.com/alutov/ESP32-R4sGate-for-Redmond)