esphome/issues #3781: Config files locked in docker installation, cannot back up

jpmiller25, 2022-11-12:
### The problem

TLDR; the esphome docker installation seems to be locking the config files more than necessary, and leaving them locked when the container stops. This prevents me from doing a proper backup. 

I have esphome installed in docker alongside home assistant. I use duplicati to back up my home assistant config directory and would like to back up my esphome as well. However when I run the backup I get over 137 warnings of files that could not be accessed or backed up. They are all within my esphome-config folder that’s bind mounted to the container. I have tried stopping the container and running the backup, but it doesn’t help. I notice that all the files created in my esphome-config are owned by root, perhaps it would help if the container was not running as root?
HA 2022.11.2
Esphome 2022.10

### Which version of ESPHome has the issue?

2022.10

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

N/A

### Component causing the issue

Esphome docker container

### Anything in the logs that might be useful for us?

```txt
Sample warning from duplicati:
2022-11-12 15:19:06 -06 - [Warning-Duplicati.Library.Main.Operation.Backup.FileEnumerationProcess-FileAccessError]: Error reported while accessing file: /source/home/josh/homeassistant/esphome-config/.esphome/platformio/platforms/espressif32/
```

### Additional information

I don’t get this error on config files for several different docker containers I use even when they are running.

randybb, 2022-11-13:
My config files are not owned by root, I can do whatever do with them anytime. And if you stop esphome container and it doesn't help then there must be other problem not related to esphome.

jpmiller25, 2022-11-14:
> My config files are not owned by root, I can do whatever do with them anytime. And if you stop esphome container and it doesn't help then there must be other problem not related to esphome.

Thanks, I'll try to start over with a fresh docker compose and container. Do you set any environment variables for puid/pgid in your container definition?
Or, could it make a difference that my host user is 1000:1003, instead of the typical 1000:1000?

closed 2024-01-21 by jpmiller25