esphome/issues #3782: Display ssd1306_i2c causes wifi connection failure, reason='Auth Expired'

bdurrett, 2022-11-13:
### The problem

I am trying to add a dht11 sensor and a ssd1306 display to a nodeMCU esp8266. The sensor alone works fine but when adding the display, wifi fails, but only when actually connected (code works fine if display physically disconnected). The sensor reads properly, and the display renders the "Goodbye World!" properly, the only failure is wifi can not connect if the display is physically connected. 

The yaml I am using is below:

```
esphome:
  name: temp-humid

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "[REDACTED]"

ota:
  password: "[REDACTED]"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

sensor:
  - platform: dht
    pin: D5    # GPI14, D5
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s
    model: DHT11
    
i2c:
  sda: D1     # GPIO5, D1
  scl: D2     # GPIO4, D2

font:
  # gfonts://family[@weight]
  - file: "gfonts://Roboto"
    id: roboto
    size: 16

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(roboto), "Goodbye World!");
```

The logs are as follows:

```
[20:48:54][I][wifi:255]: WiFi Connecting to '[REDACTED]'...
[20:48:56][W][wifi_esp8266:482]: Event: Disconnected ssid='[REDACTED]' bssid=[redacted] reason='Auth Expired'
[20:48:56][W][wifi:547]: Error while connecting to network.
[20:48:56][W][wifi:583]: Restarting WiFi adapter...
```

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

ssd1306_i2c

### Example YAML snippet

```yaml
i2c:
  sda: D1     # GPIO5, D1
  scl: D2     # GPIO4, D2

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(roboto), "Goodbye World!");
```

### Anything in the logs that might be useful for us?

```txt
[20:48:54][I][wifi:255]: WiFi Connecting to '[REDACTED]'...
[20:48:56][W][wifi_esp8266:482]: Event: Disconnected ssid='[REDACTED]' bssid=[redacted] reason='Auth Expired'
[20:48:56][W][wifi:547]: Error while connecting to network.
[20:48:56][W][wifi:583]: Restarting WiFi adapter...
```

ssieb, 2022-11-13:
Try increasing the i2c frequency.

bdurrett, 2022-11-13:
I discovered increasing 12c frequency as a fix (well, workaround?), also increasing the value of the display update_interval to 2s seems to work:
```
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    update_interval: 2s   # fixes the wifi connection problem 
```
I realize this will likely result in the issue not being addressed, but I suggest that since this seems to be a common issue that blocks users / requires debugging, a log message would be appropriate.

jurgenstef, 2022-11-13:
i had the exact same problem. adding setup_priority solved my problem.
`i2c:
  sda: D1
  scl: D2
  scan: false
  setup_priority: -100

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    setup_priority: -100
    reset_pin: D0
    address: 0x3D`

weboide, 2022-11-23:
Having the same issue with the same combination of esp8266 and ssd1306_i2c. The workaround suggested by @jurgenstef fixes the problem! Wifi now connects right away.
You saved me from quite a bit of frustration!