esphome/issues #3783: Wemos S2 Mini - I2S Iissue

alfwro13, 2022-11-13:
### The problem

I am trying to use max98357a I2S Amplifier with Wemos S2 mini. According to the [Wemos S2 mini documentation](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf) I should be able to use any GPIO pins with I2S.
My ESPHome config:
```
media_player:
  - platform: i2s_audio
    name: ${friendly_name} Speaker
    dac_type: external
    i2s_lrclk_pin: 12
    i2s_dout_pin: 33
    i2s_bclk_pin: 11
    mode: mono
    id: media_out
```
but when I try to install it I get this:
```
INFO Reading configuration /config/esphome/szopa-multisensor.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing szopa-multisensor (board: lolin_s2_mini; framework: arduino; platform: platformio/espressif32 @ 4.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- ESP32-audioI2S @ 2.1.0
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/i2s_audio/i2s_audio_media_player.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/core/component.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/core/component_iterator.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/core/controller.cpp.o
In file included from src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:1:
src/esphome/components/i2s_audio/i2s_audio_media_player.h:28:30: error: 'i2s_dac_mode_t' has not been declared
   void set_internal_dac_mode(i2s_dac_mode_t mode) { this->internal_dac_mode_ = mode; }
                              ^~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.h:54:3: error: 'i2s_dac_mode_t' does not name a type; did you mean 'i2s_mode_t'?
   i2s_dac_mode_t internal_dac_mode_{I2S_DAC_CHANNEL_DISABLE};
   ^~~~~~~~~~~~~~
   i2s_mode_t
src/esphome/components/i2s_audio/i2s_audio_media_player.h: In member function 'void esphome::i2s_audio::I2SAudioMediaPlayer::set_internal_dac_mode(int)':
src/esphome/components/i2s_audio/i2s_audio_media_player.h:28:59: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
   void set_internal_dac_mode(i2s_dac_mode_t mode) { this->internal_dac_mode_ = mode; }
                                                           ^~~~~~~~~~~~~~~~~~
                                                           set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp: In member function 'virtual void esphome::i2s_audio::I2SAudioMediaPlayer::setup()':
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:106:13: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
             ^~~~~~~~~~~~~~~~~~
             set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:106:35: error: 'I2S_DAC_CHANNEL_DISABLE' was not declared in this scope
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:106:35: note: suggested alternative: 'PTHREAD_CANCEL_DISABLE'
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
                                   PTHREAD_CANCEL_DISABLE
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:107:51: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
     this->audio_ = make_unique<Audio>(true, this->internal_dac_mode_);
                                                   ^~~~~~~~~~~~~~~~~~
                                                   set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp: In member function 'virtual void esphome::i2s_audio::I2SAudioMediaPlayer::dump_config()':
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:140:13: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
             ^~~~~~~~~~~~~~~~~~
             set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:140:35: error: 'I2S_DAC_CHANNEL_DISABLE' was not declared in this scope
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:140:35: note: suggested alternative: 'PTHREAD_CANCEL_DISABLE'
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
                                   PTHREAD_CANCEL_DISABLE
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:141:19: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
     switch (this->internal_dac_mode_) {
                   ^~~~~~~~~~~~~~~~~~
                   set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:142:12: error: 'I2S_DAC_CHANNEL_LEFT_EN' was not declared in this scope
       case I2S_DAC_CHANNEL_LEFT_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:142:12: note: suggested alternative: 'I2S_DATA_ENABLE_TEST_EN'
       case I2S_DAC_CHANNEL_LEFT_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~
            I2S_DATA_ENABLE_TEST_EN
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:145:12: error: 'I2S_DAC_CHANNEL_RIGHT_EN' was not declared in this scope
       case I2S_DAC_CHANNEL_RIGHT_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:145:12: note: suggested alternative: 'I2S_CHANNEL_FMT_RIGHT_LEFT'
       case I2S_DAC_CHANNEL_RIGHT_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~~
            I2S_CHANNEL_FMT_RIGHT_LEFT
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:148:12: error: 'I2S_DAC_CHANNEL_BOTH_EN' was not declared in this scope
       case I2S_DAC_CHANNEL_BOTH_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:148:12: note: suggested alternative: 'I2S_CHANNEL_STEREO'
       case I2S_DAC_CHANNEL_BOTH_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~
            I2S_CHANNEL_STEREO
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/core/entity_base.cpp.o
*** [/data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/i2s_audio/i2s_audio_media_player.cpp.o] Error 1
========================== [FAILED] Took 2.51 seconds ==========================
```

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.2

### What platform are you using?

ESP32

### Board

Wemos S2 Mini

### Component causing the issue

I2S

### Example YAML snippet

```yaml
esphome:
  name: szopa-multisensor

esp32:
  board: lolin_s2_mini
  variant: esp32s2
  framework:
    type: arduino
    version: 2.0.3
    platform_version: 4.4.0

wifi:
<section removed for brevity>
logger:
api:
ota:
  password: <removed>

 
media_player:
  - platform: i2s_audio
    name: Speaker
    dac_type: external
    i2s_lrclk_pin: 12
    i2s_dout_pin: 33
    i2s_bclk_pin: 11
    mode: mono
    id: media_out
```

alfwro13, 2022-12-14:
Now that ESPHome supports Wemos S2 mini I updated the config with:

````
esp32:
  board: lolin_s2_mini
````
and I have re-tested it in esphome v2022.12.0 and HA v2022.12.5
No change still  failing to compile:
````
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- ESP32-audioI2S @ 2.0.6
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/i2s_audio/i2s_audio_media_player.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/media_player/media_player.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/network/util.cpp.o
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
In file included from src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:1:
src/esphome/components/i2s_audio/i2s_audio_media_player.h:28:30: error: 'i2s_dac_mode_t' has not been declared
   void set_internal_dac_mode(i2s_dac_mode_t mode) { this->internal_dac_mode_ = mode; }
                              ^~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.h:54:3: error: 'i2s_dac_mode_t' does not name a type; did you mean 'i2s_mode_t'?
   i2s_dac_mode_t internal_dac_mode_{I2S_DAC_CHANNEL_DISABLE};
   ^~~~~~~~~~~~~~
   i2s_mode_t
src/esphome/components/i2s_audio/i2s_audio_media_player.h: In member function 'void esphome::i2s_audio::I2SAudioMediaPlayer::set_internal_dac_mode(int)':
src/esphome/components/i2s_audio/i2s_audio_media_player.h:28:59: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
   void set_internal_dac_mode(i2s_dac_mode_t mode) { this->internal_dac_mode_ = mode; }
                                                           ^~~~~~~~~~~~~~~~~~
                                                           set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp: In member function 'virtual void esphome::i2s_audio::I2SAudioMediaPlayer::setup()':
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:106:13: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
             ^~~~~~~~~~~~~~~~~~
             set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:106:35: error: 'I2S_DAC_CHANNEL_DISABLE' was not declared in this scope
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:106:35: note: suggested alternative: 'PTHREAD_CANCEL_DISABLE'
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
                                   PTHREAD_CANCEL_DISABLE
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:107:51: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
     this->audio_ = make_unique<Audio>(true, this->internal_dac_mode_);
                                                   ^~~~~~~~~~~~~~~~~~
                                                   set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp: In member function 'virtual void esphome::i2s_audio::I2SAudioMediaPlayer::dump_config()':
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:140:13: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
             ^~~~~~~~~~~~~~~~~~
             set_internal_dac_mode
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:140:35: error: 'I2S_DAC_CHANNEL_DISABLE' was not declared in this scope
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:140:35: note: suggested alternative: 'PTHREAD_CANCEL_DISABLE'
   if (this->internal_dac_mode_ != I2S_DAC_CHANNEL_DISABLE) {
                                   ^~~~~~~~~~~~~~~~~~~~~~~
                                   PTHREAD_CANCEL_DISABLE
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:141:19: error: 'class esphome::i2s_audio::I2SAudioMediaPlayer' has no member named 'internal_dac_mode_'; did you mean 'set_internal_dac_mode'?
[... 7 lines cut ...]
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:142:12: note: suggested alternative: 'I2S_DATA_ENABLE_TEST_EN'
       case I2S_DAC_CHANNEL_LEFT_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~
            I2S_DATA_ENABLE_TEST_EN
[... cut ...]
src/esphome/components/i2s_audio/i2s_audio_media_player.cpp:148:12: note: suggested alternative: 'I2S_CHANNEL_STEREO'
       case I2S_DAC_CHANNEL_BOTH_EN:
            ^~~~~~~~~~~~~~~~~~~~~~~
            I2S_CHANNEL_STEREO
Compiling /data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/sensor/filter.cpp.o
*** [/data/szopa-multisensor/.pioenvs/szopa-multisensor/src/esphome/components/i2s_audio/i2s_audio_media_player.cpp.o] Error 1
========================== [FAILED] Took 2.70 seconds ==========================
````

jesserockz, 2022-12-15:
Copied from #3855 @LucaDev

See: https://github.com/espressif/esp-idf/blob/b04e33256df3a6149e75a4704393322b9149df6c/components/driver/deprecated/driver/i2s_types_legacy.h#L130
SOC_I2S_SUPPORTS_DAC = false for the ESP32 S2

Example of it being used regardless of the target IC:
https://github.com/esphome/esphome/blob/72b294333287658ecf8577de5cafd05a901d7921/esphome/components/i2s_audio/i2s_audio_media_player.h#L28

closed 2022-12-15 by jesserockz

Electronlibre2012, 2023-02-19:
Hello,

seems this fix has affected the regular ESP32 because the audio is now lowest level and poor quality.

elik745i, 2023-05-12:
Hello there, I have tried several boards of mini and I2S sound is intermittent, does not work well.

my config is following:
`esphome:
  name: mediaplayer

esp32:
  board: lolin_s2_mini
  framework:
    type: arduino

wifi:
<section removed for brevity>
logger:
api:
ota:
  password: <removed>

i2s_audio:
    i2s_lrclk_pin: 16
    i2s_bclk_pin: 17
 
media_player:
  - platform: i2s_audio
    dac_type: external
    i2s_dout_pin: 18
    mode: mono

Lateralleap, 2023-06-11:
Hi

Just wondering if you managed to get max98357a I2S Amplifier working with Wemos S2 mini, and if so, what pins you used?

Thanks