esphome/issues #3785: Unexpected Maker TinyS3 not a known board

BeardyMike, 2022-11-14:
### The problem

The [Unexpected Maker TinyS3](https://esp32s3.com/tinys3.html) fails to compile for me.

Despite appearing in the PlatformIO devices list, [um_tinys3](https://docs.platformio.org/en/latest/boards/espressif32/um_tinys3.html#unexpected-maker-tinys3), I cannot get my YAML to compile. The console tells me that the board is unknown

Thanks in advance.

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

um_tinys3

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: um_tinys3
  framework:
    type: arduino
  variant: esp32s3

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REMOVED"

ota:
  password: "REMOVED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "REMOVED"
    password: "REMOVED"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/door.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing door (board: um_tinys3; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Error: Unknown board ID 'um_tinys3'
```

ilium007, 2022-11-14:
Came here with the same issue

reubenbijl, 2022-11-27:
Looks like this will be coming soon. In the current release version of esphome 2022.11.3, it's using version 3.5.0 of the espressif toolchain on platformio.

Looks like this has been updated here.

https://github.com/esphome/esphome/commit/c2e198311ca73112fa7c9c213101277200085a6f

reubenbijl, 2022-11-27:
Installed the latest version of esphome and succesfully ran a project on my tinys3 :)

```
python3 -m pip install 'esphome @ git+https://github.com/esphome/esphome@c5f59fad626534c55d58ad6927ab994cafc6a7b3'
```

reubenbijl, 2022-11-27:
Here's a really simple file to run the built in RGB Led.

```
esphome:
  name: rgb-light-demo

esp32:
  board: um_tinys3
  framework:
    type: arduino
    
light:
  - platform: fastled_clockless
    chipset: WS2812
    rgb_order: GRB
    pin: GPIO18
    num_leds: 1
    name: "RGB Light"
    on_turn_on:
    - switch.turn_on: rgb_power
    on_turn_off:
    - switch.turn_off: rgb_power

switch:
    - platform: gpio
      pin: GPIO17
      restore_mode: ALWAYS_ON
      id: rgb_power

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: ###
  password: ###
```

BeardyMike, 2023-03-28:
Resolved.

[reubenbijl](https://github.com/reubenbijl)'s answer worked for me.

closed 2023-03-28 by BeardyMike