esphome/issues #3787: ESP32S2 - DAC problem

SmartieBjelovar, 2022-11-14:
### The problem

DAC function is not available for ESP32S2 modules.  Internal DAC pins for ESP32S2 are GPIO17 and GPIO18 but esphome allows pins GPIO25 and GPIO26. 

Am I doing something wrong, is there any other way to use an internal DAC on ESP32S2?

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32-IDF

### Board

esp32-s2-saola-1

### Component causing the issue

esp32_dac

### Example YAML snippet

```yaml
esp32:
  board: esp32-s2-saola-1
  framework:
    type: esp-idf
    version: recommended
  variant: esp32s2

output:
  - platform: esp32_dac
    pin: GPIO17
    id: fan_speed
```

### Anything in the logs that might be useful for us?

```txt
output.esp32_dac: [source /config/esphome/fan-controller.yaml:12]
  platform: esp32_dac
  
  Unknown value '17', valid options are '25', '26'.
  pin: GPIO17
  id: fan_speed
```

Bart-1992, 2023-02-14:
Did you managed to solve this? I currently have the same problem.... I need a DAC output on my ESP32-S2, but unfortunately not supported (yet?)

SmartieBjelovar, 2023-02-17:
We gave up and made a new PCB with ESP32 Wroom, works without a problem now
:)

hppi-johnsson, 2023-02-21:
I got it working by modifying the references to GPIO25 & 26 in these two files:

...\Lib\site-packages\esphome\components\esp32_dac\output.py
```
def valid_dac_pin(value):
    num = value[CONF_NUMBER]
    cv.one_of(17, 18)(num)
    return value
```

...\Lib\site-packages\esphome\components\esp32_dac\esp32_dac.cpp
`auto channel = pin_->get_pin() == 17 ? DAC_CHANNEL_1 : DAC_CHANNEL_2;`

RealDoublejan, 2023-07-31:
@jesserockz can the above named fix be added to the Output component?

RaceNJason, 2024-03-03:
Your standard HAOS install on a Pi doesn't allow access to the esp32_dac.cpp files. Would be nice if the ESPHome developers just put this in their code and updated the integration.

Anyway, shouldn't
`auto channel = pin_->get_pin() == 17 ? DAC_CHANNEL_1 : DAC_CHANNEL_2;`

be instead (to make sure people who still use the GPIO25 and GPIO26 pins to not break)?
`auto channel = (pin_->get_pin() == 17 || pin_->get_pin() == 25 ? DAC_CHANNEL_1 : DAC_CHANNEL_2;`

ufacek, 2025-02-02:
Hello, is this already fixed or can I set the DAC output to work somewhere else?

RealDoublejan, 2025-02-02:
@ufacek I installed an gp8403 to work around it. Easily to get as little breakout

ufacek, 2025-02-07:
> [@ufacek](https://github.com/ufacek) Nainstaloval jsem GP8403, abych to obešel. Snadno získat tak malý útěk

why add another DAC module when this board can do it, it's just not well supported, can I somehow fix it myself?

closed 2025-02-14 by jesserockz