esphome/issues #3788: cwww Light reboots when set to midrange, due to watchdog timeout

glyndon, 2022-11-15:
### The problem

Full discussion thread at:
[https://community.home-assistant.io/t/cwww-light-reboots-when-set-to-midrange-due-to-watchdog-timeout/488784](url)

cwww lights running ESPHome which reboot themselves approx. every 5-6 minutes when set to midrange color-temperature and brightness values (e.g. 25% bright, 4000K).

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Board

Lolin d1_mini

### Component causing the issue

cwww Light

### Example YAML snippet

```yaml
light:
  - platform: cwww
    name: ${node_name}
    id: the_light
    cold_white: output_cool_leds
    warm_white: output_warm_leds
    cold_white_color_temperature: 6024 K
    warm_white_color_temperature: 2700 K
    default_transition_length: 1s
    constant_brightness: true

output:
  - platform: esp8266_pwm
    id: output_warm_leds
    pin: GPIO5
    inverted: false
  - platform: esp8266_pwm
    id: output_cool_leds
    pin: GPIO4
    inverted: false
```

### Anything in the logs that might be useful for us?

```txt
rst cause:4, boot mode:(3,6)
wdt reset
```

### Additional information

The devices are stable at either end of the color/brightness ranges - they only self-reset when the colors were about equal.
So, I looked around and found that the call shown below may ‘feed’ the watchdog timer sufficiently to prevent the random rebooting.
`
esphome:
  on_loop:
    then:
      - lambda: |-
          ESP.wdtFeed(); return;
`
After adding the call to WDT, the devices have remained reboot-free for 18 hours. Previously they could not last beyond a sunset/sunrise, when the circadian code in HA sets them to the troublesome midranges of color & brightness.

When I later added “instrumentation” such as sntp_time, loop time, heap fragmentation, etc, to the yaml config and removed the wdtfeed call, the problem remained absent.
That smells like a pseudo-race happening at a much lower level than the cwww module.
I looked into the cwww and Light code, and the math in the ESPHome code seems sane and unlikely to cause loop starvation. My guess is this issue is an emergent property of the underlying Arduino lib that handles software PWM (or, in this case, 2 of them simultaneously).
I'm posting this as an issue on GitHub, if only so the maintainer on the ESPHome side can be aware of it.
But I suspect the prob does not reside in the cwww code itself, but in the libs it relies upon.

glyndon, 2023-04-05:
No one else is experiencing this issue?

cc-026, 2024-12-08:
My esphome version is [2024.4.0]

I'm also troubled by this problem，but the timing when my 8266 triggers this problem is not as regular as you described. It looks more like a 'random' trigger.
[image]
[image]
[image]
[image]
[image]

When the loop time is about 300ms, the watchdog will be triggered, but when I added ESP.wdtFeed(); it is indeed prevented the watchdog from triggering once at 300ms loop time
[image]