esphome/issues #3789: ledc doc improvement

wire67, 2022-11-15:
### The problem

There's a comment in the camera component that should be added to the ledc component too:
https://esphome.io/components/output/ledc.html 

``` 
Note
Camera uses PWM timer #1. You need to manually specify a channel (with the `channel: 2` parameter)
```

### Which version of ESPHome has the issue?

website

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

wire67, 2023-03-16:
Up

nielsnl68, 2023-03-16:
Did you know, you can do this your self by creating a Pull Request on https://github.com/esphome-docs.
Read all about it here: https://esphome.io/guides/contributing.html