esphome/issues #3791: Wifi signal reporting on Raspberry Pi Pico W causes invalid data and board reboots

mattmole, 2022-11-15:
### The problem

When using wifi details sensors on the RPi Pico 2040W board, the information reported is invalid and causes board reboots.

1) When using the wifi signal module, the following is reported:

sensor:
  # WiFi Signal sensor
  - platform: wifi_signal
    name: wifi_signal
    update_interval: 10s

The following is reported and doesn't change:

[21:05:00][C][wifi_signal.sensor:009]: WiFi Signal 'wifi_signal'
[21:05:00][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:05:00][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:05:00][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:05:00][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:05:09][D][sensor:126]: 'wifi_signal': Sending state 0.00000 dBm with 0 decimals of accuracy
[21:05:19][D][sensor:126]: 'wifi_signal': Sending state 0.00000 dBm with 0 decimals of accuracy
[21:05:29][D][sensor:126]: 'wifi_signal': Sending state 0.00000 dBm with 0 decimals of accuracy

Also, every couple of minutes the board reboots

### Which version of ESPHome has the issue?

v2022.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

wifi_signal

### Example YAML snippet

```yaml
esphome:
  name: rpi-pico

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Rpi-Pico Fallback Hotspot"
    password: "REDACTED"

# Text sensors with general information
text_sensor:
  - platform: version
    name: version

sensor:
  # WiFi Signal sensor
  - platform: wifi_signal
    name: wifi_signal
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
The following is reported and doesn't change:

[21:05:00][C][wifi_signal.sensor:009]: WiFi Signal 'wifi_signal'
[21:05:00][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:05:00][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:05:00][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:05:00][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:05:09][D][sensor:126]: 'wifi_signal': Sending state 0.00000 dBm with 0 decimals of accuracy
[21:05:19][D][sensor:126]: 'wifi_signal': Sending state 0.00000 dBm with 0 decimals of accuracy
[21:05:29][D][sensor:126]: 'wifi_signal': Sending state 0.00000 dBm with 0 decimals of accuracy
```

si458, 2022-12-17:
This seems to of be fixed in the latest 2022.12.1 release of esphome 

```
INFO Successfully connected to pico-w-home.local
[07:13:18][I][app:102]: ESPHome version 2022.12.1 compiled on Dec 17 2022, 06:44:59
[07:13:18][C][wifi:504]: WiFi:
[07:13:18][C][wifi:362]:   Local MAC: 28:CD:C1:00:49:D8
[07:13:18][C][wifi:363]:   SSID: [redacted]
[07:13:18][C][wifi:364]:   IP Address: 192.168.23.107
[07:13:18][C][wifi:365]:   BSSID: [redacted]
[07:13:18][C][wifi:367]:   Hostname: 'pico-w-home'
[07:13:18][C][wifi:369]:   Signal strength: -45 dB ▂▄▆█
[07:13:18][C][wifi:373]:   Channel: 9
[07:13:18][C][wifi:374]:   Subnet: 255.255.255.0
[07:13:18][C][wifi:375]:   Gateway: 192.168.23.1
[07:13:18][C][wifi:376]:   DNS1: 192.168.23.1
[07:13:18][C][wifi:377]:   DNS2: 0.0.0.0
[07:13:18][C][logger:293]: Logger:
[07:13:18][C][logger:294]:   Level: DEBUG
[07:13:18][C][logger:295]:   Log Baud Rate: 115200
[07:13:18][C][logger:296]:   Hardware UART: USB_CDC
[07:13:18][C][uptime.sensor:031]: Uptime Sensor 'Uptime Sensor'
[07:13:18][C][uptime.sensor:031]:   Device Class: 'duration'
[07:13:18][C][uptime.sensor:031]:   State Class: 'total_increasing'
[07:13:18][C][uptime.sensor:031]:   Unit of Measurement: 's'
[07:13:18][C][uptime.sensor:031]:   Accuracy Decimals: 0
[07:13:18][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[07:13:18][C][gpio.output:010]: GPIO Binary Output:
[07:13:18][C][gpio.output:011]:   Pin: GPIO32
[07:13:18][C][restart:076]: Restart Switch 'Restart Pico W'
```

emefff, 2022-12-18:
I just wanted to open a new issue and found this. I am running esphome 2022.12.1 - December 16. My Pico W is about 3-4m from the next WiFi Mesh Repeater away, but is losing connection frequently. Reporting of WiFi signal strength is all over the place from approx. -70 to 0dBm:
[image]

When 0dBm is recorded, it is because of a loss of connection before:
[image]

I do not know, if the antenna of the Pico W is particularily bad, but an ESP32 in the same room with greater distance to the repeater does not lose its signal.

So I do not think this is resolved,

Mario.

si458, 2022-12-18:
@emefff i think your issue is maybe the signal strength of the pico w itself,
where as this issue is about the fact is was showing 0db and not it isnt in the latest update,
i would suggest you could open another issue but i dont think the is much the developers can maybe do?

emefff, 2022-12-18:
It is also rebooting! I do not know if the rebooting is the cause or the consequence. In comparison ESP32 has no trouble down to -80dBm.

si458, 2022-12-18:
> It is also rebooting! I do not know if the rebooting is the cause or the consequence. In comparison ESP32 has no trouble down to -80dBm.

my pico w isnt rebooting at all? whats your config like?
it could be an issue with the wifi bouncing between access points?

emefff, 2022-12-18:
I have a mesh repeater and one base. Bouncing should not matter, it happens a lot in another ESP32, which does not lose connection for hundreds of hours.

I did not set reboot_timeout, so it should be 15min. Clearly, the intervals in above curve are much smaller. That's strange.

.yaml:

```

esphome:
  name: esphome--14
  #platformio_options:
  #  board_build.f_cpu: 100000000L #stromverbrauch kein Thema, aber T-Einfluss? immer wieder LOC daher aus
    #https://github.com/esphome/issues/issues/3909

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:
  #level: 
    #VERBOSE

  
# Enable Home Assistant API
api:
  encryption:
    key: REDACTED

ota:
  password: REDACTED

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  output_power: 20dB #8.5-20db
  manual_ip:
    static_ip: REDACTED
    gateway: REDACTED
    subnet: 255.255.255.0
  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Esphome--14 Fallback Hotspot"
    password: REDACTED
    

dallas:
  - pin: GPIO5

i2c: 
  sda: GPIO20 
  scl: GPIO21 
  scan: true
  id: bus_a    

sensor:
  - platform: dallas
    address: 0x230922503ac7d228
    name: "Schlafzimmer 2 Pico Onboard Dallas Temperature"
    accuracy_decimals: 1  
    filters:
      - filter_out: nan
      - filter_out: 85.0   
  - platform: wifi_signal
    name: "Schlafzimmer 2 Pico WiFi Signal"
    update_interval: 300s # defaults to 60s
    accuracy_decimals: 1
  - platform: uptime
    name: "Schlafzimmer 2 Pico Uptime"
    update_interval: 300s # defaults to 60s
    filters:
      - lambda: return x / 3600;
    unit_of_measurement: "h"
    accuracy_decimals: 1
  - platform: bmp280
    temperature:
      name: "Schlafzimmer 2 Pico BMP280 Temperature"
      oversampling: 16x
    pressure:
      name: "Schlafzimmer 2 Pico BMP280 Pressure"
    address: 0x76
    update_interval: 60s
  - platform: bh1750
    name: "Schlafzimmer 2 Pico BH Illuminance"
    address: 0x23
    update_interval: 60s

button:
  - platform: restart
    name: "Schlafzimmer 2 Restart"
    id: schlafzimmer2_restart_button
    disabled_by_default: true
    on_press:
      then:
        - logger.log: Button Pressed
```

bdraco, 2026-03-12:
Closing — this was fixed back in 2022.12.1 as confirmed by @si458. If anyone is still seeing wifi signal issues on RP2040 with a current version, please open a new issue.