esphome/issues #3792: Wifi_info sensor causes erroneous information and board reboots on the Raspberry Pi Pico W

mattmole, 2022-11-15:
### The problem

When using the wifi_info sensor, the information reported is erroneous and causes regular board reboots

### Which version of ESPHome has the issue?

v2022.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

wifi_info

### Example YAML snippet

```yaml
esphome:
  name: rpi-pico

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:
  password: "REDACTED"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Rpi-Pico Fallback Hotspot"
    password: "REDACTED"

# Text sensors with general information
text_sensor:
  - platform: wifi_info
    ip_address:
      name: ip
    ssid:
      name: ssid
    bssid:
      name: bssid
```

### Anything in the logs that might be useful for us?

```txt
[20:48:36][W][api.connection:083]: Home Assistant 2022.11.2 (192.168.1.11): Connection closed
[20:48:36][D][api:102]: Accepted 192.168.1.11
[20:48:36][D][api.connection:917]: Home Assistant 2022.11.2 (192.168.1.11): Connected successfully
[20:48:37][D][text_sensor:067]: 'ssid': Sending state ''
[20:48:38][D][text_sensor:067]: 'ssid': Sending state ''
[20:48:40][D][text_sensor:067]: 'ssid': Sending state 't'
[20:48:41][D][text_sensor:067]: 'ssid': Sending state ''
[20:48:42][D][text_sensor:067]: 'ssid': Sending state ''
[20:48:44][D][text_sensor:067]: 'ssid': Sending state ''
[20:48:45][D][text_sensor:067]: 'ssid': Sending state '\x900'
```

### Additional information

[image]

si458, 2022-12-09:
i have noticed the same thing even on the latest 2022.11.5 version
the SSID is completely wrong and the is no signal or channel information?
```
[20:10:52][I][app:102]: ESPHome version 2022.11.5 compiled on Dec  6 2022, 09:45:14
[20:10:52][C][wifi:504]: WiFi:
[20:10:52][C][wifi:362]:   Local MAC: 28:CD:C1:00:49:D8
[20:10:52][C][wifi:363]:   SSID: '9   '
[20:10:52][C][wifi:364]:   IP Address: 192.168.23.107
[20:10:52][C][wifi:365]:   BSSID: EE:EE:EE:EE:EE:EE
[20:10:52][C][wifi:367]:   Hostname: 'pico-w-home'
[20:10:52][C][wifi:369]:   Signal strength: 0 dB ▂▄▆█
[20:10:52][C][wifi:373]:   Channel: 0
[20:10:52][C][wifi:374]:   Subnet: 255.255.255.0
[20:10:52][C][wifi:375]:   Gateway: 192.168.23.1
[20:10:52][C][wifi:376]:   DNS1: 192.168.23.1
[20:10:52][C][wifi:377]:   DNS2: 0.0.0.0
[20:10:52][C][logger:293]: Logger:
[20:10:52][C][logger:294]:   Level: DEBUG
[20:10:52][C][logger:295]:   Log Baud Rate: 115200
[20:10:52][C][logger:296]:   Hardware UART: USB_CDC
[20:10:52][C][uptime.sensor:031]: Uptime Sensor 'Uptime Sensor'
[20:10:52][C][uptime.sensor:031]:   Device Class: 'duration'
[20:10:52][C][uptime.sensor:031]:   State Class: 'total_increasing'
[20:10:52][C][uptime.sensor:031]:   Unit of Measurement: 's'
[20:10:52][C][uptime.sensor:031]:   Accuracy Decimals: 0
[20:10:52][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[20:10:52][C][gpio.output:010]: GPIO Binary Output:
[20:10:52][C][gpio.output:011]:   Pin: GPIO32
[20:10:52][C][restart:022]: Restart Switch 'Restart Pico W'
[20:10:52][C][restart:022]:   Icon: 'mdi:restart'
[20:10:52][C][adc:087]: ADC Sensor 'Core Temperature'
[20:10:52][C][adc:087]:   Device Class: 'voltage'
[20:10:52][C][adc:087]:   State Class: 'measurement'
[20:10:52][C][adc:087]:   Unit of Measurement: '°C'
[20:10:52][C][adc:087]:   Accuracy Decimals: 2
[20:10:52][C][adc:121]:   Pin: Temperature
[20:10:52][C][adc:125]:   Update Interval: 30.0s
[20:10:52][C][version.text_sensor:021]: Version Text Sensor 'ESPHome Version'
[20:10:52][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[20:10:52][C][wifi_info:013]: WifiInfo Mac Address 'ESP Mac Wifi Address'
[20:10:52][C][mdns:103]: mDNS:
[20:10:52][C][mdns:104]:   Hostname: pico-w-home
[20:10:52][C][ota:093]: Over-The-Air Updates:
[20:10:52][C][ota:094]:   Address: pico-w-home.local:2040
[20:10:52][C][ota:097]:   Using Password.
[20:10:52][C][api:138]: API Server:
[20:10:52][C][api:139]:   Address: pico-w-home.local:6053
[20:10:52][C][api:141]:   Using noise encryption: YES
[20:10:52][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi Signal Sensor'
[20:10:52][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[20:10:52][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[20:10:52][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[20:10:52][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[20:10:52][C][wifi_info:009]: WifiInfo IPAddress 'ESP IP Address'
[20:10:52][C][wifi_info:012]: WifiInfo BSSID 'ESP Connected BSSID'
```

si458, 2022-12-17:
This might of been fixed wit the latest release of 2022.12.1 

```
[07:13:18][I][app:102]: ESPHome version 2022.12.1 compiled on Dec 17 2022, 06:44:59
[07:13:18][C][wifi:504]: WiFi:
[07:13:18][C][wifi:362]:   Local MAC: 28:CD:C1:00:49:D8
[07:13:18][C][wifi:363]:   SSID: [redacted]
[07:13:18][C][wifi:364]:   IP Address: 192.168.23.107
[07:13:18][C][wifi:365]:   BSSID: [redacted]
[07:13:18][C][wifi:367]:   Hostname: 'pico-w-home'
[07:13:18][C][wifi:369]:   Signal strength: -45 dB ▂▄▆█
[07:13:18][C][wifi:373]:   Channel: 9
[07:13:18][C][wifi:374]:   Subnet: 255.255.255.0
[07:13:18][C][wifi:375]:   Gateway: 192.168.23.1
[07:13:18][C][wifi:376]:   DNS1: 192.168.23.1
[07:13:18][C][wifi:377]:   DNS2: 0.0.0.0
[07:13:18][C][logger:293]: Logger:
[07:13:18][C][logger:294]:   Level: DEBUG
[07:13:18][C][logger:295]:   Log Baud Rate: 115200
[07:13:18][C][logger:296]:   Hardware UART: USB_CDC
[07:13:18][C][uptime.sensor:031]: Uptime Sensor 'Uptime Sensor'
[07:13:18][C][uptime.sensor:031]:   Device Class: 'duration'
[07:13:18][C][uptime.sensor:031]:   State Class: 'total_increasing'
[07:13:18][C][uptime.sensor:031]:   Unit of Measurement: 's'
[07:13:18][C][uptime.sensor:031]:   Accuracy Decimals: 0
[07:13:18][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[07:13:18][C][gpio.output:010]: GPIO Binary Output:
[07:13:18][C][gpio.output:011]:   Pin: GPIO32
[07:13:18][C][restart:076]: Restart Switch 'Restart Pico W'
[07:13:18][C][restart:078]:   Icon: 'mdi:restart'
[07:13:18][C][restart:098]:   Restore Mode: restore defaults to OFF
[07:13:18][C][adc:087]: ADC Sensor 'Core Temperature'
[07:13:18][C][adc:087]:   Device Class: 'voltage'
[07:13:18][C][adc:087]:   State Class: 'measurement'
[07:13:18][C][adc:087]:   Unit of Measurement: '°C'
[07:13:18][C][adc:087]:   Accuracy Decimals: 2
[07:13:18][C][adc:121]:   Pin: Temperature
[07:13:18][C][adc:126]:   Update Interval: 30.0s
[07:13:18][C][version.text_sensor:021]: Version Text Sensor 'ESPHome Version'
[07:13:18][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[07:13:18][C][wifi_info:013]: WifiInfo Mac Address 'ESP Mac Wifi Address'
[07:13:18][C][mdns:103]: mDNS:
[07:13:18][C][mdns:104]:   Hostname: pico-w-home
[07:13:18][C][ota:093]: Over-The-Air Updates:
[07:13:18][C][ota:094]:   Address: pico-w-home.local:2040
[07:13:18][C][ota:097]:   Using Password.
[07:13:18][C][api:138]: API Server:
[07:13:18][C][api:139]:   Address: pico-w-home.local:6053
[07:13:18][C][api:141]:   Using noise encryption: YES
[07:13:18][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi Signal Sensor'
[07:13:18][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[07:13:18][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[07:13:18][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[07:13:18][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[07:13:18][C][wifi_info:009]: WifiInfo IPAddress 'ESP IP Address'
[07:13:18][C][wifi_info:012]: WifiInfo BSSID 'ESP Connected BSSID'
``1

bdraco, 2026-03-12:
Closing — this was fixed back in 2022.12.1 as confirmed by @si458. If anyone is still seeing wifi_info issues on RP2040 with a current version, please open a new issue.