esphome/issues #3797: Version 2022.11.x causes ESP32 to reboot constantly

Protoncek, 2022-11-17:
### The problem

I have an esp32 module, a couple of dallas sensors and BT proxy in my basement. Until today it ran on  esphome version 2022.10.2. Today i compiled uploaded latest esphome 2022.11.1 and now my esp32 constantly reboots. It starts, connects to HA for a short time, but when BT part starts it reboots. The last log output before reboot is this:
```
 [09:37:58][I][esp32_ble_client:154]: [0] [ff:ff:10:11:98:e1] Service UUID: 0x1800
[09:37:58][I][esp32_ble_client:156]: [0] [ff:ff:10:11:98:e1]  start_handle: 0x1  end_handle: 0x5
[09:37:58][I][esp32_ble_client:059]: [0] [ff:ff:10:11:98:e1]  characteristic 0x2A00, handle 0x3, properties 0x12
[09:37:58][I][esp32_ble_client:059]: [0] [ff:ff:10:11:98:e1]  characteristic 0x2A01, handle 0x5, properties 0x2
[09:37:58][I][esp32_ble_client:154]: [0] [ff:ff:10:11:98:e1] Service UUID: 0x180F
[09:37:58][I][esp32_ble_client:156]: [0] [ff:ff:10:11:98:e1]  start_handle: 0x6  end_handle: 0x8
[09:37:58][I][esp32_ble_client:059]: [0] [ff:ff:10:11:98:e1]  characteristic 0x2A19, handle 0x8, properties 0x12
[09:37:58][I][esp32_ble_client:154]: [0] [ff:ff:10:11:98:e1] Service UUID: 0x1802
[09:37:58][I][esp32_ble_client:156]: [0] [ff:ff:10:11:98:e1]  start_handle: 0x9  end_handle: 0xb
[09:37:58][I][esp32_ble_client:059]: [0] [ff:ff:10:11:98:e1]  characteristic 0x2A06, handle 0xb, properties 0x1c
[09:37:58][I][esp32_ble_client:154]: [0] [ff:ff:10:11:98:e1] Service UUID: 0xFFE0
[09:37:58][I][esp32_ble_client:156]: [0] [ff:ff:10:11:98:e1]  start_handle: 0xc  end_handle: 0xe
[09:37:58][I][esp32_ble_client:059]: [0] [ff:ff:10:11:98:e1]  characteristic 0xFFE1, handle 0xe, properties 0x12
INFO 192.168.0.92: Ping timed out!
INFO Disconnected from ESPHome API for 192.168.0.92
```
i tried clean compile in arduino and in esp-idf format, but same reboots happen. Then i installed esphome 2022.11.0, compiled and uploaded, but same reboots. Finally i went back to version 2022.10.2 and it works fine again. So, the problem, as is seems, started with version 11. 
I didn't, however, try to upload a program with BT part disabled...

### Which version of ESPHome has the issue?

2022.11.x

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.3

### What platform are you using?

ESP32

### Board

esp32 devkit v4

### Component causing the issue

BT part of board

bdraco, 2022-11-18:
Commenting out the `this->already_discovered` code seems to fix it for me

bdraco, 2022-11-18:
https://github.com/esphome/esphome/pull/3905

Protoncek, 2022-11-18:
Uh... i guess i'll wait for an update, since i don't know where to comment above line. I'm not that experienced in programming and github stuff...
But i'm happy that you found a problem!

codyc1515, 2022-11-18:
Experiencing a similar issue where my (inverted) keeps coming on repeatedly, presumably due to the restarts. Started in 2022.11.0.

mforbiz, 2022-11-21:
> i tried clean compile in arduino and in esp-idf format, but same reboots happen. Then i installed esphome 2022.11.0, compiled and uploaded, but same reboots. Finally i went back to version 2022.10.2 [...]
Hi Protoncek .
How may I downgrade ESP Home firmware version on my board that has just installed 2022.11.x ?
Thanks in advance.

Protoncek, 2022-11-21:
@mforbiz : when you updated i hope that you had enabled option "make backup before update" - in that case just go to system-backups, find "addon_xxxxxxxx_esphome_2022.10.2" and restore it. And disable automatic updates.

mforbiz, 2022-11-21:
> @mforbiz : when you updated i hope that you had enabled option "make backup before update" - in that case just go to system-backups, find "addon_xxxxxxxx_esphome_2022.10.2" and restore it. And disable automatic updates.

Fantastic !
It now works fine again.
Two  days'work will go for nothing  because of a wrong official update.
Many tanks.

Protoncek, 2022-11-23:
Hi!
Jusst for info: new version 2022.11.2 doesn't solve this problem - it's still the same problem. I tried with arduino and esp-idf.

mforbiz, 2022-11-23:
> Hi! Jusst for info: new version 2022.11.2 doesn't solve this problem - it's still the same problem. I tried with arduino and esp-idf.

I've just tried but even 2022.11.2 don't works fine. I must rollback 2022.10.2 again.

bdraco, 2022-11-23:
Try reflashing with the branch in this comment:

https://github.com/esphome/issues/issues/3761#issuecomment-1324176431

Protoncek, 2022-11-23:
Hm... i have "some problems" with that... can you advice how to include this into code? 
(sorry, i'm a pretty much total newbie in github...)

bdraco, 2022-11-23:
Clone the remote

checkout the branch

run ‘pip3 install --upgrade .’

use the esphome cli to reflash

dyeves, 2022-11-29:
> Try reflashing with the branch in this comment:
> 
> [#3761 (comment)](https://github.com/esphome/issues/issues/3761#issuecomment-1324176431)

I confirm it worked for me. Thanks

bdraco, 2022-11-29:
All the changes are merged to dev now and will be in 2022.12.x

Protoncek, 2022-11-30:
Thanks! I didn’t try above, since i’m not even sure how exactly…too complicated for me.
I’ll rather wait for v12.x.

mforbiz, 2022-12-16:
Hi,
After last 2022.12.1 the issue is fixed.
All works fine.

Protoncek, 2022-12-16:
Yes, issue is fixed with V12.x. Many thanks, bdraco!!!

closed 2022-12-16 by Protoncek