esphome/issues #3800: esp32_camera always idle

SebastianBrase85, 2022-11-17:
### The problem

With

- A AI-Thinker ESP32-CAM
- Using a esp32_camera: component
- Added to Home Assistant

The Camera will always show idle state and only deliver the idle framerate, no matter what I do in Home assistant. If I open the camera entity or make a panel on the dashboard, the status (and framerate) stays the same.

### Which version of ESPHome has the issue?

ESPHome version 2022.10.2 compiled on Nov 17 2022

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Home Assistant 2022.3.8

### What platform are you using?

ESP32

### Board

AI-Thinker ESP32-CAM

### Example YAML snippet

```yaml
esphome:
  name: espcam
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Esp Cam Fallback Hotspot"
    password: "UeGqYlVrJUt4"

captive_portal:

logger:
api:
ota:

esp32_camera:
  name: esp32_cam
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  max_framerate: 20fps
  idle_framerate: 1fps
  resolution: 1280x1024

# Example configuration entry
#esp32_camera_web_server:
#  - port: 8080
#    mode: stream
#  - port: 8081
#    mode: snapshot

output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4

binary_sensor:
  - platform: status
    name: "ESP32Cam status"
    
sensor:
  - platform: wifi_signal
    name: "ESP32Cam WiFi Signal"
    update_interval: 60s
    
light:
  - platform: binary
    output: gpio_4
    name: "ESP32Cam light"
```

disconn3ct, 2023-03-18:
nostale

ElVit, 2023-06-12:
Hi,
I have also the same issue. 
For some years I bought a [TTGO T-Camera V1.7](https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series).
It's been lying around ever since because I couldn't get it to work with ESPHome.
Please fix this issue 🙏🏼

Th3H0ff, 2023-06-12:
As a workaround I ended up running a Motioneye docker instance and having that pull the stream from the ESP32-cam. Apparently Motioneye can connect to the ESPhome camera web server stream, port 8080 or whatevery you wish. Home Assistant has a native integration for Motioneye so that was pretty straight forward.
HA ESP32-cam stream with a few extra steps.

Redwire1, 2024-01-27:
Had the same issue using a Freenove ESP32-Wrover. 
Gets the first frame only and status stays at idle.
Very slow even on 640x480 res.
Seems like the trigger to activate the stream does not work.

Rudd-O, 2024-02-14:
Extremely slow camera feed in HA compared to esp32_camera_web_server, but even that is slow compared to just the Arduino camera sketch.

What's happening behind the scenes that makes everything so slow?

I'm using latest ESPHome beta (Feb 2024 beta).

ronald-winter, 2024-02-29:
Same issue here. Camera always reports to be in idle. No picture at all. Not even in preview. Even though it works fine in the beginning.
Log shows a setup error:

```
[11:56:53][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 29 2024, 11:37:53
[11:56:53][C][wifi:577]: WiFi:
[11:56:53][C][wifi:409]:   Local MAC: [removed]
[11:56:53][C][wifi:414]:   SSID: [redacted]
[11:56:53][C][wifi:415]:   IP Address: [removed]
[11:56:53][C][wifi:417]:   BSSID: [redacted]
[11:56:53][C][wifi:418]:   Hostname: 'esp-cam01'
[11:56:53][C][wifi:420]:   Signal strength: -44 dB ▂▄▆█
[11:56:53][C][wifi:424]:   Channel: 6
[11:56:53][C][wifi:425]:   Subnet: 
[11:56:53][C][wifi:426]:   Gateway: [removed]
[11:56:53][C][wifi:427]:   DNS1: [removed]
[11:56:53][C][wifi:428]:   DNS2: 0.0.0.0
[11:56:53][C][logger:447]: Logger:
[11:56:53][C][logger:448]:   Level: DEBUG
[11:56:53][C][logger:449]:   Log Baud Rate: 115200
[11:56:53][C][logger:451]:   Hardware UART: UART0
[11:56:53][C][gpio.output:010]: GPIO Binary Output:
[11:56:53][C][gpio.output:011]:   Pin: GPIO4
[11:56:53][C][light:103]: Light 'flashlight'
[11:56:53][C][esp32_camera:048]: ESP32 Camera:
[11:56:53][C][esp32_camera:049]:   Name: My ESP32-CAM
[11:56:53][C][esp32_camera:050]:   Internal: NO
[11:56:53][C][esp32_camera:052]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[11:56:53][C][esp32_camera:053]:   VSYNC Pin: 25
[11:56:53][C][esp32_camera:054]:   HREF Pin: 23
[11:56:53][C][esp32_camera:055]:   Pixel Clock Pin: 22
[11:56:53][C][esp32_camera:056]:   External Clock: Pin:0 Frequency:20000000
[11:56:53][C][esp32_camera:060]:   I2C Pins: SDA:26 SCL:27
[11:56:53][C][esp32_camera:062]:   Reset Pin: -1
[11:56:53][C][esp32_camera:080]:   Resolution: 640x480 (VGA)
[11:56:53][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_FOUND
[11:56:53][C][psram:020]: PSRAM:
[11:56:53][C][psram:021]:   Available: YES
[11:56:53][C][psram:024]:   Size: 4095 KB
[11:56:53][C][captive_portal:088]: Captive Portal:
[11:56:53][C][mdns:115]: mDNS:
[11:56:53][C][mdns:116]:   Hostname: esp-cam01
[11:56:53][C][ota:096]: Over-The-Air Updates:
[11:56:53][C][ota:097]:   Address: esp-cam01.local:3232
[11:56:53][C][ota:100]:   Using Password.
[11:56:53][C][ota:103]:   OTA version: 2.
[11:56:53][C][api:139]: API Server:
[11:56:53][C][api:140]:   Address: esp-cam01.local:6053
[11:56:53][C][api:142]:   Using noise encryption: YES
```
YAML is roughly the same as shown above.

Coldness00, 2024-03-23:
Experiencing the same issue as well :/

tzven0, 2024-04-07:
same here, any workarounds for this?

dskw, 2024-05-15:
Same here starting now with the latest esphome (2024.5.0) update in home assistant.

I just updated the firmare and suddenly the camera reports:
[17:58:13][E][esp32_camera:123]:   Setup Failed: ESP_ERR_NOT_FOUND
[17:58:13][E][component:082]:   Component esp32_camera is marked FAILED

Coldness00, 2024-05-15:
In my case it s working fine now. I just  updated to the latest firmware.
No change in yaml code nor WiFi :/
I ll not complain, but it's kind of strange

dskw, 2024-05-17:
Yeah it started working for me too. I think it was solved by a hard reboot (unplug power for some seconds) of the board.
Maybe the last update introduced some memory corruption between flashing the new binary and the reboot?
It is working fine for me now at last.