esphome/issues #3804: Deleting external component from github repo does not revert to stock component; causes error.

bkaufx, 2022-11-17:
### The problem

When using external components from a github repo, deleting one of the components in the repo does not properly revert back to using the stock ESPHome component with the same name.   When ESPHome updates the repo, the local copy of the deleted external component's folder is emptied because it was deleted on the repo, but the folder still sticks around because of the `__pycache__` subfolder.

ESPHome then tries to proceed with basically an empty folder for one of the components instead of reverting to the stock component like it should, resulting in an error.

I'm going to keep looking at this and try to fix it, but figured I'd throw it on here to get feedback.  I think there are at least a couple potential solutions, but I'm not sure what would be preferable.

- when updating github repos, delete folders that only include `__pycache__` subfolders.
- delete all `__pycache__` subfolders before updating a repo.
- ignore component folders that are empty or that only have `__pycache__` subfolders during config checking.

### Which version of ESPHome has the issue?

2022.11.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP8266

### Anything in the logs that might be useful for us?

```txt
Component select cannot be loaded via YAML (no CONFIG_SCHEMA).
```

jesserockz, 2022-11-18:
Did you not remove that component from the list of `components` on the `external_component` entry?

bkaufx, 2022-11-18:
I just have it including all components.  Your suggestion is a reasonable workaround but this still feels like a bug to me.  ESPHome is causing an empty directory to exist and then throwing an error because it exists.

Probably the simplest solution I can see so far is changing the git command to wipe out local changes instead of saving them.  I'm not sure how important it is to save `__pycache__` folders.  No one should be making changes within the .esphome directory and expecting them to be saved long term.
https://github.com/esphome/esphome/blob/d8746266625c5a4c9127577297876e445cffe859/esphome/git.py#L92

Other ideas:

Maybe git.py could traverse through the repo_dir after refreshing a repo and delete any folder that only contains a `__pycache__` folder?
https://github.com/esphome/esphome/blob/d8746266625c5a4c9127577297876e445cffe859/esphome/git.py#L77-L103

Folders that only contain a `__pycache__` folder can be deleted in _process_git_config() of external_components before returning components_dir.
https://github.com/esphome/esphome/blob/d8746266625c5a4c9127577297876e445cffe859/esphome/components/external_components/__init__.py#L100-L119

In the external_components component, would it work to traverse through the directory structure and construct allowed_components based on directories that have `__init__.py`, rather than just assigning allowed_components to none at line 143?  
https://github.com/esphome/esphome/blob/d8746266625c5a4c9127577297876e445cffe859/esphome/components/external_components/__init__.py#L134-L152

Can you think of anywhere else this problem could be solved relatively easily?  Do you think this is something that doesn't need fixed?