esphome/issues #3806: Periodic Disconnect

FedgarLurch, 2022-11-18:
### The problem

For the past year or so, my ESP32 based esphome devices have been disconnecting very regularly. You can basically set a clock to how often they disconnect. It is one 5 minute disconnect, once per hour. But it is not based on the time of day. 

I am not entirely sure this isn't something to do with my router / configuration though. I have an Asus AC3200 running dd-wrt. I have replaced it and the problem persists, but I still can't 100% say it isn't related. 

[image]
In the photo, you will see 3 ESPhome devices. 
The Bed Sensor is a devkit doit V1. 
The aqm1 is a wemos d1 mini
The bench light plug is running on a sonoff that has been reflashed. I believe it is an ESP8266.  It does not have the boot issue. 

### Which version of ESPHome has the issue?

This problem has been here since 1.14 when I started using ESPhome. I am now on  ESPHome version 2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

All of them. I am currently up to date I believe. 2022.11.3?

### What platform are you using?

ESP32

### Board

DevKit Doit V1 / Wemos D1 mini / 1 other I can't recall right now. 

### Component causing the issue

It still happens even with nothing but an uptime sensor

### Example YAML snippet

```yaml
#Name. 
# Remember, some wifi networks don't like dashes. (especially Hidden SSIDs?)
esphome:
  name: $explicit_devicename

#Board type 
esp32:
  board: esp32dev
  framework:
    type: arduino
    
#Substitutions woo! 
substitutions:
  explicit_devicename: bedsensor
  devicename: Bed Sensor

#The most important part, wifi credentials
wifi:
  networks:
  - ssid: !secret home_wifi_ssid
    password: !secret home_wifi_password
    manual_ip:
      static_ip: 10.10.10.132
      gateway: 10.10.10.1
      subnet: 255.255.255.0
  #use_address: 10.10.10.98
# Secondary backup network.
  - ssid: !secret rashas_wifi_ssid
    password: !secret rashas_wifi_password
# Added, then removed, then re-added to combat instability
# Turn this off if you need to save power. 
  power_save_mode: light
  reboot_timeout: 0s
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$devicename Hotspot"
    password: !secret AP_hotspot_password

#Need this if you want to use the fallback hotspot
captive_portal:

logger:

api:

ota:
```
```

### Anything in the logs that might be useful for us?

```txt
I am honestly not sure what I can provide, but please tell me and I will be happy to update!
```

### Additional information

Thanks for all of the hard work you folks do!

0x90ChuckTessela, 2022-11-22:
Just a shot in the dark -- please forgive me it I'm stating something you've already checked.  I'm wondering if it might be the devices dropping off the network, and then rebooting after they lost connectivity (after about 10-15 minutes), which would then restore connectivity.  If that's the case, it might be something with the access points/network side of the house external to esp home.

FedgarLurch, 2023-03-10:
> 

It very well might be an access point issue. I am wondering if anyone else out here is using DDWRT and has set things up successfully. Maybe its just me and or my configuration. On the other hand it is weird that my ESP8266's are running fine, but the ESP32's have this problem. You would think the more sophisticated of the two would be more reliable.