esphome/issues #3807: BedJet - Can't connect to my Bedjet V3

firstdominator, 2022-11-18:
### The problem

Not connected, will not send status.

[image]

### Which version of ESPHome has the issue?

2022.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

Node-MCU

### Component causing the issue

Bedjet

### Anything in the logs that might be useful for us?

```txt
This is all I have at the moment on the code.

I know I don't have the climate and fan, jsut trying to debug, puttin gless info possible.

When I can for BLE, I see my BEDJET v3, and the mac address is good. Check over 10 times.
```

### Additional information

substitutions:
  name: esp32-ble-proxy-and-bedjet
packages:
  esphome.bluetooth-proxy: github://esphome/bluetooth-proxies/esp32-generic.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ble_client:
  - mac_address: 94:B5:55:30:E8:1E
    id: BEDJET_V3

bedjet:
  - id: bedjet_1
    ble_client_id: BEDJET_V3

firstdominator, 2022-11-18:
One more thing my esp32 is just next to the Bedjet.

firstdominator, 2022-11-19:
OK so after some more digging.... I started from fresh and remove the BLE proxy from the ESP32.
Now it's seems I'm going somewhere since it's getting some data from the Bedjet.... BUt I'm still stuck.

As soon it's get some data from the BedJet the ESPHOME API crashed and disconnects the ESP32.

[image]

If I remove the bedjet component the API doesn't crash anymore. 

My code is pretty simple it's the minimum possible. Don't even want to add the BedJet climate component since I want first to get this API not crashing.

Here is my code.
[image]

Can someone try to help me here? I actually brought the Bedjet V3 just for that.... If I can't control it from HA i don't want it. PLEASE  @jhansche 

Thanks!

jhansche, 2022-11-19:
My first thought was also the ble proxy, because I see app_id 4, and the esp32 only supports 3 connections.

But without the proxy, it looks like your esp32 is crashing, but because you're getting logs over the air, you can't see the crash. Can you try plugging the device into USB and get logs directly from the UART console instead? That'll be the only way to see the actual crash logs. Basically, instead of powering it with a wall wart, plug it into your laptop or computer; and when you select logs or install, instead of choosing "wirelessly", select the option for USB over this computer. You can also use the esphome command line.

Once you have the UART console log, it should show where the crash is happening.

Is your BedJet firmware up to date? Have you been able to pair it with your phone app yet? And have you tried power cycling the BedJet (just unplug and replug)?

firstdominator, 2022-11-19:
@jhansche 
Here is the log. I just letft in there one intance for clearer reading. 
[logs.txt](https://github.com/esphome/issues/files/10047730/logs.txt)

Yes my firmware is up to date V.120 (140).
Yes it has been paired with the phone. I know the BLE_Client isn't found when the app is open, but closed what mentionned above happens.

I did try factory reset of bedjet, unplug replug, hold down the LOW Power for pairing, unplug the bedjet remote, disabling wifi on it. I tried all sort of combinations.

Any posibility that my esp32 board is too old. Got those like 1.5-2 years ago.
NODEMCU - ESP-32s

I did try with 3 different ESP32 boards, (2x NODEMCU - ESP-32s and 1x ESP32 DEVKITv1)

Thanks for your help.

jhansche, 2022-11-19:
Yeah, usually when you run with logs from the same source that compiled the image, it should unwind the stacktrace. This panic can't be read unless you decode it with your compiled symbols table.

How did you run this? Was it through the browser or through the command line esphome? If it was through the command line, can you copy your yaml config to a file in the directory, and install it using the command line as well? That will ensure that it has the compiled symbols available. Otherwise if it was installed over the browser, and pulling logs from somewhere else, it wouldn't decode it. Might need some help figuring out how to get a proper stack trace

firstdominator, 2022-11-21:
This was done with Edge Broswer (since Brave doesn't support WebSerial.)

With the WebESPHome and the device connected to the computer. This is how I got the logs.

firstdominator, 2022-11-21:
So i just tough I could try this onver HTTPS with my NabuCasa link.

I was able to connect to with with Microsoft Edge and the ESPHome addons from my NabuCasa instead of the ESP HOME WEB.

Same result, visually. 

Here the log from NabuCasa ESPHome addon on my HA instance. 

[bedjet_logs.txt](https://github.com/esphome/issues/files/10051389/bedjet_logs.txt)

Thanks!

jhansche, 2022-11-21:
It's hard to say without having the stack trace. When you're connecting it via USB and getting the console logs, are you also installing at the same time, or only clicking "logs"?  It does seem to panic right as it's connecting, but without the stack trace I can't tell what is causing it. The panic you see in the logs just shows memory addresses, but that doesn't help by itself - those addresses have to be decoded against your compiled symbols in order to know what function was executing when the panic happened.

You could try to narrow down what could be causing it, by commenting out the BedJet component, and see if it still happens, since it might be just the ble tracker itself. In fact, I don't see esp32_ble_tracker in your config, and I thought that was required, but I may be misremembering. I'm away for the next week or so, so I wouldn't be able to make any fixes anyway until I'm back. But I have a feeling this is either a misconfiguration, or a hardware issue. If you can either rule out the BedJet component, or get the decoded stack trace, that will help figure out the root cause.

You could also try jumping on the esphome discord to get some help decoding the panic stack. I can't seem to find the documentation about that right now, but I know it is supposed to decode the panic for you automatically, so I'm not sure why it isn't.

jhansche, 2022-11-21:
As for supported boards, you can check https://esphome.io/components/esp32.html and https://registry.platformio.org/packages/platforms/platformio/espressif32/boards

gardiner4, 2022-11-22:
I have the same issue. I have been using it successfully for months and the latest update seems to be crashing the ESP32.

jhansche, 2022-11-22:
@gardiner4 are you able to get a decoded panic stack trace?

gardiner4, 2022-11-22:
Is this it?

[14:56:03][D][wifi:386]: Starting sca\xc0\xe0\xe0\xf0\xe0\xe0\xe0j5
[14:56:03][D][esp32_ble_tracker:809]: Found device 52:A2:32:53:8C:12 RSSI=-70
[14:56:03][D][esp32_ble_tracker:830]:   Address Type: RANDOM
[14:56:03][D][esp32_ble_tracker:809]: Found device 4C:C9:5E:0D:14:EB RSSI=-91
[14:56:03][D][esp32_ble_tracker:830]:   Address Type: PUBLIC
[14:56:04]\xe0\xc0\xc0\xc0\xc0\xe0\xe0]j\x95\x91\xa5хѥ\xbd\xb9\x81Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[14:56:04]Memory dump at 0x401de784: 82ff3361 788c0016 026d440c
[14:56:04]Core 0 register dump:
[14:56:04]PC      : 0x401de78a  PS      : 0x00060933  A0      : 0x801cf0a2  A1      : 0x3ffd2970  
WARNING Decoded 0x401de78a: r_rwip_sleep
[14:56:04]A2      : 0x00000001  A3      : 0x3ffc3ec8  A4      : 0x3ffb8d40  A5      : 0x3ffc3eb0  
[14:56:04]A6      : 0x3ffb8bcc  A7      : 0x3ffc3eb8  A8      : 0x00000000  A9      : 0x3ffd2950  
[14:56:04]A10     : 0x00000001  A11     : 0x00000000  A12     : 0x0000001f  A13     : 0x00000001  
[14:56:04]A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000000  
[14:56:04]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[14:56:04]
[14:56:04]ELF file SHA256: 0000000000000000
[14:56:04]
[14:56:04]Backtrace: 0x401de78a:0x3ffd2970 0x401cf09f:0x3ffd29b0 0x401cf69d:0x3ffd29d0 0x400909d6:0x3ffd2a00
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x401de78a: r_rwip_sleep
WARNING Decoded 0x401cf09f: r_rw_schedule
WARNING Decoded 0x401cf69d: btdm_controller_task
WARNING Decoded 0x400909d6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[14:56:04]
[14:56:04]Rebooting...

jhansche, 2022-11-22:
Yeah that's the decoded trace I'm looking for. Unfortunately none of that belongs to the BedJet component, so it might be an Arduino or espressif platform issue. 

Have either of you tried using the esp-idf framework instead of Arduino? I would recommend trying that next, just to rule out the Arduino framework.

gardiner4, 2022-11-22:
When I comment out the BedJet lines it does not crash. Esp-idf hangs at starting scan......

[16:10:14][D][esp-idf:000]: I (1637) wifi_init: tcpip mbox: 32
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1637) wifi_init: udp mbox: 6
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1646) wifi_init: tcp mbox: 6
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1656) wifi_init: tcp tx win: 5744
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1657) wifi_init: tcp rx win: 5744
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1667) wifi_init: tcp mss: 1440
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1667) wifi_init: WiFi IRAM OP enabled
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1677) wifi_init: WiFi RX IRAM OP enabled
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1691) wifi:
[16:10:14][D][esp-idf:000]: mode : sta (0c:b8:15:c1:41:98)
[16:10:14][D][esp-idf:000]: 
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1698) wifi:
[16:10:14][D][esp-idf:000]: enable tsf
[16:10:14][D][esp-idf:000]: 
[16:10:14]
[16:10:14][D][esp-idf:000]: I (1708) wifi:
[16:10:14][D][esp-idf:000]: Set ps ty\x95\xe9 1
[16:10:14]
[16:10:14][D][esp-idf:000]: 
[16:10:14]
[16:10:14][D][wifi:386]: Starting scan...

gardiner4, 2022-11-22:
When I comment out BedJet the esp32 does not crash. I just tried esp-idf and it does briefly connect to the BedJet, but eventually crashes in a similar fashion.

gardiner4, 2022-11-23:
I did not have a backup so I used Portainer to recreate the docker image with the 2022.10.2 version. The issue is gone, and my BedJet is now working. I see some other open issues of ESP32s restarting in 2022.11.xx.

ChrisCarde, 2022-11-23:
Can confirm - this also broke in the approximate timeframe of upgrading ESPHome -- or perhaps even HA itself -- to 2022.11.x. I had been using it successfully for a month until then -- notably *on the same ESP32 device*.

kquinsland, 2022-11-24:
I have ESPhome 2022.11 working with my bedjet but I have noticed some increased instability.
I noticed that something wasn't quite right because a portion of my HA dashboard was missing.

I have ESPhome expose the fan speed as a slider directly to HA because there still isn't a built in UI widget for that.

```yaml
# We already have fan speed control via the climate entity in HA but there's no automagic UI element for easy
#  fan speed control. We fix that here.
number:
  - name: "${friendly_name_short} Fan Speed"
    platform: template
    id: inp_fan_speed
    icon: "mdi:fan"
    entity_category: config
    unit_of_measurement: "%"
    mode: "slider"
    min_value: 0
    max_value: 100
    step: 5
    # We'll get a string like `85%` back. Delete the last char, turn into a float and we're golden :)
    lambda: |-
      std::string speed = id(climate_bedjet_88fa).custom_fan_mode.value();
      speed.pop_back();
      return atoi(speed.c_str());

    # Called when HA sets the number
    set_action:
      then:
        - lambda: |-
            auto call = id(climate_bedjet).make_call();
            call.set_fan_mode(str_sprintf("%i%%", (int)x));
            call.perform();
```

It was _this_ slider that wasn't showing up as expected that tipped me off.
I had ~35 min I could spend on investigation so I didn't get far. Here's what I tried

- Update to `v2022.11.1`

After a new flash / cold boot, there's usually some difficulty in connecting. This time was no exception and it took a few crash -> boot cycles before the module managed to connect to the bedjet.

I was working from a terminal that didn't have a particularly large scrollback at the time so I don't have the exact details but I distinctly remember one of the errors just before the stack trace was dumped. It was lamenting the lack of room for JSON buffer.

That would make sense; if there's some issue w/ allocating the JSON buffer, the auto-discovery payload wouldnt get published and the slider wouldn't show up.

Unfortunately that was only one of the ~5 stack traces I saw before the module appeared to stabalize.

I was running out of time so I put a simple `on_time:` automation in place so the module reboots every day at 2 PM. It's been stable since then.

¯\_(ツ)_/¯

nochlab1, 2022-11-25:
I am having i guess same issue tried 2 different esp32 and both times getting this

[22:48:37][I][bedjet:216]:      - Command char: 0x33
[22:48:37][I][bedjet:217]:      - Status char: 0x2a
[22:48:37][I][bedjet:218]:        - config descriptor: 0x2b
[22:48:37][I][bedjet:219]:      - Name char: 0x2d
[22:48:37][D][bedjet:237]: [30:c6:f7:39:46:86] Services complete: obtained char handles.
[22:48:37]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[22:48:37]Core 1 register dump:
[22:48:37]PC      : 0x4000c2e0  PS      : 0x00060230  A0      : 0x800deaec  A1      : 0x3ffce4e0  
[22:48:37]A2      : 0x3ffce574  A3      : 0x00000000  A4      : 0x00000012  A5      : 0x3ffce574  
[22:48:37]A6      : 0x00060223  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000001  
[22:48:37]A10     : 0x3fff5a84  A11     : 0x3fff67dd  A12     : 0x3fff6544  A13     : 0x3fff65d4  
[22:48:37]A14     : 0x3fff6525  A15     : 0x3fff65d5  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[22:48:37]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[22:48:37]
[22:48:37]ELF file SHA256: 0000000000000000
[22:48:37]
[22:48:37]Backtrace: 0x4000c2e0:0x3ffce4e0 0x400deae9:0x3ffce4f0 0x400ddd2e:0x3ffce510 0x400dab15:0x3ffce5b0 0x401f52a1:0x3ffce5e0 0x400df815:0x3ffce600 0x401f5559:0x3ffce700 0x401f5635:0x3ffce720 0x400e5092:0x3ffce740 0x400e757a:0x3ffce770 0x400f42dc:0x3ffce790 0x400909d6:0x3ffce7b0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400deae9: esphome::esp32_ble_tracker::ESPBTUUID::get_uuid() const at /config/esphome/.esphome/build/bedjet/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:567
WARNING Decoded 0x400ddd2e: esphome::esp32_ble_client::BLEClientBase::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /config/esphome/.esphome/build/bedjet/src/esphome/components/esp32_ble_client/ble_client_base.cpp:286
WARNING Decoded 0x400dab15: esphome::ble_client::BLEClient::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /config/esphome/.esphome/build/bedjet/src/esphome/components/ble_client/ble_client.cpp:54
WARNING Decoded 0x401f52a1: esphome::esp32_ble_tracker::ESP32BLETracker::real_gattc_event_handler_(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /config/esphome/.esphome/build/bedjet/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:368
WARNING Decoded 0x400df815: esphome::esp32_ble_tracker::ESP32BLETracker::loop() at /config/esphome/.esphome/build/bedjet/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:78
WARNING Decoded 0x401f5559: esphome::Component::call_loop() at /config/esphome/.esphome/build/bedjet/src/esphome/core/component.cpp:150
WARNING Decoded 0x401f5635: esphome::Component::call() at /config/esphome/.esphome/build/bedjet/src/esphome/core/component.cpp:150
WARNING Decoded 0x400e5092: esphome::Application::loop() at /config/esphome/.esphome/build/bedjet/src/esphome/core/application.cpp:74
WARNING Decoded 0x400e757a: loop() at /config/esphome/.esphome/build/bedjet/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.h:166
WARNING Decoded 0x400f42dc: loopTask(void*) at /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23
WARNING Decoded 0x400909d6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[22:48:38]
[22:48:38]Rebooting..

firstdominator, 2022-12-04:
Still same issue going on... Anyone got a resolution?

jhansche, 2022-12-04:
@kquinsland unrelated: you could try the new `fan` platform to replace your template number slider. The hub component also exposes fan speed controls now, without having to use the custom fan preset string, if you want to continue using the number template. The docs explain how to add the BedJet fan component.

As for the crash, my esp32 has been running 2022.11.1 since Nov 22, without any issues. I'm using `framework: esp-idf`, without HA BLE proxy, and with log level set to `INFO`. Note that using a more verbose log level has caused stability issues for me in the past. I'm also using some slightly relaxed BLE scan parameters, though I'm not sure that it's making any difference:

```
esp32_ble_tracker:
  scan_parameters:
    interval: 1sec # 320ms
    window: 300ms  # 30ms
    duration: 5min # 5min
```

At this point I've seen several different stack traces that all point to different places in the code that don't appear to be directly caused by the BedJet component. There has been a lot of work done recently in the BLE stack (including BLE proxy), so unless I can reproduce it myself, or get a consistent stack trace that shows BedJet is the culprit, I can only guess that is caused by something else in the stack.

nochlab1, 2022-12-04:
> Still same issue going on... Anyone got a resolution?

i used linux to install a older version of esphom and it works

kquinsland, 2022-12-04:
> @kquinsland unrelated: you could try the new `fan` platform to replace your template number slider. The hub component also exposes fan speed controls now, without having to use the custom fan preset [...]

Thanks for confirming. I was thinking that there might be some instability induced in the bed jet component by way of changes in other components.

Regardless, the reason for the number slider is that HA does not have any fan card. If you add a `fan.my_bedjet` entity to a lovelace card, you'll get a single toggle switch. You have to resort to a custom card or other DIY automations if you want to toggle the speed.

I did recently discover the mushroom cards, though, and there is a _fantastic_ fan card that does support percentages so I've switched to that :).

> using `framework: esp-idf`, without HA BLE proxy, and with log level set to `INFO`. Note that using a more verbose log level has caused stability issues for me in the past. I'm also using some [...]

Good tip! In addition to removing the template number, i'll remove my daily reboot and switch to ESP-IDF w/ those relaxed values.

> At this point I've seen several different stack traces that all point to different places in the code that don't appear to be directly caused by the BedJet component. There has been a lot of work done recently in the BLE stack (including BLE proxy), so unless I can reproduce it myself, or get a consistent stack trace that shows BedJet is the culprit, I can only guess that the issue is caused by something else in the stack.

While implementing the changes I mentioned above, I got two. Both happened immediately after flashing / initial reset.
The stack traces come from a build using your scan parameters and:

```yaml
esp32:
  board: mhetesp32minikit
  framework:
    type: esp-idf
```

The build does _not_ have the template `number` component but does have the bedjet `fan` component and this version;

```shell
❯ esphome version
Version: 2022.11.4
```

```
[09:56:23][I][esp32_ble_client:054]: [0] [de:ad:be:ef:88:fa] Attempting BLE connection
[09:56:26][I][esp32_ble_client:154]: [0] [de:ad:be:ef:88:fa] Service UUID: 0x1801
[09:56:26][I][esp32_ble_client:156]: [0] [de:ad:be:ef:88:fa]  start_handle: 0x1  end_handle: 0x5
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 0x2A05, handle 0x3, properties 0x20
[09:56:26][I][esp32_ble_client:154]: [0] [de:ad:be:ef:88:fa] Service UUID: 0x1800
[09:56:26][I][esp32_ble_client:156]: [0] [de:ad:be:ef:88:fa]  start_handle: 0x14  end_handle: 0x1c
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 0x2A00, handle 0x16, properties 0x2
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 0x2A01, handle 0x18, properties 0x2
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 0x2AA6, handle 0x1a, properties 0x2
[09:56:26][I][esp32_ble_client:154]: [0] [de:ad:be:ef:88:fa] Service UUID: 00001000-BED0-0080-AA55-4265644A6574
[09:56:26][I][esp32_ble_client:156]: [0] [de:ad:be:ef:88:fa]  start_handle: 0x28  end_handle: 0xffff
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002000-BED0-0080-AA55-4265644A6574, handle 0x2a, properties 0x1a
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002001-BED0-0080-AA55-4265644A6574, handle 0x2d, properties 0xa
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002002-BED0-0080-AA55-4265644A6574, handle 0x2f, properties 0xa
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002003-BED0-0080-AA55-4265644A6574, handle 0x31, properties 0x8
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002004-BED0-0080-AA55-4265644A6574, handle 0x33, properties 0x8
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002005-BED0-0080-AA55-4265644A6574, handle 0x35, properties 0xa
[09:56:26][I][esp32_ble_client:059]: [0] [de:ad:be:ef:88:fa]  characteristic 00002006-BED0-0080-AA55-4265644A6574, handle 0x37, properties 0xa
[09:56:26][I][bedjet:215]: [de:ad:be:ef:88:fa] Discovered service characteristics: 
[09:56:26][I][bedjet:216]:      - Command char: 0x33
[09:56:26][I][bedjet:217]:      - Status char: 0x2a
[09:56:26][I][bedjet:218]:        - config descriptor: 0x2b
[09:56:26][I][bedjet:219]:    Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[09:56:26]
[09:56:26]Core  1 register dump:
[09:56:26]PC      : 0x4000c46c  PS      : 0x00060030  A0      : 0x800d339a  A1      : 0x3ffcf720  
[09:56:26]A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x00000023  A5      : 0x0000000c  
[09:56:26]A6      : 0x00000001  A7      : 0x00000002  A8      : 0xfa000000  A9      : 0x00000000  
[09:56:26]A10     : 0x0f100000  A11     : 0xb9800000  A12     : 0x00060e23  A13     : 0x00060e23  
[09:56:26]A14     : 0x00000001  A15     : 0x0000ea60  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
[09:56:26]EXCVADDR: 0x0000000c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000001  
[09:56:26]
[09:56:26]Backtrace:0x4000c469:0x3ffcf720 0x400d3397:0x3ffcf730 0x400d4524:0x3ffcf750 0x400d6192:0x3ffcf7a0 0x401b9d69:0x3ffcf7d0 0x400d96e7:0x3ffcf7f0 0x401ba87d:0x3ffcf8f0 0x401ba969:0x3ffcf910 0x400e4ada:0x3ffcf930 0x400e7412:0x3ffcf960 0x400d729b:0x3ffcfa80
WARNING Found stack trace! Trying to decode it
[... cut ...]
WARNING Decoded 0x400e4ada: esphome::Application::setup() at /home/karl/Projects/esphome/hardware/projects/.esphome/build/bedjet-controller/src/esphome/core/application.cpp:52 (discriminator 2)
WARNING Decoded 0x400e7412: setup() at /home/karl/Projects/esphome/hardware/projects/.esphome/build/bedjet-controller/projects/bedjet.yaml:251 (discriminator 4)
WARNING Decoded 0x400d729b: esphome::loop_task(void*) at /home/karl/Projects/esphome/hardware/projects/.esphome/build/bedjet-controller/src/esphome/components/esp32/core.cpp:72
[10:01:39]
[10:01:39]
[10:01:39]ELF file SHA256: 353ab9a5bf847275
[10:01:39]
[10:01:39]Rebooting...
[10:01:39]ets Jun  8 2016 00:22:57
```

jhansche, 2022-12-04:
OK, yeah those 3 stack traces point to:
https://github.com/esphome/esphome/blob/dev/esphome/components/bedjet/bedjet_codec.cpp#L95
which is clearing the memory of the status packet that was already allocated here: https://github.com/esphome/esphome/blob/dev/esphome/components/bedjet/bedjet_codec.h#L187
Since the class already owns that memory, there's no reason that it should be prohibited, and it's not even a memory allocation issue, because it is not a pointer so the class already allocates that memory when it's first created. You could be right that there's some kind of memory issue in that board, but I don't know how I would track that down since it doesn't seem to happen for me. I haven't used the `mhetesp32minikit` board - Mine is a generic `board: esp32dev`.

I'll see if I can get some time to verify that: this could actually be happening to me as well, but the watchdog causes it to reboot after panic automatically, and maybe that's why my "edit/edit2" comments above took longer than usual for it to come online 🤔  If that's the case, then I would also only see the logs while plugged into USB.

The fan index sounds like a bug, and your explanation makes sense. Feel free to open a ticket or PR for that and I'll take a look. Should be simple enough. I think the root of the issue is that HA wants to allow setting 0%, but that should actually be "off". I don't think we have a way to tell HA that we want to start at min=5% ("fan speed index 0 == 5%"), so we probably need to update the fan component to have a range of 0-20 and subtract one when sending the speed to BedJet, with a special case for 0==off.

kquinsland, 2022-12-04:
> OK, yeah those 3 stack traces point to:
> https://github.com/esphome/esphome/blob/dev/esphome/components/bedjet/bedjet_codec.cpp#L95
> which is clearing the memory of the status packet that was already [...]

Thanks for the explanation. I didn't go digging through the code but could see the call stack and the basic "handler/hook/callback function -> decode function" pattern just smelled like some sort of unexpected bytes which isn't unheard of on embedded electronics ... especially when doing a "warm" boot just after flashing.

> Since the class already owns that memory, there's no reason that it should be prohibited, and it's not even a memory allocation issue, because it is not a pointer so the class already allocates that [...]

This is the board in question: https://www.amazon.com/Bluetooth-Internet-Development-Functional-Ardu-ino/dp/B08MFCC4SR

(that was the first link I found; the board is sold under a zillion different brands / stores / etc. I buy mine in bulk from Ali Express for ~$3 each)

It's basically as close to a generic/dev unit as you can get. It's a basic USB <-> UART chip with ESD protection and an LDO to power the ESP32 module.

There might be some difference in the _generation_ of ESP32 chips, perhaps? E.G.: the way the compiler works for `extensa` stuff results in a prohibited exception from time to time but the same class is compiled in an exception-free way for the newer RISC based cores?  I'm grasping at straws to explain it :).

> I'll see if I can get some time to verify that: this could actually be happening to me as well, but the watchdog causes it to reboot after panic automatically, and maybe that's why my "edit/edit2" comments above took longer than usual for it to come online thinking If that's the case, then I would also only see the logs while plugged into USB.

Not sure what `edit/edit2` you're referring to. I was getting the provided stack traces via USB so it could very well be that log lines emitted to MQTT are not seen before the reboot.

It could also be a bug in the underlying SDK/APIs that's exacerbated by my wireless environment.  /shrug.

> Feel free to open a ticket or PR for that and I'll take a look. Should be simple enough.

Will do!  Didn't know if you wanted a distinct thread or not.

nochlab1, 2022-12-04:
It works on a older version so its not a issue in hardware it was broken in software by mistake

jhansche, 2022-12-05:
> Not sure what `edit/edit2` you're referring to. I was getting the provided stack traces via USB so it could very well be that log lines emitted to MQTT are not seen before the reboot.

The edits I'm talking are at the end of this comment earlier in this thread: https://github.com/esphome/issues/issues/3807#issuecomment-1324278203
I tested updating to latest esphome, but I was out of town so the update happened blindly. At first I didn't see it come back up as quickly as I'd expect so I assumed that the update also broke mine. But it did eventually come back up, with no intervention. What I'm saying now is that mine may have actually hit a crash but got automatically restarted and then remained stable after that (which sounds like that's what you're seeing now?)

> It works on a older version so its not a issue in hardware it was broken in software by mistake

I understand what you're saying, but the relationship between hardware and software is not as cut-and-dry as this implies. Applying your logic to my situation actually says the opposite: "I updated with no issues therefore the software must be fine." However that's not what I'm saying. I'm aware that multiple people are reporting this issue, but the issue is not affecting me or my devices. If it was just "broken in software" as your suggesting, then it would have broken for me, and it didn't. Therefore there must be something else at play that either a) makes it work unexpectedly for me (that is, it's truly broken in software, but by some fluke mine didn't break), or b) makes it break unexpectedly for the handful of you in this thread.

Most commonly and most likely, it's something very specific, like a particular board manufacturers, something else in the esphome config or framework, the power source, flash layout, or dozens of other things that could be aligning (or misaligning) just right, to either make it work or make it break.

I'll also point out that the BedJet code itself hasn't changed in a few months (Aug 30); but the BLE proxy changes in October did precipitate some refactoring (doesn't appear to have made any functional changes however, at least not in the BedJet component itself)
https://github.com/esphome/esphome/commits/dev/esphome/components/bedjet

It's possible that something did change in the Bluetooth stack or some other hardware-specific code path, but I'm not aware of anything specific. So again, I can't really do anything here if I can't reproduce the problem myself.

firstdominator, 2022-12-06:
SO Ir ead everything, but you guys are talking a different language that I can't follow... I still have my issue and can't seem to resolve. Can someone give me the minimal code at least so I can connect to it. With my code on the orignal post, I'm not able too...

I'm still stuck at the same point.

Thanks in advance.

kquinsland, 2022-12-06:
> The edits I'm talking are at the end of this comment earlier in this thread: [#3807 (comment)](https://github.com/esphome/issues/issues/3807#issuecomment-1324278203)
> I tested updating to latest esphome, but I was out of town so the update happened blindly. At first I didn't see it come back up as quickly as I'd expect so I assumed that the update also broke mine. But it did eventually come back up, with no intervention. What I'm saying now is that mine may have actually hit a crash but got automatically restarted and then remained stable after that (which sounds like that's what you're seeing now?)

Ah, i get you now. Yeah, very likely could be the case that a watchdog hid the issue from you initially.  For what it's worth, my ESP32 module has been up a bit over 26h so far after I switched to `esp-idf` and used your relaxed bluetooth settings. That's more stable than things had been.  I'll keep an eye on it :).

> Can someone give me the minimal code at least so I can connect to it. With my code on the orignal post, I'm not able too...

@firstdominator 

We need the "decocded" stack trace to figure out what _your_ crash means. You'll want to access `esphome` program directly so you can `esphome run path/to/your.yaml`. That will compile and then upload. If the crash still happens, `esphome` should be able to detect it and will be able to put meaning to the addresses.

It'll look like this:

```
esphome run your.yaml
<...>
[10:19:22]Core  1 register dump:
[10:19:22]PC      : 0x4000c46c  PS      : 0x00060030  A0      : 0x800d3342  A1      : 0x3ffcf790  
[10:19:22]A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x00000023  A5      : 0x0000000c  
[10:19:22]A6      : 0x00000001  A7      : 0x00000002  A8      : 0xfa000000  A9      : 0x00000000  
[10:19:22]A10     : 0x0eec0000  A11     : 0xe3800000  A12     : 0x00060e23  A13     : 0x00060e23  
[10:19:22]A14     : 0x00000001  A15     : 0x0000ea60  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
[10:19:22]EXCVADDR: 0x0000000c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000001  
[10:19:22]
[10:19:22]Backtrace:0x4000c469:0x3ffcf790 ....
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400d333f: esphome::bedjet::BedjetCodec::decode_notify(unsigned char const*, unsigned short) at /home/karl/Projects/esphome/hardware/projects/.esphome/build/bedjet-controller/src/esphome/components/bedjet/bedjet_codec.cpp:95
<....>
```

nochlab1, 2023-03-30:
> Ah, i get you now. Yeah, very likely could be the case that a watchdog hid the issue from you initially. For what it's worth, my ESP32 module has been up a bit over 26h so far after I switched to [...]

[03:14:44][D][bedjet:239]: [94:B5:55:17:E8:CA] Services complete: obtained char handles.
[03:14:44][D][esp32_ble_tracker:246]: Starting scan...
[03:14:44][D][bedjet:435]: [94:B5:55:17:E8:CA] wrote notify=true to status config 0x002b, for conn 0
[03:14:44]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[03:14:44]
[03:14:44]Core  1 register dump:
[03:14:44]PC      : 0x400909a0  PS      : 0x00060330  A0      : 0x800d9a51  A1      : 0x3ffcf540
WARNING Decoded 0x400909a0: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150
[03:14:44]A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x00000023  A5      : 0x0000000c
[03:14:44]A6      : 0x3ffc3e24  A7      : 0x00000002  A8      : 0xfa000000  A9      : 0x00000000
[03:14:44]A10     : 0x0f990000  A11     : 0xb3400000  A12     : 0x3ffcc7dc  A13     : 0x3fff44e4
[03:14:44]A14     : 0x3ffc55c0  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d
[03:14:44]EXCVADDR: 0x0000000c  LBEG    : 0x400909a0  LEND    : 0x400909ab  LCOUNT  : 0x00000001
[03:14:44]
[03:14:44]
[03:14:44]Backtrace:0x4009099d:0x3ffcf5400x400d9a4e:0x3ffcf550 0x400dae00:0x3ffcf570 0x400dcc66:0x3ffcf5c0 0x402077b9:0x3ffcf5f0 0x400e064d:0x3ffcf610 0x40207781:0x3ffcf630 0x400df2fd:0x3ffcf650 0x40207b75:0x3ffcf670 0x40207c35:0x3ffcf690 0x400e8252:0x3ffcf6b0 0x400ea9dd:0x3ffcf6e0 0x400f5e82:0x3ffcf7e0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4009099d: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:143
WARNING Decoded 0x400d9a4e: esphome::bedjet::BedjetCodec::decode_notify(unsigned char const*, unsigned short) at /home/nochum/.esphome/build/bedjet-3/src/esphome/components/bedjet/bedjet_codec.cpp:95
WARNING Decoded 0x400dae00: esphome::bedjet::BedJetHub::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/components/bedjet/bedjet_hub.cpp:378
WARNING Decoded 0x400dcc66: esphome::ble_client::BLEClient::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/components/ble_client/ble_client.cpp:58 (discriminator 2)
WARNING Decoded 0x402077b9: esphome::esp32_ble_tracker::ESP32BLETracker::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:346 (discriminator 2)
WARNING Decoded 0x400e064d: non-virtual thunk to esphome::esp32_ble_tracker::ESP32BLETracker::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*)
WARNING Decoded 0x40207781: esphome::esp32_ble::ESP32BLE::real_gattc_event_handler_(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/components/esp32_ble/ble.cpp:208 (discriminator 2)
WARNING Decoded 0x400df2fd: esphome::esp32_ble::ESP32BLE::loop() at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/components/esp32_ble/ble.cpp:158
WARNING Decoded 0x40207b75: esphome::Component::call_loop() at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/core/component.cpp:76
WARNING Decoded 0x40207c35: esphome::Component::call() at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/core/component.cpp:98
WARNING Decoded 0x400e8252: esphome::Application::setup() at /home/_Redacted_/.esphome/build/bedjet-3/src/esphome/core/application.cpp:52 (discriminator 2)
WARNING Decoded 0x400ea9dd: setup() at /home/_Redacted_/.esphome/build/bedjet-3/src/main.cpp:330
WARNING Decoded 0x400f5e82: loopTask(void*) at /home/_Redacted_/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
[03:14:45]
[03:14:45]
[03:14:45]
[03:14:45]
[03:14:45]ELF file SHA256: 0000000000000000
[03:14:45]

is this good?

jhansche, 2023-04-01:
Yeah, and it looks like the same problem as reported earlier in the comments: memset() from BedjetCodec::decode_notify.

What doesn't make sense however, is that I'm not manually allocating that memory address - it's allocated automatically. So I don't understand how it can be getting `StoreProhibited` here, if that object was already allocated by the platform.

```
    // Clear old buffer
    memset(&this->buf_, 0, sizeof(BedjetStatusPacket));
```
```
class BedjetCodec {
  ...
  BedjetStatusPacket buf_;
}
```

The `buf_` field is allocated for us.

jhansche, 2023-04-01:
For those of you having this crash, can you try with this branch, and provide the logs of the same stack trace, as well as the `bedjet` logs leading up to the crash?

```
# for debugging https://github.com/esphome/issues/issues/3807
external_components:
 - source: github://jhansche/esphome@test/3807
   components: [ bedjet ]
   refresh: 5sec
```

I've added a new DEBUG level log message before the crashing code. The log I'm looking for will look like:

```
[00:00:00][D][bedjet:000]: Writing ## bytes to @0x####, from @0x####…
```

cc @nochlab1 @kquinsland

nochlab1, 2023-04-09:
> For those of you having this crash, can you try with this branch, and provide the logs of the same stack trace, as well as the `bedjet` logs leading up to the crash?
>
> ```
> # for debugging [...]

uto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x001a5fff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 1661312 bytes to 1098030...
Wrote 1661312 bytes (1098030 compressed) at 0x00010000 in 25.0 seconds (effective 531.2 kbit/s)...
Hash of data verified.
Compressed 17440 bytes to 12127...
Wrote 17440 bytes (12127 compressed) at 0x00001000 in 0.5 seconds (effective 275.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 520.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 737.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[00:39:17][I][logger:258]: Log initialized
[00:39:17][C][ota:469]: There have been 5 suspected unsuccessful boot attempts.
[00:39:17][D][esp32.preferences:113]: Saving 1 preferences to flash...
[00:39:17][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[00:39:17][I][app:029]: Running through setup()...
[00:39:17][C][esp32_ble:026]: Setting up BLE...
[00:39:18][D][esp32_ble:042]: BLE setup complete
[00:39:18][D][esp32_ble_tracker:246]: Starting scan...
[00:39:18][C][wifi:037]: Setting up WiFi...
[00:39:18][C][wifi:038]:   Local MAC: EC:62:60:75:D5:30
[00:39:18][D][wifi:386]: Starting scan...
[00:39:18][D][switch:056]: 'Enable Bedjet': Sending state ON
[00:39:18][D][esp32_ble_client:048]: [0] [94:B5:55:17:E8:CA] Found device
[00:39:18][D][esp32_ble_tracker:214]: Pausing scan to make connection...
[00:39:18][D][esp32_ble_tracker:214]: Pausing scan to make connection...
[00:39:18][I][esp32_ble_client:064]: [0] [94:B5:55:17:E8:CA] 0x00 Attempting BLE connection
[00:39:21][I][esp32_ble_client:196]: [0] [94:B5:55:17:E8:CA] Connected
[00:39:21][I][bedjet:217]: [94:B5:55:17:E8:CA] Discovered service characteristics: 
[00:39:21][I][bedjet:218]:      - Command char: 0x33
[00:39:21][I][bedjet:219]:      - Status char: 0x2a
[00:39:21][I][bedjet:220]:        - config descriptor: 0x2b
[00:39:21][I][bedjet:221]:      - Name char: 0x2d
[00:39:21][D][bedjet:239]: [94:B5:55:17:E8:CA] Services complete: obtained char handles.
[00:39:21][D][esp32_ble_tracker:246]: Starting scan...
[00:39:21][D][bedjet:435]: [94:B5:55:17:E8:CA] wrote notify=true to status config 0x002b, for conn 0
[00:39:21][D][bedjet:096]: Writing 20 bytes to @0xc, from @0x3fff40dc. Clearing 0..35(35) first.
[00:39:21]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[00:39:21]
[00:39:21]Core  1 register dump:
[00:39:21]PC      : 0x400909a0  PS      : 0x00060130  A0      : 0x800d91d8  A1      : 0x3ffcf640  
WARNING Decoded 0x400909a0: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:150
[00:39:21]A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x00000023  A5      : 0x0000000c  
[00:39:21]A6      : 0x00000014  A7      : 0x00000002  A8      : 0x800e7c91  A9      : 0x3ffcf5c0  
[00:39:21]A10     : 0x00000005  A11     : 0x3f400f31  A12     : 0x00000060  A13     : 0x3f400f38  
[00:39:21]A14     : 0x3ffcf744  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
[00:39:21]EXCVADDR: 0x0000000c  LBEG    : 0x400909a0  LEND    : 0x400909ab  LCOUNT  : 0x00000001  
[00:39:21]
[00:39:21]
[00:39:21]Backtrace:0x4009099d:0x3ffcf6400x400d91d5:0x3ffcf650 0x400da588:0x3ffcf680 0x400dc07a:0x3ffcf6d0 0x40205e89:0x3ffcf700 0x400dfa61:0x3ffcf720 0x40205e51:0x3ffcf740 0x400de711:0x3ffcf760 0x40206235:0x3ffcf780 0x402062f5:0x3ffcf7a0 0x400e718a:0x3ffcf7c0 0x400e935e:0x3ffcf7f0 0x400f4762:0x3ffcf8d0 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4009099d: memset at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:143
WARNING Decoded 0x400d91d5: esphome::bedjet::BedjetCodec::decode_notify(unsigned char const*, unsigned short) at /config/esphome/.esphome/build/bedjet/src/esphome/components/bedjet/bedjet_codec.cpp:99
WARNING Decoded 0x400da588: esphome::bedjet::BedJetHub::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /config/esphome/.esphome/build/bedjet/src/esphome/components/bedjet/bedjet_hub.cpp:378
[... 5 lines cut ...]
WARNING Decoded 0x40206235: esphome::Component::call_loop() at /config/esphome/.esphome/build/bedjet/src/esphome/core/component.cpp:76
WARNING Decoded 0x402062f5: esphome::Component::call() at /config/esphome/.esphome/build/bedjet/src/esphome/core/component.cpp:98
WARNING Decoded 0x400e718a: esphome::Application::setup() at /config/esphome/.esphome/build/bedjet/src/esphome/core/application.cpp:52 (discriminator 2)
WARNING Decoded 0x400e935e: setup() at /config/esphome/.esphome/build/bedjet/src/main.cpp:290
WARNING Decoded 0x400f4762: loopTask(void*) at /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
[00:39:22]
[00:39:22]
[00:39:22]
[00:39:22]
[00:39:22]ELF file SHA256: 0000000000000000
[00:39:22]
[00:39:22]Rebooting...
[00:39:22]ets Jul 29 2019 12:21:46
[00:39:22]
[00:39:22]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[00:39:22]configsip: 0, SPIWP:0xee
[00:39:22]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[00:39:22]mode:DIO, clock div:2
[00:39:22]load:0x3fff0030,len:1184
[00:39:22]load:0x40078000,len:13132
[00:39:22]load:0x40080400,len:3036
[00:39:22]entry 0x400805e4
[00:39:22][I][logger:258]: Log initialized

(after crashing 2 times it connected)

nochlab1, 2023-04-09:
i reflashed and now it works

jhansche, 2023-04-09:
> i reflashed and now it works

Reflashed from my branch, and it stopped crashing? 

The 0xC memory address does not look normal to me, and almost certainly highlights the root of this bug. I expect it to be more similar to the other address 0x3fff40dc.

nochlab1, 2023-04-11:
> @nochlab1 that is helpful - I have a few more log statements to verify my suspicions, if you wouldn't mind rebuilding and redeploying from the same branch, and report back what logs you get.
> 
> I suspect that this is a race condition between the component setup order: BLEClient is setup using AFTER_BLUETOOTH, while BedJet is setup with AFTER_WIFI. BLEClient gets setup first, and then BedJet later. Normally that's probably not a problem, but if the BedJet is actively broadcasting notification packets (while there's no client listening to those packets), then as soon as BLEClient is initialized, it will start trying to dispatch these gatt events, including the notify packet. But if BedJet hasn't been initialized yet, it will be trying to handle a packet on an incomplete component, so the memory address will be uninitialized (null / 0), and trying to access the status packet buffer on that, results in a memory address of `0xc` which is invalid, hence the crash you're seeing.

it didnt build 
INFO Reading configuration /config/esphome/bedjet.yaml...
INFO Updating https://github.com/jhansche/esphome.git@test/3807
INFO Generating C++ source...
INFO Compiling app...
Processing bedjet (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/bedjet/.pioenvs/bedjet/src/esphome/components/bedjet/bedjet_codec.cpp.o
In file included from src/esphome/components/bedjet/bedjet_codec.h:4,
                 from src/esphome/components/bedjet/bedjet_codec.cpp:1:
src/esphome/components/bedjet/bedjet_codec.cpp: In member function 'bool esphome::bedjet::BedjetCodec::decode_notify(const uint8_t*, uint16_t)':
src/esphome/components/bedjet/bedjet_codec.cpp:96:36: error: lvalue required as unary '&' operand
              length, &this->buf_, &this, sizeof(BedjetStatusPacket), sizeof(this->buf_));
                                    ^~~~
src/esphome/core/log.h:92:89: note: in definition of macro 'esph_log_d'
   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^~~~~~~~~~~
src/esphome/components/bedjet/bedjet_codec.cpp:95:5: note: in expansion of macro 'ESP_LOGD'
     ESP_LOGD(TAG, "Writing %d bytes to @%p, on @%p. Clearing 0..%d(%d) first.",
     ^~~~~~~~
*** [/data/bedjet/.pioenvs/bedjet/src/esphome/components/bedjet/bedjet_codec.cpp.o] Error 1
========================== [FAILED] Took 6.12 seconds ==========================

(side point esphome version v2022.9.0 works normally)

jhansche, 2023-04-11:
Whoops, my mistake. Should have tested it first 😳  Try now.

kquinsland, 2023-04-11:
> I suspect that this is a race condition between the component setup order: BLEClient is setup using AFTER_BLUETOOTH, while BedJet is setup with AFTER_WIFI. BLEClient gets setup first, and then [...]

Thinking through it, that would certainly explain it. I'll try to build/flash my bedjet ESP32 today w/ your branch and get back to you.  Do you want me to first try "stock" `2023.03.02` but with some custom `setup_priority` values on the two components to serve as a control of this theory, @jhansche ?

jhansche, 2023-04-11:
And if this is what's causing it, then the actual problem leading to the crash is that the BedJet unit is still broadcasting notify packets while there's no client associated, so when BLEClient initializes, it's immediately getting packets that the BedJet component hasn't requested yet. If the unit were reset or power cycled before esphome starts up, then that situation wouldn't happen, and it won't crash. This is likely why I haven't observed it myself.

We might be able to emulate that bad state by power cycling the esp32 while it's connected to the BedJet unit. In that case the BedJet doesn't know that the client has disappeared, so when esphome starts back up, it'll still be broadcasting the BLE notify events.

kquinsland, 2023-04-11:
As promised, here's my results:

Prep:

```shell
# Confirm on stock 2023.03.02 release
pip install esphome --upgrade
# Confirm build_root cache is empty
esphome clean projects/bedjet.yaml
```

Below are (lightly) edited outputs. The first I had logger set to `INFO`, second was after I switched to `DEBUG`.

```log
Linking .pioenvs/bedjet-controller/firmware.elf
RAM:   [==        ]  17.8% (used 58272 bytes from 327680 bytes)
Flash: [=======   ]  74.7% (used 1370841 bytes from 1835008 bytes)
<...>
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:56:24]\xa5\xd9\xe92
[15:56:24]load:0x3fff0030,len:6660
[15:56:24]load:0x40078000,len:14944
[15:56:24]ho 0 tail 12 room 4
[15:56:24]load:0x40080400,len:3816
[15:56:24]entry 0x40080698
[15:56:24]I (28) boot: ESP-IDF 4.4.4 2nd stage bootloader
[15:56:24]I (28) boot: compile time 15:55:50
[15:56:24]I (28) boot: chip revision: v1.0
[15:56:24]I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
[15:56:24]I (38) boot.esp32: SPI Speed      : 40MHz
[15:56:24]I (43) boot.esp32: SPI Mode       : DIO
[15:56:24]I (47) boot.esp32: SPI Flash Size : 4MB
[15:56:24]I (52) boot: Enabling RNG early entropy source...
[15:56:24]I (57) boot: Partition Table:
[15:56:24]I (61) boot: ## Label            Usage          Type ST Offset   Length
[15:56:24]I (68) boot:  0 otadata          OTA data         01 00 00009000 00002000
[15:56:24]I (76) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[15:56:24]I (83) boot:  2 app0             OTA app          00 10 00010000 001c0000
[15:56:24]I (91) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[15:56:24]I (98) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[15:56:24]I (106) boot: End of partition table
[15:56:24]I (110) boot: No factory image, trying OTA 0
[15:56:24]I (115) boot_comm: chip revision: 1, min. application chip revision: 0
[15:56:24]I (122) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3453ch (214332) map
[15:56:24]I (208) esp_image: segment 1: paddr=00044564 vaddr=3ffbdb60 size=053f8h ( 21496) load
[15:56:24]I (217) esp_image: segment 2: paddr=00049964 vaddr=40080000 size=066b4h ( 26292) load
[15:56:24]I (228) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=f93e4h (1020900) map
[15:56:24]I (598) esp_image: segment 4: paddr=0014940c vaddr=400866b4 size=16d18h ( 93464) load
[15:56:24]I (650) boot: Loaded app from partition at offset 0x10000
[15:56:24]I (686) boot: Set actual ota_seq=1 in otadata[0]
[15:56:24]I (687) boot: Disabling RNG early entropy source...
[15:56:24]I (697) cpu_start: Pro cpu up.
[15:56:24]I (698) cpu_start: Starting app cpu, entry point is 0x40082308
[15:56:24]I (0) cpu_start: App cpu up.
[15:56:25]I (714) cpu_start: Pro cpu start user code
[15:56:25]I (714) cpu_start: cpu freq: 160000000
[15:56:25]I (714) cpu_start: Application information:
[15:56:25]I (718) cpu_start: Project name:     bedjet-controller
[15:56:25]I (724) cpu_start: App version:      2023.3.2
[15:56:25]I (729) cpu_start: Compile time:     Apr 11 2023 15:55:37
[15:56:25]I (735) cpu_start: ELF file SHA256:  93154bdb1ea08ab7...
[15:56:25]I (741) cpu_start: ESP-IDF:          4.4.4
[15:56:25]I (746) heap_init: Initializing. RAM available for dynamic allocation:
[15:56:25]I (753) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
[15:56:25]I (759) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
[15:56:25]I (765) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
[15:56:25]I (772) heap_init: At 3FFCBF00 len 00014100 (80 KiB): DRAM
[15:56:25]I (778) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
[15:56:25]I (784) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
[... 1406 lines cut ...]
[15:59:51][D][climate:392]: 'BedJet Climate' - Sending state:
[15:59:51][D][climate:395]:   Mode: OFF
[15:59:51][D][climate:397]:   Action: IDLE
[15:59:51][D][climate:400]:   Fan Mode: OFF
[15:59:51][D][climate:403]:   Custom Fan Mode: 20%
[15:59:51][D][climate:415]:   Current Temperature: 22.00°C
[15:59:51][D][climate:421]:   Target Temperature: 32.00°C
[15:59:55][D][bedjet:379]: codec = 0x3ffd255c
[15:59:55][D][bedjet:096]: Writing 20 bytes to @0x3ffeb888, on @0x3ffeb87c. Clearing 0..35(35) first.

<...>
# Pattern like this repeats;
[16:11:37][D][bedjet:379]: codec = 0x3ffd255c
[16:11:37][D][bedjet:096]: Writing 20 bytes to @0x3ffeb888, on @0x3ffeb87c. Clearing 0..35(35) first.
[16:11:42][D][bedjet:098]: [BedJet Fan] update()
[16:11:42][D][bedjet:102]: [BedJet Fan] update_status result=true
<...>
[16:11:52][D][bedjet:379]: codec = 0x3ffd255c
[16:11:52][D][bedjet:096]: Writing 20 bytes to @0x3ffeb888, on @0x3ffeb87c. Clearing 0..35(35) first.
[16:12:06][D][bedjet:512]: [BEDJET-VICE] Notifying 2 children of latest status @0x3ffeb888.
```

----

It looks like your hunch was correct; the only crashes are at the failed `assert()` checks for null pointers you added: https://github.com/esphome/esphome/compare/dev...jhansche:esphome:test/3807

jhansche, 2023-04-12:
I tested the PR on my own BedJet and it appears to work fine. And it's already approved, so will likely be merged soon for the next release.

Thanks to everyone here for all the logs and help tracking this down.

Iceman248, 2023-05-17:
Reading through this. Can ESP32 not work properly with a BedJet while also being a Bluetooth Proxy?

jhansche, 2023-05-17:
This is explained in the BLE proxy docs.

It should work, but I have not tested it explicitly. It largely depends on whether the proxied device needs a connection or not. There's a limit on the number of devices that esp32 can connect to at one time. But if it's just passive BLE scanning, there's no limit.

I am able to connect to two separate BedJet units with a single esp32, so I know that it can connect to multiple devices. 

But I don't see how you arrived at your conclusion by reading through this. This issue has nothing to do with BLE proxy. If you're having a specific problem then I suggest you open a new issue with relevant information for your issue.

Iceman248, 2023-05-17:
Will do, thanks.