esphome/issues #3809: Button (GPIO Binary Sensor) keeps going on / off since 2022.11.0

codyc1515, 2022-11-18:
### The problem

Since updating to 2022.11.0, my (inverted) button keeps coming on then off repeatedly. As far as I can tell this wasn’t happening in 2022.10.0. Using M5Stack Atom Lite and it’s happening on all of my devices.

### Which version of ESPHome has the issue?

2022.11.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.11.0

### What platform are you using?

ESP32

### Board

M5Stack Atom Lite

### Component causing the issue

binary_sensor, gpio

### Additional information

[image]
[image]

SkipperDo, 2022-11-25:
Hello...

Maybe a pull-up / pull-down resistor solves this issue?
(I don't know how ESPHome solves inputs... Quickly i can't find anything about this... - Maybe they don't setup software-pull-ups / pull-downs in the soucrecode...)

Try it... Good luck...

Kind Regards..

codyc1515, 2022-11-25:
This only became an issue in 2022.11, otherwise I’d agree.

SkipperDo, 2022-11-25:
See my edit... But note the code structure is modified by this side...

So better take a look here: https://esphome.io/components/binary_sensor/gpio.html
and combine the "inverting values" and the "activating internal pullups" section...

Software changes... Maybe before they automatically activate internal pull-ups before... And disable this "feature" in a following version...

codyc1515, 2022-11-27:
So, I tried to use the pullups and what do you know...

`GPIO39 (34-39) does not support pullups.`

Any more thoughts on this one? Bit strange for it to just pop up like this or perhaps I simply just didn't notice before.

SkipperDo, 2022-11-28:
Change the Pin if possible... Or add an external Pullup...

codyc1515, 2024-01-09:
Worker around this by using the debounce parameters but the root cause clearly remains.