esphome/issues #3810: Sprinkler with one valve cannot have main_switch - unable to trigger extended behaviours

schinckel, 2022-11-19:
### The problem

I have a multi-zone sprinkler that has a distinct valve designated for "recreation" purposes. The main system is responsible for watering the lawn, but there is a misting valve that the kids use on the trampoline.

To make it easier, I moved the misting valve into it's own sprinkler item - this means that it should be possible to trigger the main lawn program to pause (if it's running) when the mister is turned on, and then resume when it's turned off.

However, because the mist sprinkler only has a single valve, then I'm unable to define a main valve, and as such unable to build this automation.

I'm aware I could have the mister as a disabled valve on the main system, but that seems like a harder-to manage solution.

### Which version of ESPHome has the issue?

2022.11.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP8266

### Board

Wemos D1

### Component causing the issue

sprinkler

### Example YAML snippet

```yaml
- id: lawn
      # ... these parts aren't important ...
    - id: trampoline_mist
      main_switch:
        name: "Trampoline Mister"
        on_turn_on:
          - sprinkler.pause: lawn
        on_turn_off:
          - sprinkler.resume: lawn
      valves:
        - valve_switch: "Trampoline Mist"
          pump_switch_id: relay_0
          valve_switch_id: relay_8
          run_duration: 10min
```

### Anything in the logs that might be useful for us?

```txt
Failed config

sprinkler: [source irrigation.yaml:207]
  
  Do not define main_switch with only one valve.

Attempting to add on_turn_o* to a valve_switch fails with a different error.
```

kbx81, 2022-11-20:
Since this single-valve controller has only one valve switch, why not just move the `on_turn_o*` action to the valve switch? With a single valve, I can't really think of a reason to not do this.

schinckel, 2022-11-20:
Ah, you mean to the GPIO switch that is being controlled by the sprinkler valve?

Yeah, I didn't think of that. That seems to work (it compiles, at least).

closed 2022-11-20 by schinckel