esphome/issues #3814: Incorrect readout for negative values for PMS5003T

spyfly, 2022-11-19:
### The problem

When the temperature dips below 0.0 °C the readout inverts and goes to 6553,5 °C. This equals the value 65535 without rounding.

When investigating the [source code](https://github.com/esphome/esphome/blob/d8746266625c5a4c9127577297876e445cffe859/esphome/components/pmsx003/pmsx003.cpp#L198), it can be seen that an unsigned 16-bit integer is used, which is *not* compatible with getting negative values.

So the solution would be reading out the temperature as a signed integer.

### Which version of ESPHome has the issue?

2022.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.3

### What platform are you using?

ESP32

### Board

az-delivery-devkit-v4

### Component causing the issue

pmsx003

### Additional information

I have attached an example temperature readout below: 
[image]

This issue has also been previously reported here:  #3013

spyfly, 2022-11-19:
This is the fix I came up with for now:
```
    temperature:
      name: "Wetterstation Strassenseite Temperature"
      id: "strassenseite_temp"
      filters: 
        - lambda: !lambda |-
            // Fix negative values for now
            if (x > 6000) {
              return x - 6553.6;
            }
            return x;
```

aderusha, 2023-12-19:
Issue still out there, this should be a one-line fix: https://github.com/esphome/esphome/blob/dev/esphome/components/pmsx003/pmsx003.cpp#L198

KarolKozlowski, 2024-01-11:
> This is the fix I came up with for now:
> 
> ```
>     temperature:
>       name: "PMS5003T Temp"
>       id: "PMS5003T_temp"
>       filters: 
>         - lambda: !lambda |-
>             // Fix negative values for now
>             if (x > 6000) {
>               return x - 6553.6;
>             }
>             return x;
> ```

Casting to correct type seems like a better solution:
```
      filters:
        - lambda: !lambda |-
            return (int16_t) x;
```

ssieb, 2024-01-12:
You could have made a PR.  :-)   Please test it.

closed 2024-01-12 by kbx81

KarolKozlowski, 2024-01-12:
Indeed I could, but I wasn't sure how to test it, but since you called me to the board I guess I have no choice now... ;)

I've added the following stanza to my project and removed the lambdas, it this the right way to use the updated code?

```
external_components:
  - source: github://esphome/esphome/@dev
    components:
      - pmsx003
```

The data seems be coming in correctly, however I will have to wait for the temperatures to drop to be sure the issue is solved.

ssieb, 2024-01-12:
Yes, since it's merged to dev now, that's the way to do it.

jfurtner, 2024-01-15:
Can confirm this fixes the temperature readings from PMS5003Ts (2 in an Airgradient Pro). Currently reading -6.1 instead of the 6547.5 it was before.