esphome/issues #3815: Tuya climate does not support visual: tag attributes

jazzmonger, 2022-11-19:
### The problem

Trying to use visual tag with Tuya climate and it renders the climate control unusable. 

[image]

### Which version of ESPHome has the issue?

All

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

All

### What platform are you using?

ESP32

### Board

TYWE1S

### Component causing the issue

Climate

### Example YAML snippet

```yaml
#
climate:
  - platform: tuya
    reports_fahrenheit: true
    name: "House Wood Pellet Stove"
    switch_datapoint: 1
    target_temperature_datapoint: 106
    current_temperature_datapoint: 107
#   visual:
#     temperature_step: 1 °F
#     min_temperature: 50 °F
#     max_temperature: 75 °F
```

### Anything in the logs that might be useful for us?

```txt
No log issues just Lovelace display issues.
Adjusting any climate entity with the round control and decimal points is tedious at best. Climate should always default to round numbers.
```

ssieb, 2022-11-20:
Can you explain what the issue is?  The HA interface is the same for all climate components.  So if it's a problem with the HA interface, then it's an HA issue, not esphome.

jazzmonger, 2022-11-20:
[image]

the climate on the left is Tuya Climate from ESP home. Notice the decimal point in the set temperature. The climate entity on the right is a standard home assistant climate entity, not Tuya from ESP home. Noticed there is no decimal place in the single digits only.

why are they different?

ssieb, 2022-11-20:
How is it unusable to have a `.0` on the set temp?  What is your actual config?  Is that how it looks with the `visual:` part commented out or do you have something else?

ssieb, 2022-11-20:
Also, post logs of the tuya climate config from the device.

jazzmonger, 2022-11-20:
trying to set 68.0 or 70.0 is next to impossible.  Try it.
Why doesn't the option to set  temperature_step work? 
```
climate:
  - platform: tuya
    reports_fahrenheit: true
    name: "House Wood Pellet Stove"
    switch_datapoint: 1
    target_temperature_datapoint: 106
    current_temperature_datapoint: 107
#   visual:
#     temperature_step: 1 °F
#     min_temperature: 50 °F
#     max_temperature: 75 °F
```

Logs:
```
INFO Reading configuration /config/esphome/house-pellet-stove.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/4032/head
INFO Detected timezone 'America/Denver'
INFO Starting log output from 192.168.1.80 using esphome API
INFO Successfully connected to 192.168.1.80
[22:36:00][I][app:102]: ESPHome version 2022.12.0-dev compiled on Nov 19 2022, 09:49:49
[22:36:00][C][wifi:507]: WiFi:
[22:36:00][C][wifi:365]:   Local MAC: CC:50:E3:FE:FE:2D
[22:36:00][C][wifi:366]:   SSID: 'Van Slyke'[redacted]
[22:36:00][C][wifi:367]:   IP Address: 192.168.1.80
[22:36:00][C][wifi:368]:   BSSID: E0:63:DA:15:93:89[redacted]
[22:36:00][C][wifi:370]:   Hostname: 'house-wood-pellet-stove'
[22:36:00][C][wifi:372]:   Signal strength: -56 dB ▂▄▆█
[22:36:00][C][wifi:376]:   Channel: 6
[22:36:00][C][wifi:377]:   Subnet: 255.255.255.0
[22:36:00][C][wifi:378]:   Gateway: 192.168.1.1
[22:36:00][C][wifi:379]:   DNS1: 8.8.8.8
[22:36:00][C][wifi:380]:   DNS2: 0.0.0.0
[22:36:00][C][logger:293]: Logger:
[22:36:00][C][logger:294]:   Level: DEBUG
[22:36:00][C][logger:295]:   Log Baud Rate: 0
[22:36:00][C][logger:296]:   Hardware UART: UART0
[22:36:00][C][uart.arduino_esp8266:102]: UART Bus:
[22:36:00][C][uart.arduino_esp8266:103]:   TX Pin: GPIO15
[22:36:00][C][uart.arduino_esp8266:104]:   RX Pin: GPIO13
[22:36:00][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[22:36:00][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[22:36:00][C][uart.arduino_esp8266:109]:   Data Bits: 8
[22:36:00][C][uart.arduino_esp8266:110]:   Parity: NONE
[22:36:00][C][uart.arduino_esp8266:111]:   Stop bits: 1
[22:36:00][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[22:36:00][C][uptime.sensor:031]: Uptime Sensor 'sensor_uptime'
[22:36:00][C][uptime.sensor:031]:   Device Class: 'duration'
[22:36:00][C][uptime.sensor:031]:   State Class: 'total_increasing'
[22:36:00][C][uptime.sensor:031]:   Unit of Measurement: 's'
[22:36:00][C][uptime.sensor:031]:   Accuracy Decimals: 0
[22:36:00][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[22:36:01][C][template.sensor:023]: Template Sensor 'House Wood Pellet Stove Uptime'
[22:36:01][C][template.sensor:023]:   Device Class: 'timestamp'
[22:36:01][C][template.sensor:023]:   State Class: ''
[22:36:01][C][template.sensor:023]:   Unit of Measurement: ''
[22:36:01][C][template.sensor:023]:   Accuracy Decimals: 0
[22:36:01][C][template.sensor:024]:   Update Interval: never
[22:36:01][C][restart.button:017]: Restart Button 'ESPH House Stove Restart'
[22:36:01][C][tuya.select:042]: Tuya Select 'ESPH House Power Select'
[22:36:01][C][tuya.select:043]:   Select has datapoint ID 4
[22:36:01][C][tuya.select:044]:   Options are:
[22:36:01][C][tuya.select:047]:     0: P1
[22:36:01][C][tuya.select:047]:     1: P2
[22:36:01][C][tuya.select:047]:     2: P3
[22:36:01][C][tuya.select:047]:     3: P4
[22:36:01][C][tuya.select:042]: Tuya Select 'ESPH House ECO Select'
[... 89 lines cut ...]
[22:36:29][D][select:015]: 'ESPH House Power Select': Sending state P2 (index 1)
[22:36:29][D][sensor:126]: 'ESPH House Stove-Power Mode': Sending state 1.00000  with 0 decimals of accuracy
[22:36:29][D][tuya:300]: Datapoint 101 update to 0
[22:36:29][D][select:015]: 'ESPH House ECO Select': Sending state ECO1 (index 0)
[22:36:29][D][tuya:288]: Datapoint 106 update to 68
[22:36:29][D][sensor:126]: 'ESPH House Stove-Set Temp': Sending state 68.00000  with 0 decimals of accuracy
[22:36:29][D][climate:385]: 'House Wood Pellet Stove' - Sending state:
[22:36:29][D][climate:388]:   Mode: OFF
[22:36:29][D][climate:390]:   Action: OFF
[22:36:29][D][climate:408]:   Current Temperature: 20.00°C
[22:36:29][D][climate:414]:   Target Temperature: 20.00°C
[22:36:29][D][tuya:288]: Datapoint 107 update to 68
[22:36:29][D][sensor:126]: 'ESPH House Stove-Current Temp': Sending state 68.00000  with 0 decimals of accuracy
[22:36:29][D][climate:385]: 'House Wood Pellet Stove' - Sending state:
[22:36:29][D][climate:388]:   Mode: OFF
[22:36:29][D][climate:390]:   Action: OFF
[22:36:29][D][climate:408]:   Current Temperature: 20.00°C
[22:36:29][D][climate:414]:   Target Temperature: 20.00°C
[22:36:29][D][tuya:288]: Datapoint 108 update to 138
[22:36:29][D][sensor:126]: 'ESPH House Stove-Pipe Temp': Sending state 138.00000  with 0 decimals of accuracy
[22:36:29][D][tuya:288]: Datapoint 109 update to 91
[22:36:29][D][sensor:126]: 'ESPH House Stove-Protect Temp': Sending state 91.00000  with 0 decimals of accuracy
[22:36:29][D][tuya:317]: Datapoint 104 update to 00000000
[22:36:29][D][sensor:126]: 'ESPH House Stove-Error Code': Sending state 0.00000  with 0 decimals of accuracy
````

ssieb, 2022-11-20:
> Why doesn't the option to set temperature_step work?

Because you have it commented out?

jazzmonger, 2022-11-20:
I know.  Perhaps you didn't understand - this is a PR.

Its commented out bc it causes the lovelace control to completely stop working. If its uncommented, Temp can only be adjusted 0.8 degrees higher. I don't think this is the intended operation/feature....

jazzmonger, 2022-11-20:
is my syntax wrong or am I trying to do this in the wrong place or, is there actually a bug here?

randybb, 2022-11-20:
temperature_step (Optional, **float**): The granularity with which the target temperature can be controlled.
Remove °F units. Just tested it - if value is float or in °C, then it is fine. It is broken with °F.

jazzmonger, 2022-11-21:
Well... I removed degrees F... if I lived in hell it would look great.  Here on earth it doesn't look so good.  Better to actually FIX the bug,

[image]

closed 2022-11-24 by jesserockz