esphome/issues #3816: Sonoff Basic Devices Offline After Upgrading Firmware to 2022.11.1

grantalewis, 2022-11-19:
### The problem

I have 14 different devices running ESPHome, most of them Sonoff Basics. After recently updating from 2022.10.2 to 2022.11.1, I started the OTA update process but then noticed that a couple devices didn't come back online after updating, so I stopped the upgrade and then rolled back to 2022.10.2. 

The yaml for one of the offline devices is below. Although this same yaml has worked for literally years, something's evidently disagreeing with the new firmware. I noticed the breaking change, "Don’t Use Base Network Manual IP for WiFi AP" [esphome#3902](https://github.com/esphome/esphome/pull/3902), but that doesn't seem like it would cause this kind of a problem.

Any guidance greatly appreciated.

### Which version of ESPHome has the issue?

2022.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.10.5

### What platform are you using?

ESP8266

### Board

Sonoff Basic

### Component causing the issue

unknown

### Example YAML snippet

```yaml
esphome:
  name: attic-sonoff

esp8266:
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.48
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    gateway: 192.168.1.1
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Attic-Sonoff Fallback Hotspot"
    password: [redacted]

captive_portal:

api:

logger:

ota:
  password: [redacted]

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: Attic Sonoff ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: Attic Sonoff IP
    ssid:
      name: Attic Sonoff SSID
    bssid:
      name: Attic Sonoff BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Attic Sonoff Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: Attic Sonoff WiFi Signal
    update_interval: 60s

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: Attic Sonoff Button
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: Attic Sonoff Relay
    pin: GPIO12
    id: relay

status_led:
  pin:
    number: GPIO13
    inverted: yes
```

### Anything in the logs that might be useful for us?

```txt
Nothing of note.
```

ssieb, 2022-11-20:
The AP isn't coming up?  Could you possibly get serial logs?

grantalewis, 2022-11-20:
I'm able to connect to the AP, but barely. Connectivity is very flaky. My laptop is merely inches away from the Sonoff, so it's not a proximity thing. The web interface appears long enough for me to select the correct WiFi network and supply the network key, but then it never does anything after that. I just continue to get a flashing green light on the Sonoff.

I'd need to be connected to get serial logs, right?

I guess the next step is to disassemble and re-flash, but if my yaml is at fault, my guess is that I'm likely to run into the same problem again in the future. Do you see anything wrong with the yaml?

bkaufx, 2022-11-20:
Try reducing Wi-Fi power to 17.  ESPHome's default of 20 causes lots of weird problems like this.

https://esphome.io/components/wifi.html?highlight=output_power#configuration-variables

e.g.: https://github.com/esphome/issues/issues/3747#event-7715512349

zerocritical, 2022-11-21:
I have the same issue. Devices appear online on 2022.10.2 but offline on 2022.11.1 (only in ESP home). controlling the device via integration works fine.

I can ping the devices using devicename and I get a response. Some issue with ESP Home trying to resolve dns name I guess

nnmalex, 2022-11-28:
I have this issue as well but it only happens if I have the `time` component enabled. Whether sntp or ha in config would cause this.

grantalewis, 2022-11-29:
It appeared that no resolution was coming, so I flashed the failed devices with Tasmota, and I guess I'll avoid upgrading any remaining devices with updates to ESPHome. I guess I'll eventually move to Tasmota entirely.

closed 2022-11-29 by grantalewis

Mechotronic, 2022-12-09:
I have always had this issue with SonOff Minis and SonOff basics./ The AP is usually not available in the list, and randomly shows up for me to connect to it. It fails 90% of the time. I'll try 17, but this means I have to flip breakers to recover it.