esphome/issues #3818: PID Thermostat incorrect value transfer

CCChelios, 2022-11-20:
### The problem

My goal: for the thermostat (air temperature) to set the temperature of the thermostat (heat carrier temperature), I pass the temperature value through the output template, but instead of the value 0-85, I get 0 or 1 on the thermostat.

### Which version of ESPHome has the issue?

 2022.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

PID Climate

### Example YAML snippet

```yaml
output:
  - platform: template
    id: target_temperature_ten
    type: float
    write_action:
      - climate.control:
          id: thermostat_on_heater_1
          mode: "HEAT"
          target_temperature: !lambda return state * 85;

climate:
  - platform: thermostat # Thermostat TEN #1
    name: "Thermostat on heater #1"
    id: thermostat_on_heater_1
    sensor: coolant_temp
    min_heating_off_time: 0s
    min_heating_run_time: 0s
    min_idle_time: 0s
    heat_deadband: 5
    heat_overrun: 0
    heat_action:
      - switch.turn_on: ten1
    idle_action:
      - switch.turn_off: ten1
    visual:
      min_temperature: 0 °C
      max_temperature: 85 °C
    preset:
      - name: Home
        default_target_temperature_low: 85 °C

  - platform: pid
    name: "Air thermostat"
    id: air_thermostat
    sensor: air_temp
    default_target_temperature: 21°C
    heat_output: target_temperature_ten
    control_parameters:
      kp: 0.49460
      ki: 0.00487
      kd: 12.56301
    visual:
      min_temperature: 10 °C
      max_temperature: 30 °C
```

gocenik, 2022-11-25:
You need to divide it by 100

```target_temperature: !lambda return state * 85 / 100;```
https://esphome.io/components/output/index.html#config-output

CCChelios, 2022-11-25:
In this line state * 85 I get: 1*84 = 84
your suggestion will: 1*84/100 = 0.84
the state value should have more dynamic range.  while it takes 0 or 1

gocenik, 2022-11-25:
You are correct, sorry. I was doing the same thing with two PID thermostats... Anyway, try this, I think it will work:

```
output:
  - platform: template
    id: target_temperature_ten
    type: float
    write_action:
      - lambda: |-
          auto call = id(thermostat_on_heater_1).make_call();
          call.set_target_temperature(id(target_temperature_ten).state * 85);
          call.perform();
```

CCChelios, 2022-11-26:
didn't know about this design. but it still sends 1 or 0(

gocenik, 2022-11-27:
That should have worked, I've copied from my config. There is another approach, how about this:

```
output:
  - platform: template
    id: target_temperature_ten
    name: 'Thermostat TEN #1'
    type: float
    write_action:
      - then:
          - lambda: |-
              auto call = id(thermostat_on_heater_1).make_call();
              call.set_target_temperature(id(pid_output).state / 85);
              call.perform();
          - sensor.template.publish:
              id: output_pid
              state: !lambda return state;

sensor:
  - platform: pid
    id: pid_output
    name: PID Output
    type: HEAT

  - platform: template
    name: Output Result
    id: output_pid
 ```
 
What output do you get for the two sensors I've added in the example?

CCChelios, 2022-11-27:
Thanks a lot!  As it turned out, everything is correct, it transmits a value from 0.0 to 1.0, due to the fact that the pid controller is not configured, it showed only extreme values.

closed 2022-11-27 by CCChelios