esphome/issues #3819: GL-S10 reboot cycle after device configured in Home Assistant

c4weddell, 2022-11-20:
### The problem

I have flashed a new GL-S10 (v2.1) as per https://esphome.github.io/bluetooth-proxies/ and the log shows it boots up fine.  However once it has been Configured in Home Assistant as a discovered new device it connects successfully but then aborts and keeps going through a reboot cycle (until the device is deleted from Home Assistant).

[D][api:102]: Accepted ::FFFF:C0A8:1BF
[D][api.connection:917]: Home Assistant 2022.11.3 (::FFFF:C0A8:1BF): Connected successfully
abort() was called at PC 0x401d4317 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x400901dc:0x3ffccc80 0x40090459:0x3ffccca0 0x401d4317:0x3ffcccc0 0x401d435e:0x3ffccce0 0x401d3c79:0x3ffccd00 0x401d3d50:0x3ffccd20 0x400d456e:0x3ffccd40 0x400d45e3:0x3ffccd60 0x400d7a37:0x3ffccd80 0x400d7ae5:0x3ffcce70 0x400dab72:0x3ffcce90 0x401dd785:0x3ffccf90 0x401dd829:0x3ffccfb0 0x400ddcfa:0x3ffccfd0 0x400e020e:0x3ffcd000 0x400e4c08:0x3ffcd020 0x400914ce:0x3ffcd040

### Which version of ESPHome has the issue?

2022.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.3

### What platform are you using?

ESP32

### Board

GL-S10 (v2.1)

### Anything in the logs that might be useful for us?

```txt
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:243]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:465]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[I][i2c.arduino:161]: Performing I2C bus recovery
[D][binary_sensor:034]: 'Reset Button': Sending initial state OFF
[D][esp32_ble_tracker:264]: Starting scan...
[C][ethernet:039]: Setting up Ethernet...
[I][ethernet:092]: Starting ethernet connection
[I][ethernet:103]: Connected via Ethernet!
[C][ethernet:264]:   IP Address: 192.168.1.195
[C][ethernet:265]:   Hostname: 'gl-s10-bt-proxy-8f34d4'
[C][ethernet:266]:   Subnet: 255.255.255.0
[C][ethernet:267]:   Gateway: 192.168.1.1
[C][ethernet:278]:   DNS1: 192.168.1.1
[C][ethernet:279]:   DNS2: 0.0.0.0
[C][ethernet:282]:   MAC Address: 0C:B8:15:8F:34:D7
[C][ethernet:283]:   Is Full Duplex: YES
[C][ethernet:284]:   Link Up: YES
[C][ethernet:285]:   Link Speed: 100
[C][ota:089]: Over-The-Air Updates:
[C][ota:090]:   Address: gl-s10-bt-proxy.local:3232
[... 1006 lines cut ...]
[C][logger:277]:   Log Baud Rate: 115200
[C][logger:278]:   Hardware UART: UART0
[C][ethernet:146]: Ethernet:
[C][ethernet:264]:   IP Address: 192.168.1.195
[C][ethernet:265]:   Hostname: 'gl-s10-bt-proxy-8f34d4'
[C][ethernet:266]:   Subnet: 255.255.255.0
[C][ethernet:267]:   Gateway: 192.168.1.1
[C][ethernet:278]:   DNS1: 192.168.1.1
[C][ethernet:279]:   DNS2: 0.0.0.0
[C][ethernet:282]:   MAC Address: 0C:B8:15:8F:34:D7
[C][ethernet:283]:   Is Full Duplex: YES
[C][ethernet:284]:   Link Up: YES
[C][ethernet:285]:   Link Speed: 100
[C][ethernet:148]:   Power Pin: GPIO5
[C][ethernet:149]:   MDC Pin: 23
[C][ethernet:150]:   MDIO Pin: 18
[C][ethernet:151]:   Type: IP101
[C][mdns:100]: mDNS:
[C][mdns:101]:   Hostname: gl-s10-bt-proxy-8f34d4
[C][ota:089]: Over-The-Air Updates:
[C][ota:090]:   Address: gl-s10-bt-proxy.local:3232
[W][ota:099]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts

Terminal disconnected: NetworkError: The device has been lost.
```

c4weddell, 2022-11-22:
This appears to be an ethernet issue.  Reconfiguring to use wifi rather than ethernet and it is working as expected.

efaden, 2022-11-26:
Interesting..... also finding an issue with POE in general.  Had two Olimex units.  POE powered both and neither worked.  If I switch from POE to just ethernet + usb micro power it works fine.  Back to POE and it just reboots over and over......

gl-chenxiaosheng, 2022-11-28:
I think the official may have put the wrong firmware in the website, because I according to the configuration of their config [gl-s10.yaml](https://github.com/esphome/bluetooth-proxies/blob/main/gl-s10.yaml) compiled and flash the firmware is no problem, you can try the firmware provided by my attachment.
[gl-s10-bt-proxy-factory.zip](https://github.com/esphome/issues/files/10103193/gl-s10-bt-proxy-factory.zip)

gl-chenxiaosheng, 2022-11-29:
> > I think the official may have put the wrong firmware in the website, because I according to the configuration of their config [gl-s10.yaml](https://github.com/esphome/bluetooth-proxies/blob/main/gl-s10.yaml) compiled and flash the firmware is no problem, you can try the firmware provided by my attachment. [gl-s10-bt-proxy-factory.zip](https://github.com/esphome/issues/files/10103193/gl-s10-bt-proxy-factory.zip)
> 
> I used the Yaml file...

You can try the firmware that I compiled, my test here is no problem.

nagyrobi, 2022-12-08:
Please try 2022.12(Beta), Ethernet driver has been rewritten.

c4weddell, 2022-12-17:
Updated to Firmware: 2022.12.1 (Dec 17 2022, 11:40:17) and now working as expected, thank you.

closed 2022-12-17 by c4weddell