esphome/issues #3826: `Meditation Error: Core  0 panic'ed (Illegal instruction)` when booting on the ESP32-C3

alistair23, 2022-11-23:
### The problem

When trying to flash a simple ESPhome binary to an esp32-c3-devkitm-1 I see this error message:

```
phy_init: phGuru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.
```

and the device gets stuck in a reboot loop. The full log is included in the additional details.

I have tried manually flashing the binary with the below command and I see the same issue:

```shell
esptool.py --port /dev/ttyUSB0 --chip esp32c3 write_flash --flash_mode dio --flash_size detect --flash_freq 80m  0x0 espc3-water-level.bin
```

I have attached the binary as a zip file: [espc3-water-level.zip](https://github.com/esphome/issues/files/10074382/espc3-water-level.zip)

### Which version of ESPHome has the issue?

2022.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

ESP32-IDF

### Board

esp32-c3-devkitm-1

### Component causing the issue

phy_init

### Example YAML snippet

```yaml
esphome:
  name: espc3-water-level
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:
  password: "***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

[... 433 lines cut ...]
[14:04:31]3fca1370: 0x00000000 0xffffffff 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000
[14:04:31]3fca1390: 0x3fe00000 0x00000000 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb
[14:04:31]3fca13b0: 0x3fd34413 0x00000000 0x3ff00000 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000
[14:04:31]3fca13d0: 0x40140000 0x00000000 0x43500000 0x3fc8de00 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
[14:04:31]3fca13f0: 0x3fca1050 0x3fc95914 0x3fc92094 0x3fc92094 0x3fca13f0 0x3fc9208c 0x00000002 0x3fc9c94c
[14:04:31]3fca1410: 0x3fc9c94c 0x3fca13f0 0x00000000 0x00000017 0x3fc9f9ec 0x69666977 0x2f094800 0xb1f0cd0a
[14:04:31]3fca1430: 0x009d0c12 0x00000000 0x3fca13e0 0x00000017 0x00000001 0x00000000 0x00000000 0x00000000
[14:04:31]3fca1450: 0x3fc9612c 0x3fc96194 0x3fc961fc 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
[14:04:31]3fca1470: 0x00000000 0x4207d234 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[14:04:31]3fca1490: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[14:04:31]3fca14b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[14:04:31]3fca14d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[14:04:31]3fca14f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[14:04:31]3fca1510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[14:04:31]3fca1530: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca0000 0x00000054 0x3fca1548 0x3fca1548
[14:04:31]3fca1550: 0x3fca1548 0x3fca1548 0x00000000 0x3fca1560 0xffffffff 0x3fca1560 0x3fca1560 0x00000001
[14:04:31]3fca1570: 0x3fca1574 0xffffffff 0x3fc9c320 0x3fc9c320 0x00000000 0x00000001 0x00000000 0xfa00ffff
[14:04:31]3fca1590: 0x00000000 0x138b61b1 0x3fca1540 0x000007ec 0x6f6d706f 0x00006564 0x2e617473 0x64697373
[14:04:31]
[14:04:31]
[14:04:31]
[14:04:31]ELF file SHA256: e3ce15286c4002cc
[14:04:31]
[14:04:31]Rebooting...
```

bananu7, 2024-04-24:
I just hit the same issue running ESPHome Web on the old devkit C3.

TLCary, 2024-12-20:
Same: "Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled."

anishp55, 2025-01-08:
i have hit the same error.

anishp55, 2025-01-08:
@alistair23 i got it working, its because its rev 0.2 of the part.
you need these goodies
```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options: 
     CONFIG_ESP32C3_REV_MIN_2: y
     CONFIG_ESP32C3_REV_MIN_FULL: "2"
     CONFIG_ESP_REV_MIN_FULL: "2"
```

ssieb, 2025-01-08:
Thank you for the info.

closed 2025-01-08 by ssieb