esphome/issues #3827: eps32 crashing with abort and backtrace

apaperclip, 2022-11-23:
### The problem

I upgraded one of my three esp32  to 2022.10 from 2021.9.3 to test for stability/regression. All three have essentially the same config. I noticed the ble sensors were going unavailable in ha and tracked it to this esp32 was restarting. I added an uptime sensor and memory sensors and finally caught a restart while it was plugged in and capturing logs. Memory looks fine so I do not it is a leak. I have not noticed any consistent uptime related to the crashes, but I would need more occurrences to rule that out,  The last reported uptime was 121,837s about 45s before the restart.

### Which version of ESPHome has the issue?

2022.10.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

ESP32

### Board

esp32 devkit

### Example YAML snippet

```yaml
substitutions:
  esphome_name: esphome_ble_02

esphome:
  name: esphome_ble_02
  platform: ESP32
  board: mhetesp32devkit

wifi:
  ssid: "asdf"
  password: "asdf"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome Ble 02 Fallback Hotspot"
    password: "asdf"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

debug:

esp32_ble_tracker:
  scan_parameters:
    window: 300ms
    interval: 300ms
    duration: 15s
    active: false
    
binary_sensor:
  - platform: ble_presence
    ibeacon_uuid: 'asdf'
    name: "mailbox"
    device_class: opening
  - platform: ble_presence
    ibeacon_uuid: 'asdf'
[... 256 lines cut ...]
11-23 02:40:18.696abort() was called at PC 0x401e42df on core 0
11-23 02:40:18.696
11-23 02:40:18.696ELF file SHA256: 0000000000000000
11-23 02:40:18.696
11-23 02:40:18.711Backtrace: 0x4008f7dc:0x3ffd8440 0x4008fa59:0x3ffd8460 0x401e42df:0x3ffd8480 0x401e4326:0x3ffd84a0 0x401e3c41:0x3ffd84c0 0x401e3d18:0x3ffd84e0 0x400d90ad:0x3ffd8500 0x401395b1:0x3ffd8520 0x401346d2:0x3ffd8540 0x40090ace:0x3ffd8570
11-23 02:40:18.711
11-23 02:40:18.727Rebooting...
11-23 02:40:18.727ets Jun  8 2016 00:22:57
11-23 02:40:18.727
11-23 02:40:18.727rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11-23 02:40:18.727configsip: 0, SPIWP:0xee
11-23 02:40:18.727clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11-23 02:40:18.727mode:DIO, clock div:2
11-23 02:40:18.727load:0x3fff0018,len:4
11-23 02:40:18.775load:0x3fff001c,len:1044
11-23 02:40:18.775load:0x40078000,len:8896
11-23 02:40:18.775load:0x40080400,len:5828
11-23 02:40:18.775entry 0x400806ac
11-23 02:40:19.375[I][logger:243]: Log initialized
11-23 02:40:19.375[C][ota:465]: There have been 0 suspected unsuccessful boot attempts.
11-23 02:40:19.390[D][esp32.preferences:113]: Saving 1 preferences to flash...
11-23 02:40:19.390[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
11-23 02:40:19.390[I][app:029]: Running through setup()...
11-23 02:40:19.406[C][switch.gpio:011]: Setting up GPIO Switch 'esphome_ble_02 onboard_LED'...
```

apaperclip, 2022-11-24:
Based on guidance from jess, grabbing logs from the device while attached to the build host for decoding the trace. Will attach once I have it

apaperclip, 2022-11-24:
Here's a crash with the decode

[07:38:38][D][sensor:127]: 'Free Heap Memory': Sending state 71700.00000 b with 1 decimals of accuracy
[07:38:39]abort() was called at PC 0x401e43e3 on core 0
[07:38:39]
[07:38:39]ELF file SHA256: 0000000000000000
[07:38:39]
[07:38:39]Backtrace: 0x4008f7dc:0x3ffd83c0 0x4008fa59:0x3ffd83e0 0x401e43e3:0x3ffd8400 0x401e442a:0x3ffd8420 0x401e3d45:0x3ffd8440 0x401e3e1c:0x3ffd8460 0x400d9101:0x3ffd8480 0x401396b5:0x3ffd84a0 0x401347d6:0x3ffd84c0 0x40090ace:0x3ffd84f0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008f7dc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4008fa59: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x401e43e3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401e442a: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401e3d45: __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc:313
WARNING Decoded 0x401e3e1c: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x400d9101: esphome::esp32_ble_tracker::ESP32BLETracker::gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at /config/.esphome/build/esphome_ble_02/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:308
WARNING Decoded 0x401396b5: btc_gap_ble_cb_to_app at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:1070
 (inlined by) btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:914
WARNING Decoded 0x401347d6: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c:163
WARNING Decoded 0x40090ace: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[07:38:40]
[07:38:40]Rebooting...
[07:38:40]ets Jun  8 2016 00:22:57
[07:38:40]
[07:38:40]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[07:38:40]configsip: 0, SPIWP:0xee
[07:38:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[07:38:40]mode:DIO, clock div:2
[07:38:40]load:0x3fff0018,len:4
[07:38:40]load:0x3fff001c,len:1044
[07:38:40]load:0x40078000,len:10124
[07:38:40]load:0x40080400,len:5828
[07:38:40]entry 0x400806a8
[07:38:40][I][logger:258]: Log initialized
[07:38:40][C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[07:38:40][D][esp32.preferences:113]: Saving 1 preferences to flash...
[07:38:40][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[07:38:40][I][app:029]: Running through setup()...
[07:38:40][C][switch.gpio:011]: Setting up GPIO Switch 'esphome_ble_02 onboard_LED'...
[07:38:40][D][switch:017]: 'esphome_ble_02 onboard_LED' Turning OFF.
[07:38:40][D][switch:037]: 'esphome_ble_02 onboard_LED': Sending state OFF
[07:38:40][D][switch:017]: 'esphome_ble_02 onboard_LED' Turning OFF.
[07:38:41][D][esp32_ble_tracker:264]: Starting scan...
[07:38:41][C][wifi:037]: Setting up WiFi...

apaperclip, 2022-11-29:
Snagged another crash - decode here

```[16:33:34][D][esp32_ble_tracker:833]:   Address Type: PUBLIC
[16:33:36][D][xiaomi_ble:328]: Got Xiaomi LYWSD03MMC (A4:C1:38:BF:38:04):
[16:33:36][D][xiaomi_ble:331]:   Temperature: 21.2°C
[16:33:36][D][sensor:127]: 'Temp04 Temperature 02': Sending state 70.16000 °F with 1 decimals of accuracy
[16:33:38]abort() was called at PC 0x401e43e3 on core 0
[16:33:38]
[16:33:38]ELF file SHA256: 0000000000000000
[16:33:38]
[16:33:38]Backtrace: 0x4008f7dc:0x3ffd83c0 0x4008fa59:0x3ffd83e0 0x401e43e3:0x3ffd8400 0x401e442a:0x3ffd8420 0x401e3af3:0x3ffd8440 0x401e3e2a:0x3ffd8460 0x400d9101:0x3ffd8480 0x401396b5:0x3ffd84a0 0x401347d6:0x3ffd84c0 0x40090ace:0x3ffd84f0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008f7dc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4008fa59: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x401e43e3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401e442a: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401e3af3: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
WARNING Decoded 0x401e3e2a: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x400d9101: esphome::esp32_ble_tracker::ESP32BLETracker::gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at /config/.esphome/build/esphome_ble_02/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:308
WARNING Decoded 0x401396b5: btc_gap_ble_cb_to_app at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:1070
 (inlined by) btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:914
WARNING Decoded 0x401347d6: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c:163
WARNING Decoded 0x40090ace: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[16:33:38]
[16:33:38]Rebooting...
[16:33:38]ets Jun  8 2016 00:22:57
[16:33:38]
[16:33:38]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:33:38]configsip: 0, SPIWP:0xee
[16:33:38]clk_drv:0x00,q_drv:0x00,d_drv```

apaperclip, 2022-12-08:
Some feedback I had from bdraco is that this appeared to be memory related.  I moved to the idf platform and went from 60k to 100k free. Will monitor for stability.

apaperclip, 2023-02-01:
things have been rock solid with the idf platform

closed 2023-02-01 by apaperclip

ssieb, 2023-02-01:
The clear indication is this line:
`WARNING Decoded 0x401e3d45: __cxa_allocate_exception at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_alloc.cc:313`
Definitely was an out of memory problem.