esphome/issues #3831: Wemos NodeMCU CH340 not working

TheMangrove, 2022-11-25:
### The problem

I'm trying to flash a Wemos 32M flash NodeMCU V3 340G ESP8266. It appears to flash ok, I get the "success" message, however afterwards there is no serial communication, and the device does not appear connected to WiFi. 

[This](https://universal-solder.ca/product/nodemcu-lua-esp8266-wifi-4mb-32mb-usb-iot-arduino/) is the board I'm using.

I have tried the following:

- Flashing through different methods including both via USB direct from HASSIO device, as well as via the web browser interface.
- Different USB cables
- Different but incidental board
- Different NodeMCU board model - which works flawlessly ([This board](https://universal-solder.ca/product/nodemcu-v3-lua-esp8266-wifi-4mb-usb-internet-of-things/))
- Checking serial output using Arduino IDE (I just get garbage output when ESPHome is flashed). 
- Flashing a simple script using Arduino IDE (Works fine with good serial output and it connects to WiFi)
- Re-flashing after trying IDE in case that fixed something (It didn't)

Thanks for any help you can provide. 

### Which version of ESPHome has the issue?

2022.11.2 and 2022.11.3 (upgraded while troubleshooting)

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

ESP8266

### Board

NodeMCU V3 340G (Wemos)

### Example YAML snippet

```yaml
esphome:
  name: freezer

esp8266:
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Freezer Fallback Hotspot"
    password: "FuOoim0gni2d"

captive_portal:
```

TheMangrove, 2022-11-30:
So the solution is I had to use the NodeMCU flasher with the following settings:

Baud Rate: 460800
Flash Size: 4MByte
**Flash Speed: 26.7MHz**
SPI Mode: DOUT

I suspect the flash speed is the key element. I'm leaving this open since it's not technically solved, just a workaround found but this should be added as an option for NodeMCU boards.

TheMangrove, 2022-11-30:
Link for NodeMCU Flasher:

https://github.com/nodemcu/nodemcu-flasher

HAelnino80, 2023-03-18:
> Link for NodeMCU Flasher:
> 
> https://github.com/nodemcu/nodemcu-flasher

THANKS !!!!

I had the same issue, I was to put in the garbage :)