esphome/issues #3832: ezo_pmp integration compilation issues

notjj, 2022-11-25:
### The problem

The ezo_pmp component requires you to define at least one sensor of each type (binary_sensor, sensor, text_sensor) or else compilation will fail due to members not being found.

[logs_pump-test_upload.txt](https://github.com/esphome/issues/files/10088133/logs_pump-test_upload.txt)

### Which version of ESPHome has the issue?

2022.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

wESP32

### Component causing the issue

ezo_pmp

### Example YAML snippet

```yaml
# Non-compiling config, uncommenting the sensor section will cause it to compile properly.

ezo_pmp:
  id: ezo_pmp1

# sensor:
#   - platform: ezo_pmp
#     pump_voltage:
#       id: pump_voltage
#       name: Pump Voltage

text_sensor:
  - platform: ezo_pmp
    dosing_mode:
      id: dosing_mode
      name: Dosing Mode

binary_sensor:
  - platform: ezo_pmp
    pump_state:
      id: pump_state
      name: Pump State

number:
  - platform: template
    id: volume
    name: "Volume"
    optimistic: true
    min_value: 0
    max_value: 100
    step: 0.01

button:
  - platform: template
    name: "Dose Volume"
    on_press:
    - ezo_pmp.dose_volume:
        id: ezo_pmp1
        volume: !lambda |-
          return id(volume).state;
```

### Anything in the logs that might be useful for us?

```txt
Compiling /data/pump-test/.pioenvs/pump-test/src/esphome/components/i2c/i2c.cpp.o
src/esphome/components/ezo_pmp/ezo_pmp.cpp: In member function 'virtual void esphome::ezo_pmp::EzoPMP::update()':
src/esphome/components/ezo_pmp/ezo_pmp.cpp:55:81: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'max_flow_rate_'
     this->queue_command_(EZO_PMP_COMMAND_READ_MAX_FLOW_RATE, 0, 0, (bool) this->max_flow_rate_);
                                                                                 ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:56:81: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'current_volume_dosed_'
     this->queue_command_(EZO_PMP_COMMAND_READ_SINGLE_REPORT, 0, 0, (bool) this->current_volume_dosed_);
                                                                                 ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:57:86: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'total_volume_dosed_'
     this->queue_command_(EZO_PMP_COMMAND_READ_TOTAL_VOLUME_DOSED, 0, 0, (bool) this->total_volume_dosed_);
                                                                                      ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:59:39: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'absolute_total_volume_dosed_'
                          (bool) this->absolute_total_volume_dosed_);
                                       ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:68:83: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'current_volume_dosed_'
       this->queue_command_(EZO_PMP_COMMAND_READ_SINGLE_REPORT, 0, 0, (bool) this->current_volume_dosed_);
                                                                                   ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:69:88: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'total_volume_dosed_'
       this->queue_command_(EZO_PMP_COMMAND_READ_TOTAL_VOLUME_DOSED, 0, 0, (bool) this->total_volume_dosed_);
                                                                                        ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:71:41: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'absolute_total_volume_dosed_'
                            (bool) this->absolute_total_volume_dosed_);
                                         ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:74:80: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'pump_voltage_'
     this->queue_command_(EZO_PMP_COMMAND_READ_PUMP_VOLTAGE, 0, 0, (bool) this->pump_voltage_);
                                                                                ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp: In member function 'void esphome::ezo_pmp::EzoPMP::read_command_result_()':
src/esphome/components/ezo_pmp/ezo_pmp.cpp:195:56: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'last_volume_requested_'
       if (parsed_second_parameter.has_value() && this->last_volume_requested_) {
                                                        ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:196:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'last_volume_requested_'
         this->last_volume_requested_->publish_state(parsed_second_parameter.value_or(0));
               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:208:62: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'current_volume_dosed_'
       if (parsed_first_parameter.has_value() && (bool) this->current_volume_dosed_) {
                                                              ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:209:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'current_volume_dosed_'
         this->current_volume_dosed_->publish_state(parsed_first_parameter.value_or(0));
               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:214:56: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'max_flow_rate_'
       if (parsed_second_parameter.has_value() && this->max_flow_rate_)
                                                        ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:215:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'max_flow_rate_'
         this->max_flow_rate_->publish_state(parsed_second_parameter.value_or(0));
               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:227:56: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'total_volume_dosed_'
       if (parsed_second_parameter.has_value() && this->total_volume_dosed_)
                                                        ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:228:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'total_volume_dosed_'
         this->total_volume_dosed_->publish_state(parsed_second_parameter.value_or(0));
               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:232:56: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'absolute_total_volume_dosed_'
       if (parsed_second_parameter.has_value() && this->absolute_total_volume_dosed_)
                                                        ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:233:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'absolute_total_volume_dosed_'
         this->absolute_total_volume_dosed_->publish_state(parsed_second_parameter.value_or(0));
               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:251:56: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'pump_voltage_'
       if (parsed_second_parameter.has_value() && this->pump_voltage_)
                                                        ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp:252:15: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'pump_voltage_'
         this->pump_voltage_->publish_state(parsed_second_parameter.value_or(0));
               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp: In member function 'void esphome::ezo_pmp::EzoPMP::dose_continuously()':
src/esphome/components/ezo_pmp/ezo_pmp.cpp:485:79: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'current_volume_dosed_'
   this->queue_command_(EZO_PMP_COMMAND_READ_SINGLE_REPORT, 0, 0, (bool) this->current_volume_dosed_);
                                                                               ^
src/esphome/components/ezo_pmp/ezo_pmp.cpp: In member function 'void esphome::ezo_pmp::EzoPMP::dose_volume(double)':
src/esphome/components/ezo_pmp/ezo_pmp.cpp:491:79: error: 'class esphome::ezo_pmp::EzoPMP' has no member named 'current_volume_dosed_'
   this->queue_command_(EZO
```

stefanh12, 2023-01-24:
I had the same comile issue, it seems like there are default functions missing for the different sensors. If you add them to you yaml it will `comile.

`

esphome:
  name: pool-ezo
  platform: ESP32
  board: featheresp32

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "key"

ota:
  password: "ota"

wifi:
  ssid: myssid
  password: 12345679890

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pool-Ezo Fallback Hotspot"
    password: "password"

captive_portal:
i2c:
  sda: 23
  scl: 22
  scan: true
  id: bus_a

ezo_pmp:
  id: ezo_cl_pmp
  address: 103  # Default Address for the EZO-PMP.
  update_interval: 60s    

sensor:
  - platform: ezo
    name: "pH"
    id: ph_ezo
    address: 99
    unit_of_measurement: "pH"
    update_interval: 120s
    accuracy_decimals: 2
    filters:
    - sliding_window_moving_average:
        window_size: 5
        send_every: 1
    
  - platform: ezo
    name: "orp"
    id: orp_ezo
    address: 98
    unit_of_measurement: "mv"
    accuracy_decimals: 0
    update_interval: 120s
    filters:
    - sliding_window_moving_average:
        window_size: 5
        send_every: 1
        
  - platform: ezo
    name: "temp"
    id: temp_ezo
    address: 102
    unit_of_measurement: "C"    
    accuracy_decimals: 2
    update_interval: 120s

  - platform: ezo_pmp
    total_volume_dosed:
      id: total_volume_dosed
      name: Total Volume Dosed
  - platform: ezo_pmp
    absolute_total_volume_dosed:
      id: absolute_total_volume_dosed
      name: Absolute Total Volume Dosed   
  - platform: ezo_pmp
    last_volume_requested:
      id: last_volume_requested
      name: Last Volume Requested
  - platform: ezo_pmp
    max_flow_rate:
      id: max_flow_rate
      name: Max Flow Rate    
  - platform: ezo_pmp
    pump_voltage:
      id: pump_voltage
      name: Pump Voltage                 
binary_sensor:
  - platform: ezo_pmp
    pump_state:
      id: pump_state
      name: Pump State      
  - platform: ezo_pmp
    is_paused:
      id: is_paused
      name: Is Paused      
text_sensor:
  - platform: ezo_pmp
    dosing_mode:
      id: dosing_mode
      name: Dosing Mode 
  - platform: ezo_pmp
    calibration_status:
      id: calibration_status
      name: Calibration Status           
number:
  - platform: template
    id: volume
    name: "Volume"
    optimistic: true
    min_value: 0
    max_value: 100
    step: 0.01
  - platform: template
    id: duration
    name: "Duration"
    optimistic: true
    min_value: 0
    max_value: 100
    step: 1
button:
  - platform: template
    name: "Dose Over Time"
    on_press:
      - ezo_pmp.dose_volume_over_time:
          id: ezo_cl_pmp
          volume: !lambda |-
                  return id(volume).state;
          duration: !lambda |-
                  return id(duration).state;
   
    
switch:
  - platform: gpio
#    pin: 14
    pin: 12
    name: "pH On"
    inverted: yes  #Makes the ON status turn the pin LOW
    internal: true  #Hides the switch from Home Assistant so as not to clutter the interface
    restore_mode: ALWAYS_ON  #Sets the switch status ON at boot
    
  - platform: gpio
    pin: 15
    name: "Temp On"
    inverted: no  #This pin needs to be HIGH, so not inverted
    internal: true #Hides the switch from Home Assistant so as not to clutter the interface
    restore_mode: ALWAYS_ON  #Sets the switch status ON at boot
    
  - platform: gpio
#    pin: 12
    pin: 27
    name: "ORP On"
    inverted: yes   #Makes the ON status turn the pin LOW
    internal: true #Hides the switch from Home Assistant so as not to clutter the interface
    restore_mode: ALWAYS_ON  #Sets the switch status ON at boot
`

carlos-sarmiento, 2023-01-31:
Thanks for the report. It is an issue with some conditionally included code, which is something done to reduce the size of the firmware by not including code if it is not being used.

I will work on a fix, but in the meantime the workaround is to set up an entity of the required types so the compiler includes the right files. It doesn't need to be any from ezo_pmp. A template sensor, binary_sensor, etc will work just as well.

V4l3n10, 2023-03-25:
@carlos-sarmiento managed to compile it beautifully. Would you, please share a configuration example for a multi-ezo_pmp configuration?

ssieb, 2023-03-25:
@V4l3n10, that is not related to this issue and github issues are not for support requests.  Ask on the discord server or one of the forums.

nevets85, 2023-04-02:
I'm having a similar issue with the pH sensor, no ezo_pmp.

yaml
sensor:
  - platform: ezo
    id: ph_ezo
    name: "pH"
    address: 99
    accuracy_decimals: 1
    unit_of_measurement: "pH"

`Compiling /data/test-ezo/.pioenvs/poolmonitor/src/esphome/components/ezo/ezo.cpp.o
src/esphome/components/ezo/ezo.cpp: In member function 'virtual void esphome::ezo::EZOSensor::update()':`