esphome/issues #3833: On iOS/iPadOS It’s Impossible to mark/cut/paste blocks of text in the editor in the Home Assistant addon

gaussey, 2022-11-25:
### The problem

The editor was changed a few versions again and I was hoping that it’d be resolved by now but as per the title. I’m having to use file editor to edit configuration files.

### Which version of ESPHome has the issue?

2022.11.3 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

nagyrobi, 2022-11-30:
Same in Firefox

gdam12886, 2022-12-14:
Same in Android Chrome

nagyrobi, 2022-12-14:
Same on Android 13 within HA companion app (webview).

HHHartmann, 2024-12-30:
Stoll having problems editing in Android companion app.
The problem is that key strokes are not recorded correctly, especially deleting characters seems to move the cursor to a place not represented in the UI leading to garbled code. Also pasting from the system clipboard does not work in the custom context menu (long press) copy however places the marked code in the system clipboard.
Since I do not habe a Keyboard I cannot use the ctrl-v shortcut to paste