esphome/issues #3835: unable to conf. pin 14 on esp32s2 for ADC, datasheet specifies pin 14 as ADC2_CH3

Zixim, 2022-11-25:
### The problem

```
esp32:
  board: esp32-s2-saola-1
  variant: esp32s2
  framework:
    type: esp-idf
    version: recommended
```
trying to configure pin 14 as ADC, getting error : 
```
Failed config

sensor.adc: [source /config/esphome/home-buttons-01.yaml:64]
  platform: adc
  
  ESP32S2 doesn't support ADC on this pin.
  pin: GPIO14
  name: Battery Voltage
  id: batt_volt
  update_interval: 10s
  attenuation: 11db
```

### Which version of ESPHome has the issue?

2022.11.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Component causing the issue

adc

### Example YAML snippet

```yaml
sensor:
  # battery voltage adc
  - platform: adc
    pin: GPIO14
    name: $devicename Battery Voltage
    id: batt_volt
    update_interval: 10s
    attenuation: 11db
```
```

ssieb, 2022-11-25:
You can't use ADC2 and wifi at the same time.
https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s2/api-reference/peripherals/adc.html#adc-limitations

closed 2022-11-25 by ssieb

Zixim, 2022-11-26:
Any chance ADC could be read before wifi becomes active ?
This is for a device waking up from deep_sleep (either after timer or after button press).

mafrosis, 2022-12-06:
https://docs.espressif.com/projects/esp-idf/en/v4.4.1/esp32s2/api-reference/peripherals/adc.html#adc-channels

> These channels are supported:
> 
> ADC1:
> 10 channels: GPIO1 - GPIO10
> 
> ADC2:
> 10 channels: GPIO11 - GPIO20

ADC2 is not supported, but you can use ADC1 which is on GPIO1 - GPIO10