esphome/issues #3837: Custom Uart Sensor Not publishing

Morbidbystandard, 2022-11-26:
### The problem

I'm not getting a string published using a custom uart sensor.

### Which version of ESPHome has the issue?

2022.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

ESP32

### Board

esp-wrover

### Component causing the issue

Uart

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  
sensor:
- platform: custom
  lambda: |-
    auto EZO = new scaleSensor(id(uart_bus));
    App.register_component(EZO);
    return {EZO};
  sensors:
    name: "Ph Sensor"
    unit_of_measurement: Ph
    accuracy_decimals: 2
    filters:
      - offset: +0.22
```

### Anything in the logs that might be useful for us?

```txt
#include "esphome.h"

class scaleSensor : public PollingComponent, public UARTDevice, public sensor::Sensor{
 public:
  scaleSensor(UARTComponent *parent) : PollingComponent(1000), UARTDevice(parent) {}
    float weight = weight;
  
  void setup() override {
    // nothing to do here
  }
  
  void loop() override {
     char character;
    while(available()){
      character = read();
      if(character == '\n')
      break;
    }
  }
  
  void update() {
    publish_state(weight);
  }
};
```

ssieb, 2022-11-26:
GitHub issues are not for support requests.  Ask on the esphome discord server or one of the forums.

That code doesn't make any sense...  If you want a component to start with, I have a text_sensor that reads lines from the uart.
https://github.com/ssieb/custom_components/tree/master/components/serial

closed 2022-11-26 by ssieb

Morbidbystandard, 2022-11-26:
This code used to work flawlessly until the big update last year. It's really quite frustrating when changes are made and esphome.io doesn't implement the changes. The support for custom UART sensor is in c++ still.

ssieb, 2022-11-26:
That code you posted could never have worked.  I would be surprised if it even compiled.
Why don't you switch your ezo to i2c mode and use the existing component?