esphome/issues #3839: Heltec ESP32 WiFi board GPIO not able to program

ChristianSwift, 2022-11-26:
### The problem

Board type: heltec_wifi_kit_32_v2  
IDE report in first line with "'str' object has no attribute 'values'" once use GPIO for input or output  
[image]  
[image]  
[image]

### Which version of ESPHome has the issue?

2022.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

heltec_wifi_kit_32_v2

### Component causing the issue

output

ChristianSwift, 2022-11-26:
This problem was occurred when I trying update esphome version 2022.11.1 to 2022.11.3, not only happened for output component but status_led, binary light, etc..

ssieb, 2022-11-27:
Try compiling it to get a better error message.

ChristianSwift, 2022-11-29:
> Try compiling it to get a better error message.

INFO Reading configuration /config/esphome/sensor.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 955, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 933, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
  File "/esphome/esphome/config.py", line 977, in read_config
    res = load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 831, in load_config
    return _load_config(command_line_substitutions)
  File "/esphome/esphome/config.py", line 819, in _load_config
    result = validate_config(config, command_line_substitutions)
  File "/esphome/esphome/config.py", line 747, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 123, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 498, in run
    validated = schema(self.conf)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/voluptuous_schema.py", line 34, in __call__
    res = super().__call__(data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/esphome/esphome/voluptuous_schema.py", line 148, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 229, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 353, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/pins.py", line 76, in validator
    return PIN_SCHEMA_REGISTRY[CORE.target_platform][1](value_d)[CONF_NUMBER]
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 232, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/validators.py", line 351, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 387, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.9/dist-packages/voluptuous/schema_builder.py", line 818, in validate_callable
    return schema(data)
  File "/esphome/esphome/components/esp32/gpio.py", line 111, in validate_gpio_pin
    if value in board_pins.values():
AttributeError: 'str' object has no attribute 'values'

randybb, 2022-11-29:
`GPI01 != GPIO1`

randybb, 2022-11-29:
Ideally, paste the config so we can test it.

ChristianSwift, 2022-12-05:
> 理想情况下，粘贴配置，以便我们可以对其进行测试。

esphome:
  name: sensor

esp32:
  board: heltec_wifi_kit_32_v2
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

web_server:
  port: 80

ota:
  password: "a62fd43dff427e8dc8e9d6c1585389c3"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sensor Fallback Hotspot"
    password: "eN0KY51h9dvf"

captive_portal:

# font:
#   - file: "fonts/consola.ttf"
#     id: font1
#     size: 20
    
i2c:
  sda: 4
  scl: 15
  scan: True

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: 16
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");

ssieb, 2022-12-05:
For now, use a different board name like `esp32dev`.

jesserockz, 2022-12-05:
This will be fixed in https://github.com/esphome/esphome/pull/4147.
I will release in 2022.11.5

closed 2022-12-05 by jesserockz

ChristianSwift, 2022-12-06:
Confirm this problem solved!