esphome/issues #3840: api/i2c crashes when attempting to play radio stream

alfwro13, 2022-11-26:
### The problem

I have a Node MCU board with UDA1334A I2S DAC module. Playing audio files (mp3), text to speach is fine.
When trying to play online radio esphome api crashes

### Which version of ESPHome has the issue?

2022.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP32

### Board

Node MCU

### Component causing the issue

api/i2s

### Example YAML snippet

```yaml
media_player:
  - platform: i2s_audio
    name: ${friendly_name}
    dac_type: external
    i2s_lrclk_pin: 21
    i2s_dout_pin: 19
    i2s_bclk_pin: 18
    mode: stereo
    id: media_out
```

### Anything in the logs that might be useful for us?

```txt
[21:29:10][I][app:102]: ESPHome version 2022.11.2 compiled on Nov 26 2022, 21:28:30
[21:29:10][C][status_led:019]: Status LED:
[21:29:10][C][status_led:020]:   Pin: GPIO2
[21:29:10][C][wifi:504]: WiFi:
[21:29:10][C][wifi:362]:   Local MAC: [redacted]
[21:29:10][C][wifi:363]:   SSID: [redacted]
[21:29:10][C][wifi:364]:   IP Address: [redacted]
[21:29:10][C][wifi:366]:   BSSID: [redacted]
[21:29:10][C][wifi:367]:   Hostname: 'workshop-speaker'
[21:29:10][C][wifi:369]:   Signal strength: -30 dB ▂▄▆█
[21:29:10][C][wifi:373]:   Channel: 6
[21:29:10][C][wifi:374]:   Subnet: 255.255.255.0
[21:29:10][C][wifi:375]:   Gateway: [redacted]
[21:29:10][C][wifi:376]:   DNS1: [redacted]
[21:29:10][C][wifi:377]:   DNS2: [redacted]
[21:29:10][C][logger:293]: Logger:
[21:29:10][C][logger:294]:   Level: DEBUG
[21:29:10][C][logger:295]:   Log Baud Rate: 115200
[21:29:10][C][logger:296]:   Hardware UART: UART0
[21:29:10][C][mdns:103]: mDNS:
[21:29:10][C][mdns:104]:   Hostname: workshop-speaker
[21:29:10][C][ota:093]: Over-The-Air Updates:
[21:29:10][C][ota:094]:   Address: [redacted]:3232
[21:29:10][C][ota:097]:   Using Password.
[21:29:10][C][api:138]: API Server:
[21:29:10][C][api:139]:   Address: [redacted]:6053
[21:29:10][C][api:141]:   Using noise encryption: YES
[21:29:10][C][wifi_signal.sensor:009]: WiFi Signal 'Workshop Speaker WiFi Signal Sensor'
[21:29:10][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[21:29:10][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[21:29:10][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[21:29:10][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:29:10][C][audio:135]: Audio:
[21:29:39][D][sensor:127]: 'Workshop Speaker WiFi Signal Sensor': Sending state -30.00000 dBm with 0 decimals of accuracy
[21:30:01][D][api:102]: Accepted ::FFFF:C0A8:147
[21:30:01][D][api.connection:917]: Home Assistant 2022.11.4 (::FFFF:C0A8:147): Connected successfully
[21:30:03][D][media_player:059]: 'Workshop Speaker' - Setting
[21:30:03][D][media_player:069]:   Volume: 0.08
[21:30:05][D][media_player:059]: 'Workshop Speaker' - Setting
[21:30:05][D][media_player:069]:   Volume: 0.13
[21:30:13][D][media_player:059]: 'Workshop Speaker' - Setting
[21:30:13][D][media_player:066]:   Media URL: http://st01.dlf.de/dlf/01/128/mp3/stream.mp3
INFO [redacted]: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO [redacted]: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for [redacted]
WARNING Disconnected from API
INFO Successfully connected to [redacted]
[21:30:26][D][api:102]: Accepted ::FFFF:C0A8:147
[21:30:26][D][api.connection:917]: Home Assistant 2022.11.4 (::FFFF:C0A8:147): Connected successfully
[21:30:40][D][media_player:059]: 'Workshop Speaker' - Setting
[21:30:40][D][media_player:066]:   Media URL: http://stream.live.vc.bbcmedia.co.uk/bbc_radio_fourfm
INFO [redacted]: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO [redacted]: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for [redacted]
WARNING Disconnected from API
INFO Successfully connected to [redacted]
[21:31:01][D][media_player:059]: 'Workshop Speaker' - Setting
[21:31:01][D][media_player:066]:   Media URL: https://icecast.walmradio.com:8443/classic
INFO [redacted]: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO [redacted]: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for [redacted]
WARNING Disconnected from API
INFO Successfully connected to [redacted]
```

### Additional information

.

masman69, 2023-02-07:
Same issue, using ESPHome 22.12.8

Tried with both a ESP32 ESP-WROOM-32 Development (aka Amazon special) and a ESP32 Mini ESP-WROOM-32 Development board

Relevant section of log downloaded:

[ 75901][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
E (90468) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (90468) task_wdt:  - loopTask (CPU 1)
E (90468) task_wdt: Tasks currently running:
E (90468) task_wdt: CPU 0: IDLE
E (90468) task_wdt: CPU 1: IDLE
E (90468) task_wdt: Aborting.

I get this error either trying a local mp3 or Nabu Casa TTS

Relevant configuration:
substitutions:
  i2s_lrclk_pin: GPIO33
  i2s_dout_pin: GPIO22
  i2s_bclk_pin: GPIO19

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    dac_type: external
    i2s_lrclk_pin: $i2s_lrclk_pin
    i2s_dout_pin: $i2s_dout_pin
    i2s_bclk_pin: $i2s_bclk_pin
    mode: mono

randybb, 2023-02-07:
I don't have any issues with playing mentioned streams, but when it cannot connect, for example stream address is not valid or it is blocked on a FW, then it might crash.

Kosygor, 2023-06-17:
Hey. Was having similar issue.
@randybb Thank You ! As You wrote Crash can be caused by ESPHome node being unable to reach file in network. 
During research I found out that I had messed Setting > Network  > URL adress Setting. Which was greyed out (blocked) because it supposedly was configured in configuration.yaml.... (not true) in fack elevation, longitude and so one entries also block setting URL by hand ... 
Fixed that Local URL and immediately got speaker working.