esphome/issues #3841: rp2040_pwm does not work properly for frequencies lower than ~2khz

JasonYuJjyu, 2022-11-27:
### The problem

The output component rp2040_pwm does not function correctly when the frequency is set to below ~2khz. Doing so is desirable when using PWM to drive servos at 50hz.

The rp2040 clock divisor has a valid range of 1.f <= value < 256.f. Setting lower frequencies result in the divisor being set far beyond this range:

https://github.com/esphome/esphome/blob/c5f59fad626534c55d58ad6927ab994cafc6a7b3/esphome/components/rp2040_pwm/rp2040_pwm.cpp#L26

Fixing this for lower frequencies can be done by increasing the PWM wrap until the divisor becomes valid. Currently it hard coded, maybe this could be user configurable.

### Which version of ESPHome has the issue?

v2022.11.3

### What type of installation are you using?

Docker

### What platform are you using?

RP2040

### Board

Raspberry PI pico w

### Component causing the issue

components/rp2040_pwm

### Example YAML snippet

```yaml
esphome:
  name: picow-servo-test

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

api:
  services:
    - service: control_servo
      variables:
        level: float
      then:
        - servo.write:
            id: my_servo
            level: !lambda 'return level / 100.0;'
    
output:
  - platform: rp2040_pwm
    id: pwm_output
    pin: 15
    frequency: 50hz

servo:
  - id: my_servo
    output: pwm_output
```

freddiemare, 2023-01-13:
> ### The problem
> The output component rp2040_pwm does not function correctly when the frequency is set to below ~2khz. Doing so is desirable when using PWM to drive servos at 50hz.
> 
> The rp2040 clock divisor has a valid range of 1.f <= value < 256.f. Setting lower frequencies result in the divisor being set far beyond this range:
> 
> https://github.com/esphome/esphome/blob/c5f59fad626534c55d58ad6927ab994cafc6a7b3/esphome/components/rp2040_pwm/rp2040_pwm.cpp#L26
> 
> Fixing this for lower frequencies can be done by increasing the PWM wrap until the divisor becomes valid. Currently it is hard coded, maybe this could be user configurable.
> 
> ### Which version of ESPHome has the issue?
> v2022.11.3
> 
> ### What type of installation are you using?
> Docker
> 
> ### Which version of Home Assistant has the issue?
> _No response_
> 
> ### What platform are you using?
> RP2040
> 
> ### Board
> Raspberry PI pico w
> 
> ### Component causing the issue
> components/rp2040_pwm
> 
> ### Example YAML snippet
> ```yaml
> esphome:
>   name: picow-servo-test
> 
> rp2040:
>   board: rpipicow
>   framework:
>     # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
>     platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git
> 
> api:
>   services:
>     - service: control_servo
>       variables:
>         level: float
>       then:
>         - servo.write:
>             id: my_servo
>             level: !lambda 'return level / 100.0;'
>     
> output:
>   - platform: rp2040_pwm
>     id: pwm_output
>     pin: 15
>     frequency: 50hz
> 
> servo:
>   - id: my_servo
>     output: pwm_output
> ```
> 
> ### Anything in the logs that might be useful for us?
> _No response_
> 
> ### Additional information
> _No response_

That is exactly my finding also after struggling with the pi pico w all night to understand the issue at 50Hz the frequency show up on my oscilloscope as 2.72kHz

need attention since most servos require 1 to 2 millie seconds pulse width but at 2.72kHz it is at 184 microseconds thus rendering the servo useless for control.

Or any suggestions will be helpful.

closed 2023-01-16 by jesserockz

freddiemare, 2023-02-25:
confirm tested and the change does work perfectly except when the node restart then the servo makes an unplanned movement that is consistent that is not quite desirable. I try to find the reason.
Also, sometimes when I make a small change in the Yamal not even related to the servo, the servo will make a full rotation but not consistently. I think there should be a method to detach the servo completely during OTA and a clear method for isolation to prevent unplanned movements.