esphome/issues #3843: Setting up BME280 through multiplexer awkward

swifty99, 2022-11-27:
### The problem

Using the BME280 through I2C multiplexing is working. However there is probably something wrong with the setup sequence.
While logging through serial the log gets spammed with:
```
...
[C][bme280.sensor:086]: Setting up BME280...
[C][bme280.sensor:086]: Setting up BME280...
[C][bme280.sensor:086]: Setting up BME280...
[C][bme280.sensor:086]: Setting up BME280...
[C][bme280.sensor:086]: Setting up BME280...
[C][bme280.sensor:086]: Setting up BME280...
[C][bme280.sensor:086]: Setting up BME280...
...
```
 I cannot see an issue in the library itself, however multiplexing code might do something strange

### Which version of ESPHome has the issue?

2022.9.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.3

### What platform are you using?

ESP8266

### Board

wemos D1

### Component causing the issue

TCA9548A

### Example YAML snippet

```yaml
i2c:
  sda: D2
  scl: D1
  scan: true
  id: i2c_a

tca9548a:
  - address: 0x70
    id: multiplex0
    i2c_id: i2c_a
    channels:
      - bus_id: multiplex0channel0
        channel: 0
      - bus_id: multiplex0channel1
        channel: 1
      - bus_id: multiplex0channel2
        channel: 2

sensor:
  - platform: bme280
    address: 0x76
    i2c_id: multiplex0channel1
    update_interval: 1990ms  
    
    temperature:
      name: "bme mux1"
[... 301 lines cut ...]
[C][bme280.sensor:175]:   Address: 0x76
[E][bme280.sensor:178]: Communication with BME280 failed!
[C][bme280.sensor:187]:   IIR Filter: OFF
[C][bme280.sensor:188]:   Update Interval: 2.0s
[C][bme280.sensor:190]:   Temperature 'bme mux1'
[C][bme280.sensor:190]:     Device Class: 'temperature'
[C][bme280.sensor:190]:     State Class: 'measurement'
[C][bme280.sensor:190]:     Unit of Measurement: '°C'
[C][bme280.sensor:190]:     Accuracy Decimals: 1
[C][bme280.sensor:191]:     Oversampling: 16x
[C][bme280.sensor:193]:     Oversampling: 16x
[C][bme280.sensor:195]:     Oversampling: 16x
[C][bme280.sensor:174]: BME280:
[C][bme280.sensor:175]:   Address: 0x76
[C][bme280.sensor:187]:   IIR Filter: OFF
[C][bme280.sensor:188]:   Update Interval: 2.0s
[C][bme280.sensor:190]:   Temperature 'bme mux2'
[C][bme280.sensor:190]:     Device Class: 'temperature'
[C][bme280.sensor:190]:     State Class: 'measurement'
[C][bme280.sensor:190]:     Unit of Measurement: '°C'
[C][bme280.sensor:190]:     Accuracy Decimals: 1
[C][bme280.sensor:191]:     Oversampling: 16x
[C][bme280.sensor:193]:     Oversampling: 16x
[C][bme280.sensor:195]:     Oversampling: 16x
```

swifty99, 2023-01-01:
just an update,
this not related to multiplexing. bare ESP8266 node exhibits the same behaviour:

````Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:24:37]\xf2nn'l`\x90nn$`nr\x8e\x92\x92nrl`r\x92\x92nrl`\x9c\x8c$bl`\xfc\x82n\x92`[I][logger:258]: Log initialized
[15:24:37][C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[15:24:37][I][app:029]: Running through setup()...
[15:24:37][I][i2c.arduino:175]: Performing I2C bus recovery
[15:24:37][D][text_sensor:067]: 'zz-test-i2s Mac Wifi Address': Sending state 'C8:C9:A3:5E:0F:B8'
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][wifi:037]: Setting up WiFi...
[15:24:37][C][wifi:038]:   Local MAC: C8:C9:A3:5E:0F:B8
[15:24:37][D][wifi:386]: Starting scan...
[15:24:37][D][sensor:126]: 'zz-test-i2s_Uptime': Sending state 0.00000 Tage with 0 decimals of accuracy
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:37][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:38][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[15:24:39][C][bme280.sensor:086]: Setting up BME280...
[... 225 lines cut ...]
[15:24:45][C][bme280.sensor:194]:     Device Class: 'humidity'
[15:24:45][C][bme280.sensor:194]:     State Class: 'measurement'
[15:24:45][C][bme280.sensor:194]:     Unit of Measurement: '%'
[15:24:45][C][bme280.sensor:194]:     Accuracy Decimals: 1
[15:24:45][C][bme280.sensor:195]:     Oversampling: 4x
[15:24:45][C][captive_portal:088]: Captive Portal:
[15:24:45][C][mdns:103]: mDNS:
[15:24:45][C][mdns:104]:   Hostname: zz-test-i2s
[15:24:45][C][ota:093]: Over-The-Air Updates:
[15:24:45][C][ota:094]:   Address: 192.168.178.112:8266
[15:24:45][C][api:138]: API Server:
[15:24:45][C][api:139]:   Address: 192.168.178.112:6053
[15:24:45][C][api:143]:   Using noise encryption: NO
[15:24:45][C][wifi_info:009]: WifiInfo IPAddress 'IP Address zz-test-i2s'
[15:24:45][C][wifi_info:011]: WifiInfo SSID 'zz-test-i2s Connected SSID'
[15:24:45][C][wifi_info:012]: WifiInfo BSSID 'zz-test-i2s Connected BSSID'
[15:24:45][C][wifi_signal.sensor:009]: WiFi Signal 'zz-test-i2s_WiFi_Signal'
[15:24:45][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[15:24:45][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[15:24:45][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[15:24:45][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[15:26:46][D][sensor:126]: 'zz-test-i2s Temperatur': Sending state 18.68000 °C with 1 decimals of accuracy
[15:26:46][D][sensor:126]: 'zz-test-i2s Feuchtigkeit': Sending state 54.32520 % with 1 decimals of accuracy

````

rix84, 2023-01-24:
Any idea about that ?
I have the same behavior now (without multiplexing) but I have two devices on the same I2C. Can be related ??? At least looks like the sensor is working fine. 
Thanks

karolkrzempek, 2023-02-06:
Hi! 
Did you find a solution to this problem? 
I am having the same issues with my ESP + BME280. 
TY.

rix84, 2023-02-07:
Hi, Look like the problem disappear by himself with an ESPHome Update or because I change something. Sorry I don't have more info. (At least it's not a blocking issue, sensor is working fine)