esphome/issues #3846: MQTT_Custom_Component not working

AquaMCU, 2022-11-27:
### The problem

I tried to follow the documentation for the MQTT custom component. It does not work :( https://esphome.io/custom/custom_component.html#mqtt-custom-component

First I get:
src/lora_gateway.h:3:63: error: expected class-name before '{' token
 class LORAGateway : public Component, public CustomMQTTDevice {

When I add #include <custom_mqtt_device.h> I get the error
In file included from src/main.cpp:39:0:
src/lora_gateway.h:2:32: fatal error: custom_mqtt_device.h: No such file or directory

### Which version of ESPHome has the issue?

 v2022.11.1 

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: gateway
  libraries:
    - SPI
    - "LoRa"

  includes:
    - lora_gateway.h

  
esp32:
  board: heltec_wifi_lora_32_V2
  framework:
    type: arduino

logger:
   level: DEBUG

[...]
sensor:
- platform: custom
  lambda: |-
    auto LORA_Gatway = new LORAGateway();
    return {LORA_Gateway};
  sensors:
    name: "LORA Data"
```

### Anything in the logs that might be useful for us?

```txt
Please update the doc, so I can follow easyly. BTW. Awesome project!
```

AquaMCU, 2022-11-28:
Here is the code:

#include "esphome.h"
#include <custom_mqtt_device.h>

class LORAGateway : public Component, public CustomMQTTDevice {
 public:
  void setup() override {
    pinMode(6, OUTPUT);

    subscribe("the/topic", &LORAGateway::on_message);

    subscribe_json("the/json/topic", &MyCustomComponent::on_json_message);
  }
  void on_message(const std::string &payload) {
    if (payload == "ON") {
      digitalWrite(6, HIGH);
      publish("the/other/topic", "Hello World!");
    } else {
      digitalWrite(6, LOW);
      publish("the/other/topic", 42);
    }
  }

};

AquaMCU, 2022-11-29:
I found the lib. The error is gone when I change the path in the include do:

#include "/esphome/esphome/components/mqtt/custom_mqtt_device.h"

Nevertheless, I get now the following error:

src/lora_gateway.h:4:63: error: expected class-name before '{' token
 class LORAGateway : public Component, public CustomMQTTDevice {

I checked if I have typo, but the line seems to be correct. Any ideas?

AquaMCU, 2022-12-02:
Any help? Please. I want to develop a Lora Module for y'all. I need this to start and I can not figure it out.

ssieb, 2022-12-03:
You shouldn't be including that header.  You don't show your entire config, but I expect the problem is that you don't have an `mqtt:` component defined.
The github issues section is not for support requests.  You should either come to the esphome discord server or one of the forums.

closed 2022-12-03 by ssieb

AquaMCU, 2022-12-03:
THX. I thought is was a bug in the doc, but I was missing the mqtt: ;)