esphome/issues #3847: ESPHome Home page online status of devices

janneman001, 2022-11-28:
### The problem

In the ESPHome Home page of devices the online status is incorrect when the device is not on the local (wifi)network.

This happens after the device is placed on another network/domain like this:

wifi:
  ssid: !secret iot_ssid
  password: !secret iot_password
  use_address: $name_of_device.iotll

[image]

### Which version of ESPHome has the issue?

2022.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

RP2040

### Board

raspberry pico w

### Component causing the issue

home assistant addon

### Example YAML snippet

```yaml
wifi:
  ssid: !secret iot_ssid
  password: !secret iot_password
  use_address: $name_of_device.iotll
```

### Anything in the logs that might be useful for us?

```txt
the logs are fine also the integration in home assistant works fine.
It is just the representation of online/offline that is incorrect when used like this.
```

xorphar, 2022-11-30:
You can enable `status_use_ping` on the ESPHome addon configuration, this should fix it.

janneman001, 2022-11-30:
Works like a charm! Thnx.

This bug:  https://github.com/esphome/issues/issues/80
Reference here: https://esphome.io/guides/faq#docker-reference

I use home assistant add-on. I could set it when i disclosed the unused config options at addon configuration. Set it, restart addon and it is ok.

closed 2022-11-30 by janneman001