esphome/issues #3849: RP2040 Installation Preparation Fails

scotty3785, 2022-11-28:
### The problem

When using the dev branch for the RP2040 platform, I click on an RP2040 target and click Install. Then select to install via the USB drive. 
During the preparation process, a "preperation failed" message appears. When I click to "See what went wrong" I get a number of compilation errors such as "arm-none-eabi-g++: fatal error". See log below.

### Which version of ESPHome has the issue?

2022.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

RP2040

### Board

PicoW

### Anything in the logs that might be useful for us?

```txt
`INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
 - framework-arduinopico @ 1.20602.0 (2.6.2) 
 - tool-rp2040tools @ 1.0.2 
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium Sketch size: 1044480 EEPROM start: 0x101ff000 Filesystem start: 0x100ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0.0
|   |-- Updater @ 1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|   |   |-- PicoOTA @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |-- MD5Builder @ 1.0.0
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- lwIP_CYW43 @ 1
|   |   |-- SPI @ 1.0
|   |-- SPI @ 1.0
|-- LEAmDNS @ 1.2
|   |-- lwIP-Ethernet @ 1
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|   |-- WiFi @ 1.0.0
|   |   |-- Updater @ 1.0
|   |   |   |-- MD5Builder @ 1.0.0
|   |   |   |-- LittleFS @ 0.1.0
|   |   |   |-- PicoOTA @ 1.0.0
|   |   |   |   |-- LittleFS @ 0.1.0
|   |   |-- MD5Builder @ 1.0.0
|   |   |-- lwIP-Ethernet @ 1
|   |   |   |-- lwIP_CYW43 @ 1
|   |   |   |   |-- SPI @ 1.0
|   |   |   |-- SPI @ 1.0
|   |   |-- lwIP_CYW43 @ 1
|   |   |   |-- SPI @ 1.0
|   |   |-- SPI @ 1.0
|-- Updater @ 1.0
|   |-- MD5Builder @ 1.0.0
|   |-- LittleFS @ 0.1.0
|   |-- PicoOTA @ 1.0.0
|   |   |-- LittleFS @ 0.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- MD5Builder @ 1.0.0
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_connection.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/api/api_server.cpp.o
arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_connection.cpp.o] Error 1
arm-none-eabi-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
*** [/data/test/.pioenvs/test/src/esphome/components/api/api_server.cpp.o] Error 1
========================= [FAILED] Took 43.96 seconds =========================`
```

randybb, 2022-11-28:
rpi3? :D Your machine is killing gcc as it has not enough memory.

scotty3785, 2022-11-28:
> rpi3? :D Your machine is killing gcc as it has not enough memory.

Yes it is a Pi3. Funny as I have run the standard Pico SDK on a Pi3 before.

randybb, 2022-11-28:
But now esphome is running inside of a docker container. Don't know if it is limiting them based on available resources, but there is a million of logged issues with the same problem.
Could you please add the new parameter introduces in 2022.11? 
```yaml
esphome:
  compile_process_limit: 1
```
https://esphome.io/changelog/2022.11.0.html#running-esphome-on-lower-powered-machines

randybb, 2022-11-28:
What page? It goes to your device configuration https://esphome.io/components/esphome.html, so it needs to be added to config for every device, there is no global option right now.

scotty3785, 2022-11-28:
> What page? It goes to your device configuration https://esphome.io/components/esphome.html, so it needs to be added to config for every device, there is no global option right now.

Got it. I thought that adding it as a global option would add it to new configs. Adding it manually to each of my RP2040 configs works perfectly.

flouck, 2024-02-02:
@randybb I don't get it. Why is using a manual ip address for my esphome device is a breaking change when using this option? This is what I read 
```
esphome:
  compile_process_limit: 1
```

I tried adding this line but I still get the problem. I run hassio on my RPi 4.

ssieb, 2024-02-02:
@flouck how much RAM does it have?  And is the error the same as in the original description?

flouck, 2024-02-02:
Thanks for your quick reply :-) 
2 GB RAM. 
Slightly different log error when compling:
````xtensa-lx106-elf-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.``

ssieb, 2024-02-03:
Ok, that's basically the same and it's the same reason: too little RAM.  I suggest you compile on something else.

flouck, 2024-02-03:
No way? 
How come? I have been compiling with the addon for years and never had any issue. Do you think this is a temporary problem which will be solved by a future update?

ssieb, 2024-02-03:
No.  You've always been close to it not working.  It depends on what is running at the time.  But newer compilers tend to use more RAM for better optimization.  You could try rebooting it and try then, but eventually it's not going to work.  And it's going to be really slow anyway, especially since you're limiting it to one process.  This is nothing that esphome can control anyway other than the process limit that was added.

closed 2024-02-03 by ssieb