esphome/issues #3854: Compilation for Pico Fails on Raspberry Pi 4

HarvsG, 2022-11-29:
### The problem

Because of #3853, I tried compiling on my raspberry pi 4 instead, and hit a different issue.

### Which version of ESPHome has the issue?

2022.11.3

### What type of installation are you using?

Docker

### What platform are you using?

RP2040

### Board

Pico-W

### Component causing the issue

Compile

### Example YAML snippet

```yaml
esphome:
  name: pico-w

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "l"

ota:
  password: "6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pico-W Fallback Hotspot"
    password: "862oPcqca8Ct"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/pico-w.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing pico-w (board: rpipicow; framework: arduino; platform: https://github.com/maxgerhardt/platform-raspberrypi.git)
--------------------------------------------------------------------------------
Tool Manager: Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
INFO Installing earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714' requirements for your system 'linux_armv7l'
```

btwebz, 2022-12-01:
I am having the exact same issue as @HarvsG 
Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714' requirements for your system 'linux_armv7l'

ssieb, 2022-12-01:
There is no toolchain available for that variety of 32-bit ARM.  That's up to the provider of the toolchain.

closed 2022-12-01 by ssieb

maxgerhardt, 2022-12-02:
>Error: Could not find the package with 'earlephilhower/toolchain-rp2040-earlephilhower @ 5.100300.220714' requirements for your system 'linux_armv7l'

Are you running a Pi 4 on 32-bit ARM? Just use the right 64-bit version, @earlephilhower has uploaded the toolchaim for ARM64 (and linux_armv6l, which would e.g. be a Pi Zero)

btwebz, 2022-12-02:
Yes, I am running a Pi 4 with a 32-bit OS. Is there a toolchain that I can use?

maxgerhardt, 2022-12-02:
There is no package uploaded for it, so you would have to open an issue in https://github.com/earlephilhower/arduino-pico/issues to get that rolling. Technically I don't see a reason why it shouldn't be able to be compiled for 32-bit ARMv7L, but since right now it isn't, your easiest option is to just put a 64-bit on the SD card instead.

HarvsG, 2022-12-02:
Yes running - 32bit. I can't remember why but I think there was a good reason - compatibility with another application I run.

Issue solved by comment above.

btwebz, 2022-12-02:
Thanks for that - I am running ESPHome alongside Home Assistant and trying to get an rpi pico w working with it.

btwebz, 2022-12-02:
Thank you so much - I really appreciate the help!

HarvsG, 2022-12-02:
Will be in next toolchain build:
https://github.com/earlephilhower/pico-quick-toolchain/pull/17
https://github.com/earlephilhower/pico-quick-toolchain/issues/22#issuecomment-1335505719