esphome/issues #3855: I2S Audio not compiling for the ESP32 S2

LucaDev, 2022-11-29:
### The problem

Because the ESP32 S2 does not contain an internal DAC the mediaplayer will fail to compile even when a external DAC is being used  because the i2s_dac_mode_t struct isn't available.

See: https://github.com/espressif/esp-idf/blob/b04e33256df3a6149e75a4704393322b9149df6c/components/driver/deprecated/driver/i2s_types_legacy.h#L130

Example of it being used regardless of the target IC:
https://github.com/esphome/esphome/blob/72b294333287658ecf8577de5cafd05a901d7921/esphome/components/i2s_audio/i2s_audio_media_player.h#L28

### Which version of ESPHome has the issue?

dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.11.3

### What platform are you using?

ESP32

### Board

ESP32 S2

### Component causing the issue

i2s_audio

### Example YAML snippet

```yaml
esphome:
  name: audioplayer

esp32:
  board: lolin_s2_mini

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    dac_type: external
    i2s_lrclk_pin: GPIO16
    i2s_dout_pin: GPIO17
    i2s_bclk_pin: GPIO18
    mode: stereo
```

alfwro13, 2022-12-14:
this is the same issue as the one already reported here: [https://github.com/esphome/issues/issues/3783](url)

LucaDev, 2022-12-15:
Sorry I missed that. I'll provide a PR with a fix whenever I have some spare time in the next couple of weeks if the esphome team doesn't fix it in the meantime

jesserockz, 2022-12-15:
Closing as a duplicate of #3783

closed 2022-12-15 by jesserockz