esphome/issues #3860: PID autotune: Trying to reach nan °C

mgabor3141, 2022-12-01:
### The problem

I'm seeing something strange in the logs of my PID autotune. The regular "still running" entries are printing `nan`:

```
[I][pid.autotune:134]: PID Autotune:
[I][pid.autotune:179]:   Autotune is still running!
[D][pid.autotune:180]:   Status: Trying to reach nan °C
[D][pid.autotune:181]:   Stats so far:
[D][pid.autotune:182]:     Phases: 0
[D][pid.autotune:183]:     Detected 0 zero-crossings
[D][pid.autotune:184]:     Current Phase Min: nan, Max: nan
```

The sensor and climate entity output looks like this on the frontend:

[image]

Do I need to put the climate entity into HEAT mode? Is it waiting for the first new data point to be reported by the temperature sensor?

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.4

### What platform are you using?

ESP8266

### Board

Lolin D1 Mini

### Component causing the issue

pid climate

### Example YAML snippet

```yaml
output:
  - platform: slow_pwm
    id: study_pwm
    period: 270s
    turn_on_action:
      - uart.write: [0xA0, 0x01, 0x01, 0xA2]
    turn_off_action:
      - uart.write: [0xA0, 0x01, 0x00, 0xA1]

sensor:
  - platform: homeassistant
    id: study_temperature
    entity_id: sensor.study_temperature

climate:
  - platform: pid
    name: "Study Heating"
    id: study_heating
    sensor: study_temperature
    default_target_temperature: 21°C
    heat_output: study_pwm
    visual:
      min_temperature: 18
      max_temperature: 28
      temperature_step: 0.5

button:
  - platform: template
    name: "Study Heating Autotune"
    on_press: 
      - climate.pid.autotune: study_heating
```

gabormagyar, 2022-12-03:
Indeed it looks like it was just waiting for the first update from the sensor, it printed nan while the temperature was "stale". So no bug here, but maybe the logs should be updated to make this clearer.

remyderuysscher, 2022-12-31:
> Indeed it looks like it was just waiting for the first update from the sensor, it printed nan while the temperature was "stale". So no bug here, but maybe the logs should be updated to make this clearer.

I have some similar issue with a e-ink display, which has a correct temperature value but is printed on the display as nan.  Even a static template sensor displays nan.

Can I open an issue and mention you?

mgabor3141, 2022-12-31:
The solution to this is actually mentioned in the docs [here](https://esphome.io/components/climate/pid.html?highlight=pid#:~:text=The%20sensor%20should%20have%20a%20short%20update%20interval.):

> Note:
> The sensor should have a short update interval. The PID update frequency is tied to the update interval of the sensor. Set a short update_interval like 5s on the sensor.
> 
> We recommend putting a filter on the sensor (see filters in [Sensor Component](https://esphome.io/components/sensor/index.html)) and using output_averaging_samples to calm the PID sensor from a noisy input sensor.

So in esphome I created a new template temperature sensor that mirrors the original sensor:

```yaml
sensor:
  - platform: homeassistant
    id: living_room_temperature
    entity_id: sensor.living_room_temperature

  - platform: template
    name: Living Room PID Temp
    id: living_room_temperature_pid
    unit_of_measurement: "°C"
    device_class: "temperature"
    state_class: "measurement"
    lambda: !lambda return id(living_room_temperature).state;
    update_interval: 5s
    filters:
      - exponential_moving_average:
          alpha: 0.5
          send_every: 1
```

If the original sensor allows filtering that works even better, the home assistant api sensor doesn't allow the filter options.

This way no NaN's happen, the auto-tuning got better values, and the heating logic itself is much more responsive (it always waits for the first fresh temperature update before it starts heating if you turn it on).

closed 2022-12-31 by mgabor3141

remyderuysscher, 2022-12-31:
Still NaN's but since this is a different issue not related to autotune I'll open a new issue ok? I have a report from another user as well.  @mgabor3141

patrickcollins12, 2023-01-01:
To address your nan comment, I've added this INFO message when the autotuner is started:

_Autotune has started. This can take a long time depending on the responsiveness of your system. Your system output will be altered to deliberately oscillate above and below the setpoint multiple times. Until your sensor provides a reading, the autotuner may display 'nan'_

```
[11:38:11][D][switch:013]: 'PID Climate Autotune' Turning ON.
[11:38:11][I][pid.climate.console_fan_thermostat:158]: Autotune has started. This can take a long time depending on the responsiveness of your system. Your system output will be altered to deliberately oscillate above and below the setpoint multiple times. Until your sensor provides a reading, the autotuner may display 'nan'
[11:38:11][W][pid.climate.console_fan_thermostat:166]: !!! For PID autotuner you need to set AUTO (also called heat/cool) mode! Console Fan Thermostat
[11:38:11][D][pid.autotune.console_fan_thermostat:184]: PID Autotune:
[11:38:11][D][pid.autotune.console_fan_thermostat:185]:   Autotune is still running!
[11:38:11][D][pid.autotune.console_fan_thermostat:186]:   Status: Trying to reach nan °C
[11:38:11][D][pid.autotune.console_fan_thermostat:187]:   Stats so far:

```