esphome/issues #3863: Can read DS18b20

ymoona, 2022-12-01:
### The problem

ESP home fails to read temperature from a DS18B20

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 2022.11.4 

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

Dallas Temperature Sensor

### Example YAML snippet

```yaml
esphome:
  name: "esphome-test2"

esp8266:
  board: nodemcuv2

logger:
  level: DEBUG

dallas:
  - pin: D1
```

### Anything in the logs that might be useful for us?

```txt
Console output:
[I][logger:258]: Log initialized
[I][app:029]: Running through setup()...
[C][dallas.sensor:032]: Setting up DallasComponent...
[W][dallas.sensor:048]: Dallas device 0x7120320c3da10028 has invalid CRC.
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2022.11.4 compiled on Dec  1 2022, 20:39:36
[C][logger:293]: Logger:
[C][logger:294]:   Level: DEBUG
[C][logger:295]:   Log Baud Rate: 115200
[C][logger:296]:   Hardware UART: UART0
[C][dallas.sensor:075]: DallasComponent:
[C][dallas.sensor:076]:   Pin: GPIO5
[C][dallas.sensor:077]:   Update Interval: 60.0s
[W][dallas.sensor:080]:   Found no sensors!
```

### Additional information

I have tried various ports. 
There is an extra 100nf capacitor over vcc and gnd near the sensor. 
There is a 1k pullup to 3v3. But I also tried multiple variations, 1k 4k7 10k. results are the same. 

I started with 3 in parallel. But now down to a single one just to get it working. 

I have also swapped out the sensor for an other one, same result. 
I have tried different ports, same result.

When I unplug the sensor i get:
[E][dallas.sensor:112]: Requesting conversion failed
So it seems some form of communication is taking place. 

What else can I do to debug?

mmiller7, 2022-12-02:
I'm also experiencing this same issue with a D1 Mini board.  I've tried different resistors, capacitors, even a different sensor.  I've moved it between different pins on the board with no real success.

I think there may be some timing issue, it seems if I leave power connected and manually reset the board then it's able to detect them.

I'm wondering if when the power is first applied the ESPHome is trying to initialize the Dallas components before the hardware is sufficiently "booted", I've read some things saying the DS10B20 can take 750mS to "start up" but I can't find any way to delay bootup.

mmiller7, 2022-12-02:
Additional note, it seems like once you figure out the sensor address, you can hard-code it to that address and then the sensor works EVEN WHEN IT IS NOT FOUND at startup:

This to me seems to suggest some sort of timing issue where ESPHome needs to wait longer when power is first applied before attempting to configure the Dallas Hub component.

Example output from power-applied when using address instead of index - note it's "not found" at scan but then is clearly talking to the sensor anyway:
```
INFO Reading configuration /config/esphome/fireplace_propane.yaml...
WARNING 'fireplace_propane': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Starting log output from fireplace_propane.apt using esphome API
INFO Successfully connected to fireplace_propane.apt
[20:59:25][I][app:102]: ESPHome version 2022.11.4 compiled on Dec  1 2022, 20:58:35
[20:59:25][C][status_led:019]: Status LED:
[20:59:25][C][status_led:020]:   Pin: GPIO14
[20:59:25][C][wifi:504]: WiFi:
[20:59:25][C][wifi:362]:   Local MAC: 44:17:93:1E:E9:59
[20:59:25][C][wifi:363]:   SSID: [redacted]
[20:59:25][C][wifi:364]:   IP Address: 192.168.3.152
[20:59:25][C][wifi:365]:   BSSID: [redacted]
[20:59:25][C][wifi:367]:   Hostname: 'fireplace_propane'
[20:59:25][C][wifi:369]:   Signal strength: -65 dB ▂▄▆█
[20:59:25][C][wifi:373]:   Channel: 11
[20:59:25][C][wifi:374]:   Subnet: 255.255.255.0
[20:59:25][C][wifi:375]:   Gateway: 192.168.3.1
[20:59:25][C][wifi:376]:   DNS1: 192.168.3.1
[20:59:25][C][wifi:377]:   DNS2: 0.0.0.0
[20:59:25][C][logger:293]: Logger:
[20:59:25][C][logger:294]:   Level: DEBUG
[20:59:25][C][logger:295]:   Log Baud Rate: 0
[20:59:25][C][logger:296]:   Hardware UART: UART0
[20:59:25][C][switch.gpio:050]: GPIO Switch 'Sensor Power'
[20:59:25][C][switch.gpio:051]:   Pin: GPIO12
[20:59:25][C][switch.gpio:073]:   Restore Mode: Restore (Defaults to OFF)
[20:59:25][C][template.switch:058]: Template Switch 'fireplace_propane Deep Sleep Now'
[20:59:25][C][template.switch:058]:   Icon: 'mdi:sleep'
[20:59:25][C][template.switch:059]:   Restore State: NO
[20:59:25][C][template.switch:060]:   Optimistic: NO
[20:59:25][C][restart:022]: Restart Switch 'fireplace_propane Restart'
[20:59:25][C][restart:022]:   Icon: 'mdi:restart'
[20:59:25][C][dallas.sensor:075]: DallasComponent:
[20:59:25][C][dallas.sensor:076]:   Pin: GPIO13
[20:59:25][C][dallas.sensor:077]:   Update Interval: 15.0s
[20:59:25][W][dallas.sensor:080]:   Found no sensors!
[20:59:25][C][dallas.sensor:089]:   Device 'fireplace_propane Temperature'
[20:59:25][C][dallas.sensor:089]:     Device Class: 'temperature'
[20:59:25][C][dallas.sensor:089]:     State Class: 'measurement'
[20:59:25][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[20:59:25][C][dallas.sensor:089]:     Accuracy Decimals: 1
[20:59:25][C][dallas.sensor:097]:     Address: 0xfd3cece381100a28
[20:59:25][C][dallas.sensor:098]:     Resolution: 12
[20:59:25][C][adc:087]: ADC Sensor 'fireplace_propane Weight'
[20:59:25][C][adc:087]:   Device Class: 'weight'
[20:59:25][C][adc:087]:   State Class: 'measurement'
[20:59:25][C][adc:087]:   Unit of Measurement: 'lb'
[20:59:25][C][adc:087]:   Accuracy Decimals: 1
[20:59:25][C][adc:087]:   Icon: 'mdi:scale'
[20:59:25][C][adc:092]:   Pin: GPIO17
[20:59:25][C][adc:125]:   Update Interval: 15.0s
[20:59:25][C][captive_portal:088]: Captive Portal:
[20:59:26][C][mdns:103]: mDNS:
[20:59:26][C][mdns:104]:   Hostname: fireplace_propane
[20:59:26][C][ota:093]: Over-The-Air Updates:
[20:59:26][C][ota:094]:   Address: fireplace_propane.apt:8266
[20:59:26][C][api:138]: API Server:
[20:59:26][C][api:139]:   Address: fireplace_propane.apt:6053
[20:59:26][C][api:143]:   Using noise encryption: NO
[20:59:26][C][deep_sleep:049]: Setting up Deep Sleep...
[20:59:26][C][deep_sleep:052]:   Sleep Duration: 1800000 ms
[20:59:26][C][deep_sleep:055]:   Run Duration: 300000 ms
[20:59:31][D][sensor:126]: 'fireplace_propane Weight': Sending state -1.22598 lb with 1 decimals of accuracy
[20:59:33][D][dallas.sensor:143]: 'fireplace_propane Temperature': Got Temperature=24.1°C
[20:59:33][D][sensor:126]: 'fireplace_propane Temperature': Sending state 24.06250 °C with 1 decimals of accuracy

```

mmiller7, 2022-12-02:
Possible workaround - at least in my observation of this, a 500mS delay seems to make the DS18B20 properly detect reliably (at least the few times I've tried before calling it a night) directly on power-applied bootup the first time.

```
  on_boot:
      priority: 700
      then:
        - lambda: |-
            delay(500);
```

ymoona, 2022-12-07:
I have been testing various setups, both in hard and software. I ended up running some bare minimum code from Arduino IDE to only read the temperature from a single sensor without any checks. This seems to work, but none of them are addressable or able to read the device by id. 

Reading up on how many fake ds18b20 are out there, is guess I have bunch of faulty ones. So I ordered new ones, both from Ali and from a respectable webshop. Let's see how that turns out, but I guess this issue can be closed due to faulty sensors.

closed 2022-12-07 by ymoona