esphome/issues #3864: st7789v

nochlab1, 2022-12-02:
### The problem

Which esp32 works with 240x240 display all the ones i have tried all crash with E][display:022]: Could not allocate buffer for display!
Thanks

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.5

### What platform are you using?

ESP32

### Component causing the issue

st7789v

nochlab1, 2022-12-06:
> You're running out of memory on the ESP. Try enabling `eightbitcolor` as described [here](https://esphome.io/components/display/st7789v.html#configuration-variables) or use an ESP module that has PSRAM available.

i tried using eightbitcolor its still not working
i ordered ones that have psram but how do i tell esphome to use it

kbx81, 2022-12-06:
> i ordered ones that have psram but how do i tell esphome to use it

Simply add `param:` to your config. Some components will automatically add it "behind the scenes" so you may not even need to do this. Either way, give it a try. 🙂

nochlab1, 2022-12-06:
thanks adding psram: worked! i googled how to add psram to esphome and didnt see it

i tried eightbitcolor on a board and it didnt work so i assumed all my eso32 wroom wouldnt work but i see now it works by most and on a wrover board it works regular with psram
Thanks so much

closed 2022-12-06 by nochlab1