esphome/issues #3865: Captive portal does not update State for Select Control

pedjas, 2022-12-02:
### The problem

I've created simple Select Control that has few options. It works fine. Captive portal shows control and allows user to change option in control. Whole system reacts at change except captive portal. Status column for that Select Control does not change.

Also if something else changes status of Select Control, it's status is not updated in Captive Portal.

I have to manually refresh page to see change.

Here is an example of Select Control I use:

`
select:
  - platform: template
    name: Mode
    id: mode
    options: 
     - "OFF"
     - "ON"
     - "FORCED_ON"
    set_action: 
      - lambda: |-
          id(mode).state = x.c_str();

`

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

 Home Assistant 2022.11.4

### What platform are you using?

ESP8266

### Board

Sonoff Basic

### Component causing the issue

Select

### Example YAML snippet

```yaml
select:
  - platform: template
    name: Mode
    id: mode
    options: 
     - "OFF"
     - "ON"
     - "FORCED_ON"
    set_action: 
      - lambda: |-
          id(mode).state = x.c_str();
```

guix77, 2024-09-29:
It does not update the state by default, you could use the optimistic option: https://esphome.io/components/select/template.html

ssieb, 2024-09-29:
Right.  Either use optimistic or the `publish_state()` method.  You can't write to the `.state` directly.

closed 2024-09-29 by ssieb