esphome/issues #3868: sim800l call receive problem

andersonthomas, 2022-12-03:
### The problem

the cirtuit should trigger a relay when call is received and hangup the call. so when im trying to make a call to sim800 connected to esp32, sometimes it receives a call and triggers a relay but sometimes it shows just Call state is now: 4 and nothing happens. I found out, that, when i disconnect TX from sim800 it works fine, triggers a relay but just dont hangup call. When i connect TX backk, it start making problems.

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.3

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

sim800

### Example YAML snippet

```yaml
esphome:
  name: "brana-rfid-esp32"
  platform: ESP32
  board: nodemcu-32s

logger:
api:
ota:
  password: "c54ebb9b5bc3284e5d88a6cbb4a45ab5"

wifi:
  ssid: "escape_3"
  password: "xxxxxxxxxxx"

  ap:
    ssid: "Brana-Esp32 Fallback Hotspot"
    password: "2BksfCMLGdMl"

captive_portal:

switch:
  - platform: gpio
    pin: GPIO33
    name: "Svetlo_a"
    id: "relay2"
    restore_mode: ALWAYS_OFF
  - platform: template
    id: brana2
    name: "Brana2"
    icon: "mdi:gate"
    turn_on_action:
    - switch.turn_on: relay2
    - delay: 500ms
    - switch.turn_off: relay2

uart:
  baud_rate: 9600
  tx_pin: GPIO17
  rx_pin: GPIO16
#  rx_buffer_size: 512

#rdm6300:

sim800l:
  on_incoming_call:
    - logger.log:
        format: "Incoming call from '%s'"
        args: ["caller_id.c_str()"]
    - lambda: |-
        if ( caller_id == "+421xxxxxxxxx" ) {
        id(brana2).turn_on();
        }
    - lambda: |-
        if ( caller_id == "+421xxxxxxxxx" ) {
        id(brana2).turn_on();
        }

    - sim800l.disconnect

#sensor:
#  - platform: sim800l
#    rssi:
#      name: "ESP sim800 signal"

#binary_sensor:
#  - platform: sim800l
#    registered:
#      name: "ESP sim800 status"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/brana-rfid-esp32.yaml...
INFO Starting log output from brana-rfid-esp32.local using esphome API
INFO Successfully connected to brana-rfid-esp32.local
[14:26:00][I][app:102]: ESPHome version 2022.11.4 compiled on Dec  3 2022, 14:04:20
[14:26:00][C][wifi:504]: WiFi:
[14:26:00][C][wifi:362]:   Local MAC: 08:3A:F2:7B:DD:E0
[14:26:00][C][wifi:363]:   SSID: 'escape_3'[redacted]
[14:26:00][C][wifi:364]:   IP Address: 10.10.70.35
[14:26:00][C][wifi:366]:   BSSID: 0A:55:31:2E:39:6C[redacted]
[14:26:00][C][wifi:367]:   Hostname: 'brana-rfid-esp32'
[14:26:00][C][wifi:369]:   Signal strength: -65 dB ▂▄▆█
[14:26:00][C][wifi:373]:   Channel: 11
[14:26:00][C][wifi:374]:   Subnet: 255.255.255.0
[14:26:00][C][wifi:375]:   Gateway: 10.10.70.1
[14:26:00][C][wifi:376]:   DNS1: 1.1.1.1
[14:26:00][C][wifi:377]:   DNS2: 8.8.8.8
[14:26:00][C][logger:293]: Logger:
[14:26:00][C][logger:294]:   Level: DEBUG
[14:26:00][C][logger:295]:   Log Baud Rate: 115200
[14:26:00][C][logger:296]:   Hardware UART: UART0
[14:26:00][C][uart.arduino_esp32:108]: UART Bus 1:
[14:26:00][C][uart.arduino_esp32:109]:   TX Pin: GPIO17
[14:26:00][C][uart.arduino_esp32:110]:   RX Pin: GPIO16
[14:26:00][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[14:26:00][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[14:26:00][C][uart.arduino_esp32:115]:   Data Bits: 8
[14:26:00][C][uart.arduino_esp32:116]:   Parity: NONE
[14:26:00][C][uart.arduino_esp32:117]:   Stop bits: 1
[14:26:00][C][switch.gpio:050]: GPIO Switch 'Svetlo_a'
[14:26:00][C][switch.gpio:051]:   Pin: GPIO33
[14:26:00][C][switch.gpio:073]:   Restore Mode: Always OFF
[14:26:00][C][template.switch:058]: Template Switch 'Brana2'
[14:26:00][C][template.switch:058]:   Icon: 'mdi:gate'
[14:26:00][C][template.switch:059]:   Restore State: NO
[14:26:00][C][template.switch:060]:   Optimistic: NO
[14:26:00][C][sim800l:463]: SIM800L:
[14:26:00][C][captive_portal:088]: Captive Portal:
[14:26:00][C][mdns:103]: mDNS:
[14:26:00][C][mdns:104]:   Hostname: brana-rfid-esp32
[14:26:00][C][ota:093]: Over-The-Air Updates:
[14:26:00][C][ota:094]:   Address: brana-rfid-esp32.local:3232
[14:26:00][C][ota:097]:   Using Password.
[14:26:00][C][api:138]: API Server:
[14:26:00][C][api:139]:   Address: brana-rfid-esp32.local:6053
[14:26:00][C][api:143]:   Using noise encryption: NO
[14:26:31][I][sim800l:395]: Incoming call from +421xxxxxxxxx
[14:26:31][D][main:314]: Incoming call from '+421xxxxxxxxx'
[14:26:31][D][switch:013]: 'Brana2' Turning ON.
[14:26:31][D][switch:013]: 'Svetlo_a' Turning ON.
[14:26:31][D][switch:037]: 'Svetlo_a': Sending state ON
[14:26:32][D][switch:017]: 'Svetlo_a' Turning OFF.
[14:26:32][D][switch:037]: 'Svetlo_a': Sending state OFF
[14:26:33][D][sim800l:200]: Registered OK
[14:26:34][D][sim800l:234]: RSSI: 18
[14:26:34][I][sim800l:039]: Disconnecting...
[14:26:39][D][sim800l:234]: RSSI: 18
[14:26:44][D][sim800l:234]: RSSI: 18
[14:26:48][I][sim800l:395]: Incoming call from +421xxxxxxxxx
[14:26:48][D][main:314]: Incoming call from '+421xxxxxxxxx'
[14:26:48][D][switch:013]: 'Brana2' Turning ON.
[14:26:48][D][switch:013]: 'Svetlo_a' Turning ON.
[14:26:48][D][switch:037]: 'Svetlo_a': Sending state ON
[14:26:48][I][sim800l:039]: Disconnecting...
[14:26:49][D][switch:017]: 'Svetlo_a' Turning OFF.
[14:26:49][D][switch:037]: 'Svetlo_a': Sending state OFF
[14:26:54][D][sim800l:234]: RSSI: 16
[14:26:59][D][sim800l:234]: RSSI: 16
[14:26:59][D][sim800l:289]: Call state is now: 4
[14:27:05][W][sim800l:104]: Received unexpected OK. Ignoring
[14:27:05][W][sim800l:104]: Received unexpected OK. Ignoring
[14:27:09][D][sim800l:234]: RSSI: 16
[14:27:14][D][sim800l:234]: RSSI: 16
```

andriej, 2023-02-15:
I can confirm same behavior - it was working ok a while ago.
Right now, with the same config, it's:
```
[14:11:10][D][sim800l:291]: Call state is now: 4
[14:11:10][W][sim800l:104]: Received unexpected OK. Ignoring
[14:11:10][W][sim800l:104]: Received unexpected OK. Ignoring
[14:11:14][W][sim800l:104]: Received unexpected OK. Ignoring
```
and those are the only present in log regarding call incoming.
Just tested against latest esphome 2023.2

Could you take a look @glmnet?

allmazz, 2024-12-01:
any updates?
the bug is still reproducible on the latest esphome version

DJGLTD, 2024-12-05:
I've also running into the same issue, My issues are triple folded, 

Firstly I am running into the state 4, Which digging around in the code and comparing to the manual means the below, 

**incoming (MT Call)**

MT
Mobile Terminated messages are sent from a server or application to a mobile device. For example, when a brand sends a message to consumers' phones, it's an MT message.

Now what that means and why we are getting I have no idea. 

Secondly I am using a LilyGo-T-Call-SIM800 (https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800) My issue here is the start up process which caused me a fair few issues to get working, I need a way to automate pins on and off during the initial part of the script, If I try and run these post boot it doesn't work. But if I run them as part of an automation, Then change the code to exclude them magically it starts to work on that second soft boot. Issue here is I don't want to be manually uploading code twice to the device every time I reboot the thing. 

Third and Finally is a two part issue, Firstly calls once connected seem to auto disconnect, Now I don't have a microphone and speaker connected but I would expect the call to have an open line still, However if I got the open line the code has zero support for DTMF tone detection. which is the next hurdle I will run into when the call not being instantly ended is fixed.

DJGLTD, 2024-12-06:
It's been a long two hours of messing around but I have no resolved the booting of the sim800l issue and got that automated and working correctly. Still seeing the same issue around "Call state is now: 4"

DJGLTD, 2024-12-09:
If it's possible i have also raised ESP32 - Sim800L DTMF Tones #2974 as a feature request ?

glmnet, 2024-12-09:
can you try adding a delay before invoking the - disconnect ?

i.e. 
```

sim800l:
  on_incoming_call:
    - logger.log:
        format: "Incoming call from '%s'"
        args: ["caller_id.c_str()"]
    - lambda: |-
        if ( caller_id == "+421xxxxxxxxx" ) {
        id(brana2).turn_on();
        }
    - lambda: |-
        if ( caller_id == "+421xxxxxxxxx" ) {
        id(brana2).turn_on();
        }
# add delay before disconnect
    - delay: 2s
    - sim800l.disconnect
````