esphome/issues #3869: Heap Memory Allocation - Bug!

IgnacioHR, 2022-12-03:
### The problem

Hi, I started to develop in ESPHome in July 2022 and it took me almost 6 months to find this bug. I think this is more important that you may think at the beginning.

1. I started having problems with MAX7219 Digit Display because after a couple of minutes working, when the ESP32 react to interrupts (GPIO-Pins binary sensors) the display becomes unstable. I reported this bug in discord a couple of months ago, but, at that time, I'd not enough information to find the root of the problem.
2. A couple of weeks ago I started creating a driver for a VL53L4CX (distance sensor) And I found very strange behaviours. The device could not be initialised in the ESP32 but was working find using plain arduino code outside the ESPhome code.

So. At a certain moment I found this document: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html that describes how the memory is used by the 2 Cores of the ESP32. In fact, this explains the two mayor issues I'd in my developments and I've been able to move forward. Maybe, the solution I implemented is no the "best one" so I can't provide right now a solution but it is worth explaining it so people with more in deep knowledge of the ESP32 can find the proper way of fixing it.

Explanation of the issue:

I'll use the MAX7219digit.h class to explain it.
Line 118 `std::vector<std::vector<uint8_t>> [max_displaybuffer_](https://esphome.io/api/classesphome_1_1max7219digit_1_1_m_a_x7219_component.html#afaedfb216e28893653d78426dfe74d1a);` define a class variable that, according to the document about ESP32 memory allocation, is allocated in the HEAP of the TASK that created the class instance. IF For any reason (and this is the bug) the class is used by another TASK, the memory is no longer available and it is pure garbage for the other task.

How I solved it. The dirty way. I defined the variable out of the class as a static . Then, I modified the access to that variable to use the static instance. The static variables are not allocated inside any TASK so they are available across al the tasks. This fixed the problem for me, not only here, but also in the problem I'd with the driver for the VL53L4CX.

I don't like to define things as "static" because I cannot instantiate the same class multiple times. The solution just works for me, because I'm not using more than one instance. But there is a bug, somewhere, that is related with the compiler options, the task creations or the lack of macros used in the class variables (I don't know exactly how to approach the best solution) that is causing a lot of problems for us.

Please take a look at this issue because I believe this is going to change some things somewhere...

Regards
Ignacio

### Which version of ESPHome has the issue?

2022.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

all

### What platform are you using?

ESP32

### Board

ESP32

### Component causing the issue

Several, custom and, for sure, MAX7219digit

### Example YAML snippet

```yaml
display:
  - platform: max7219digit
    id: display0
    spi_id: spi0
    cs_pin: GPIO33
    num_chips: 1
    intensity: 8
    rotate_chip: 90
    scroll_enable: false
    update_interval: 469ms # cada tic, quitar un punto encendido
    lambda: |-
      static uint8_t heartBits[] = {1, 2, 11, 12, 5, 6, 15, 23, 31, 38, 45, 52, 51, 42, 33, 24, 16, 8};
      static uint8_t matrix[] = {0,0,0,0,0,0,0,0};
      static int maxRand = -1;
      static bool dispOnOff = false;
      if (maxRand == -1) {
        maxRand = 64;
        dispOnOff = false;
        for (int row = 0; row < 8; row++) {
          matrix[row] = 0x00;
        }
        ESP_LOGD("display", "Display OFF");
        it.turn_on_off(false);
        id(sensor_recien_cerrada).publish_state(false);
        return;
      }
      if (id(sensor_puerta).state) { /* se ha abierto la puerta */
        maxRand = 64;
        matrix[7] = (uint8_t)0x18;
        matrix[6] = (uint8_t)0x7c;
        matrix[5] = (uint8_t)0x5e;
        matrix[4] = (uint8_t)0x5e;
        matrix[3] = (uint8_t)0x5e;
        matrix[2] = (uint8_t)0x5e;
        matrix[1] = (uint8_t)0x4e;
        matrix[0] = (uint8_t)0x7e;
        if (!dispOnOff) {
          dispOnOff = true;
          ESP_LOGD("display", "Display ON");
          it.turn_on_off(true);
        }
      } else if (dispOnOff) { /* comienza o continúa la cuenta atrás */
        if (maxRand == 64) {
          ESP_LOGD("display", "Matrix initialized all ON");
          it.turn_on_off(true);
          for (int row = 0; row < 8; row++) {
            matrix[row] = (uint8_t)0xFF;
          }
        } else {
          if (maxRand > 17) {
            uint8_t rowB;
            int randCol;
            int randRow;
            bool searchAgain;
            do {
              searchAgain = false;
              randRow = random_uint32() % 8;
              rowB = matrix[randRow];
              randCol = random_uint32() % 8;
              if ((rowB >> randCol) & 0x1) {
                int randBit = randRow * 8 + randCol;
                for (int idx = 0; !searchAgain && idx < 18; ++idx) {
                  if (randBit == heartBits[idx])
                    searchAgain = true;
                }
              } else {
                searchAgain = true;
              }
            } while (searchAgain);
            rowB &= ~(1UL << randCol);
            matrix[randRow] = rowB;
          } else {
            uint8_t nextLed = heartBits[17 - maxRand];
            int row = nextLed / 8;
            uint8_t rowB = matrix[row];
            int col = nextLed % 8;
            rowB &= ~(1UL << col);
            matrix[row] = rowB;
          }
        }
        maxRand -= 1;
      }
      if (dispOnOff) {
        for (int row = 0; row < 8; row++) {
          uint8_t rowB = matrix[row];

          for (int col = 0; col < 8; col++) {
            if ((rowB >> col) & 0x01) {
              it.draw_pixel_at(row, col, COLOR_ON);
            } else {
              it.draw_pixel_at(row, col, COLOR_OFF);
            }
          }
        }
      }
```

### Anything in the logs that might be useful for us?

```txt
There are no specific "error messages", the problem with the MAX7219digit is the display not operating consistently and representing lines as they was the negative of what was indicated the display to do. In the case of the distance sensor VL53L4CX, the initialisation done in the setup() function worked, but using it in the loop() method didn't work because the internal structures was created in a different TASK.
```

### Additional information

Pointers to the relevant documentation:

ST Community thread where I explained my problem and solution with the distance sensor:
https://community.st.com/s/question/0D53W00001uMUKcSAO/unable-to-get-measures-from-vl53l4cx

How memory allocation works in the ESP32. Note, the memory is not allocated as you may think!
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html

ssieb, 2022-12-17:
Where do you see anything about task heaps in that document?  There is nothing like that.  There's one heap and allocated chunks can be used by any task.

Also, binary sensors don't use interrupts.

IgnacioHR, 2022-12-18:
Thank you @ssieb for jumping into this. Unfortunately (for me) the root problem with this device has not been fully fixed, but it is not related with the memory allocation. I’m going to close this bug now.

closed 2022-12-18 by IgnacioHR