esphome/issues #3870: delay() in lambda should issue an error

pedjas, 2022-12-03:
### The problem

delay() in lambda does not work. It does nothing, just silently skips further.

Novice users do not know that and are loosing their minds expecting it to work.

Compiler should report error like "delay() does not work in this environment." And stop compiling.

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.5

### What platform are you using?

ESP8266

### Board

Sonoff Basic

### Example YAML snippet

```yaml
I needed LED to blink to show progress while executing code in lambda. I tried this:

id (red_led).turn_on().perform();
delay(300);
id (red_led).turn_off().perform();

LED did not blink as expected. After lots of time lost trying to find out what I did wrong I finally found some explanation to other guy having the same issue and that that said delay() does not work. It just skips. That is why you cannot make LED blink this way. It turns it on and then instantly turns it off without any delay.
```

ssieb, 2023-01-26:
The compiler doesn't know anything about that.  You can put any C++ code in a lambda.  You could put a `while (true);` there as well and crash it.

pedjas, 2023-01-29:
> The compiler doesn't know anything about that. You can put any C++ code in a lambda. You could put a `while (true);` there as well and crash it.

Well, that is not same. `while (true);` is obvious. delay() silently not doing what is supposed to do is a problem. Who knoq, what else misbehaves like that?

ssieb, 2023-01-31:
`delay()` is doing exactly what it's supposed to do.  But what it does is not compatible with esphome.