esphome/issues #3873: BedJet fan speed is off by 5%

kquinsland, 2022-12-04:
### The problem

This comes from discussion in #3807 .

Essentially, there seems to be an "off by one" error with the speed range that the bedjet component reports to HA and the value that HA sends back to the bedjet component when the user selects 100% fan.

Using the [`mushroom/fan` card](https://github.com/piitaya/lovelace-mushroom/blob/main/docs/cards/fan.md), I moved the slider all the way to 100% and then saw this in the ESPHome logs:

```
[10:20:37][W][bedjet:056]: Invalid fan speed index 20, expecting 0-19.
```

Additionally, the speed reported to the user via the HA ui is 5% off from what the bedjet remote shows. The remote is always 5% ahead.

----

Here's the autoconfig payload that's sent to HA:

Topic: `homeassistant/fan/bj_controller-8<...>0/bedjet_fan/config`

Payload:

```json`
{
    "pct_cmd_t": "bj_controller/fan/bedjet_fan/speed_level/command",
    "pct_stat_t": "bj_controller/fan/bedjet_fan/speed_level/state",
    "spd_rng_max": 20,
    "name": "BedJet Fan",
    "stat_t": "bj_controller/fan/bedjet_fan/state",
    "cmd_t": "bj_controller/fan/bedjet_fan/command",
    "avty_t": "bj_controller/status",
    "uniq_id": "ESPfanbedjet_fan",
    "dev": {
        "ids": "3<...>0",
        "name": "bj_controller-8<...>0",
        "sw": "esphome v2022.11.4 Dec  4 2022, 10:18:41",
        "mdl": "mhetesp32minikit",
        "mf": "espressif"
    }
}
````

Looking [at the docs](https://www.home-assistant.io/integrations/fan.mqtt/#speed_range_min) for `speed_range_min` attribute: 

```
The minimum of numeric output range (off not included, so speed_range_min - 1 represents 0 %). The number of speeds within the speed_range / 100 will determine the percentage_step.
```

Docs say that the default value is `1` which seems to make sense. If you sent a value of `0`, HA might think that sending a speed value of `-1` was how you set the fan to the lowest possible speed ... which seems wrong :/.

> The fan index sounds like a bug, and your explanation makes sense. Feel free to open a ticket or PR for that and I'll take a look. Should be simple enough. I think the root of the issue is that HA wants to allow setting 0%, but that should actually be "off". I don't think we have a way to tell HA that we want to start at min=5% ("fan speed index 0 == 5%"), so we probably need to update the fan component to have a range of 0-20 and subtract one when sending the speed to BedJet, with a special case for 0==off.

Based on what I'm seeing in the HA docs, I think you are right. A small tweak to the mqtt payload so the value for `payload_off` is set to `0` instead of the default `OFF` should be sufficient.

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32-IDF

### Component causing the issue

bedjet

nochlab1, 2023-01-11:
Was this ever fixed?

nochlab1, 2023-01-11:
ok thanks

jhansche, 2023-01-14:
@kquinsland @nochlab1 I just pushed a fix to https://github.com/jhansche/esphome/tree/pr/fix-3873. You can test it with external components:
```yaml
external_components:
  - source: github://pr#4292
    components: [ bedjet ]
```

closed 2023-01-16 by jesserockz