esphome/issues #3875: Wemos NodeMCU ESP8266 not working after defining components

leodrivera, 2022-12-05:
### The problem

I have a Wemos NodeMCU ESP8266 v3 - CH340 (https://botland.store/img/art/inne/10908_3.jpg) and I noticed that after installing esphome on it, the device stopped responding (it didn't even ping response).

The flashing process proceeds normally without any errors.

I performed the troubleshooting, seeking to isolate the problem. In this process, I did the following:

- Tried the proposed solution in these two issues(#3831 and #3583)
- Tested on a second board of the same model
- Removed Home Assistant API encryption

None of the previous attempts worked. Finally, as a last attempt, I compiled the code without any components, as shown below, and performed the flashing process:

```yaml
esphome:
  name: wemos-test

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret wifi_fallback_password

captive_portal:
```

Worked perfectly.

To confirm my suspicion, I added a simple remote_receiver component and flashed it again, as shown below, and the problem happened again.

```yaml
esphome:
  name: wemos-test

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret wifi_fallback_password

captive_portal:

remote_receiver:
  pin:
    number: D6
    inverted: true
    mode:
      input: true
      pullup: true
  dump: all
```

PS: I have other esps running in my HA (ESP-12f and ESP-01) and never had problems like that.
PS2: I've changed the board type to generic esp8266 (`esp01_1m`) and the number pin to the corresponding GPIO (`GPIO12`) and the result was the same.

### Which version of ESPHome has the issue?

2022.11.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2022.11.5

### What platform are you using?

ESP8266

### Board

Wemos NodeMCU ESP8266 v3 - CH340

### Component causing the issue

any component

schinckel, 2022-12-07:
Are you using an i2c bus? I needed to increase the bus speed after connecting a display using i2c.

Loatan, 2022-12-07:
Similar problem. I'm struggling with this problem to no avail.
All advice from similar topics has been done, there is no significant difference.
At first, removing the "ota", "ap", "logging" sections when adding components helped, but with a certain number of added component parameters, this no longer works.
NODEMCU v3 and WEMOS d1 mini
```
wifi_apply_hostname_(lothost-ir): lwIP error -16 on interface st (index 0)
Event: Disconnected ssid='LOTHOST2' bssid=3C:7C:3F:AB:38:78 reason='Association Leave'
```
`ap_probe_send over, rest wifi status to disassoc`

leodrivera, 2022-12-07:
> Are you using an i2c bus? I needed to increase the bus speed after connecting a display using i2c.

I've tested it with different components, including a component that made use of the i2c bus. In all cases where I used a component, I had the problem I described above.

leodrivera, 2022-12-08:
> Similar problem. I'm struggling with this problem to no avail. All advice from similar topics has been done, there is no significant difference. At first, removing the "ota", "ap", "logging" [...]

I've got this same error when trying one of the possible solutions (to use `board: d1_mini` ):
```log
[W][wifi_esp8266:482]: Event: Disconnected ssid='Leo Wi-Fi' bssid=[redacted] reason='Beacon Timeout'
[W][wifi:125]: WiFi Connection lost... Reconnecting...
[W][wifi:585]: Restarting WiFi adapter...
[I][wifi:257]: WiFi Connecting to 'Leo Wi-Fi'...
[W][wifi_esp8266:212]: wifi_apply_hostname_(wemos-test): lwIP error -16 on interface st (index 0)
[D][text_sensor:067]: 'Wemos Test IP': Sending state '0.0.0.0'
[I][wifi:518]: WiFi Connected!
[C][wifi:362]:   Local MAC: [redacted]
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.1.203
[C][wifi:365]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'wemos-test'
[C][wifi:369]:   Signal strength: -38 dB ▂▄▆█
[C][wifi:373]:   Channel: 6
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.1.1
[C][wifi:376]:   DNS1: 192.168.1.1
[C][wifi:377]:   DNS2: 192.168.1.1
[D][text_sensor:067]: 'Wemos Test IP': Sending state '192.168.1.203'
[W][wifi_esp8266:482]: Event: Disconnected ssid='Leo Wi-Fi' bssid=[redacted] reason='Beacon Timeout'
[W][wifi:125]: WiFi Connection lost... Reconnecting...
[W][wifi:585]: Restarting WiFi adapter...
[I][wifi:257]: WiFi Connecting to 'Leo Wi-Fi'...
[W][wifi_esp8266:212]: wifi_apply_hostname_(wemos-test): lwIP error -16 on interface st (index 0)
[D][text_sensor:067]: 'Wemos Test IP': Sending state '0.0.0.0'
[I][wifi:518]: WiFi Connected!
[C][wifi:362]:   Local MAC: [redacted]
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.1.203
[C][wifi:365]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'wemos-test'
[C][wifi:369]:   Signal strength: -33 dB ▂▄▆█
[C][wifi:373]:   Channel: 6
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.1.1
[C][wifi:376]:   DNS1: 192.168.1.1
[C][wifi:377]:   DNS2: 192.168.1.1
[D][text_sensor:067]: 'Wemos Test IP': Sending state '192.168.1.203'
[W][wifi_esp8266:482]: Event: Disconnected ssid='Leo Wi-Fi' bssid=[redacted] reason='Beacon Timeout'
[W][wifi:125]: WiFi Connection lost... Reconnecting...
[W][wifi:585]: Restarting WiFi adapter...
[I][wifi:257]: WiFi Connecting to 'Leo Wi-Fi'...
[W][wifi_esp8266:212]: wifi_apply_hostname_(wemos-test): lwIP error -16 on interface st (index 0)
[D][text_sensor:067]: 'Wemos Test IP': Sending state '0.0.0.0'
[I][wifi:518]: WiFi Connected!
[C][wifi:362]:   Local MAC: [redacted]
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.1.203
[C][wifi:365]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'wemos-test'
[C][wifi:369]:   Signal strength: -34 dB ▂▄▆█
[C][wifi:373]:   Channel: 6
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.1.1
[C][wifi:376]:   DNS1: 192.168.1.1
[C][wifi:377]:   DNS2: 192.168.1.1
[D][text_sensor:067]: 'Wemos Test IP': Sending state '192.168.1.203'
[D][sensor:126]: 'Wemos Test Wi-Fi Signal': Sending state -34.00000 dBm with 0 decimals of accuracy
[W][wifi_esp8266:482]: Event: Disconnected ssid='Leo Wi-Fi' bssid=[redacted] reason='Beacon Timeout'
[W][wifi:125]: WiFi Connection lost... Reconnecting...
[W][wifi:585]: Restarting WiFi adapter...
[I][wifi:257]: WiFi Connecting to 'Leo Wi-Fi'...
[W][wifi_esp8266:212]: wifi_apply_hostname_(wemos-test): lwIP error -16 on interface st (index 0)
[D][text_sensor:067]: 'Wemos Test IP': Sending state '0.0.0.0'
[I][wifi:518]: WiFi Connected!
[C][wifi:362]:   Local MAC: [redacted]
[C][wifi:363]:   SSID: [redacted]
[C][wifi:364]:   IP Address: 192.168.1.203
[C][wifi:365]:   BSSID: [redacted]
[C][wifi:367]:   Hostname: 'wemos-test'
[C][wifi:369]:   Signal strength: -35 dB ▂▄▆█
[C][wifi:373]:   Channel: 6
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: 192.168.1.1
[C][wifi:376]:   DNS1: 192.168.1.1
[C][wifi:377]:   DNS2: 192.168.1.1
[D][text_sensor:067]: 'Wemos Test IP': Sending state '192.168.1.203'
[D][sensor:126]: 'Wemos Test Uptime': Sending state 55.83400 s with 0 decimals of accuracy
[D][sensor:126]: 'Wemos Test Wi-Fi Signal': Sending state -35.00000 dBm with 0 decimals of accuracy
[D][sensor:126]: 'Wemos Test Uptime': Sending state 115.83400 s with 0 decimals of accuracy
```

38decibel, 2023-01-27:
Hi,

I have similar problem with an AzDelivery D1 mini based on esp8266ex ; reason='Probe Request Unsuccessful'.
I solved this with an aluminum foil around the card, avoiding the antenna.

[image]

FGOD1983, 2024-03-08:
apart from the joke, is there any update on a solution? because this is really frustrating

ssieb, 2024-03-08:
That's not a joke.  I expect that at least in that case, there's interference between the antenna and the rest of the board.  You could try reducing the tx power.  This has been a known issue with some 8266 boards.

FGOD1983, 2024-03-11:
@ssieb the most strange thing about this is that I ordered 3 boards recently (my mistake that I selected these EX ones), and out of 3 1 has no issues at all and 2 have. All with the same config being loaded onto them. 

So I just tossed those 2 in the bin and kept the one working now, but it kinda starts to feel more like a bad wifi chip on those board that sometimes is of better quality, then it actually being a design flaw like it is suggested

mevst, 2024-05-12:
I had this problem on my 2 Sonoff basic R1 devices bought in 2016. I didn't use them until now.
I tried all the advices from the issues listed in the original problem description. None of them actually solve the issue while some of them have a temporary impact, like removing some particular component from the code helps but adding some other brings the issie back. However the same code works OK on other ESP8266 devices. Code size is not the root of the issue
After many experiments with FW and HW I figured out that this is a HW issue related to the crystal on the board. 
The crystal on my Sonoff Basic R1 units is very tiny, has no clear marking and as I suspect is faulty.
I replaced it with little bit bigger crystal desoldered from perfectly working ESP01s module and now the previously faulty Sonoff Basic R1 works perfectly.
I think that these small chrystal starts slowly at power up and at moment when boot code measures the CLK frequency and configures PLL the chrystal oscillation is not stable yet. Changes in code probably cause change in the boot timing.
I see Sonoff uses similar chrystal in other devices (I see it on my Sonoff 4CH R3) and they works without problems. So the problim looks to be specific for a particular manyfacturing batches.
The chrystal replacement isn't easy, I used hot air rework station and soltering paste to do it, so it isn't a solution for everyone. 
However I hope it explains the root reason of the issue.

milenBG, 2025-02-02:
Hello,
I had the following issue:

`[15:24:20][C][safe_mode:079]: There have been 5 suspected unsuccessful boot attempts
[15:24:20][I][app:029]: Running through setup()...
[15:24:20][C][gpio.one_wire:011]: Setting up 1-wire bus...
[15:24:20][C][dallas.temp.sensor:081]: setting up Dallas temperature sensor...
[15:24:20][C][wifi:048]: Setting up WiFi...
[15:24:20][C][wifi:061]: Starting WiFi...
[15:24:20][C][wifi:062]:   Local MAC: YY:YY:YY:YY:YY:YY
[15:24:20][D][wifi:482]: Starting scan...
[15:24:20][W][component:157]: Component wifi set Warning flag: scanning for networks
[15:24:20][D][dallas.temp.sensor:054]: 'Bedroom temperature': Got Temperature=18.8°C
[15:24:20][D][sensor:093]: 'Bedroom temperature': Sending state 18.81250 °C with 1 decimals of accuracy
[15:24:26][D][wifi:497]: Found networks:
[15:24:26][D][wifi:545]: - 'somenetwork' (ZZ:ZZ:ZZ:ZZ:ZZ:ZZ) ▂▄▆█
[15:24:26][D][wifi:545]: - 'othernetwork' (XY:ZX:YZ:XY:ZX:YZ) ▂▄▆█
[15:24:26][W][wifi:551]: No matching network found!
[15:24:31][D][wifi:482]: Starting scan...
[15:24:36][D][wifi:497]: Found networks:
[15:24:36][I][wifi:540]: - 'ala-bala' (XX:XX:XX:XX:XX:XX) ▂▄▆█
[15:24:36][D][wifi:542]:     Channel: 11
[15:24:36][D][wifi:543]:     RSSI: -81 dB
[15:24:36][D][wifi:545]: - 'othernetwork' (XY:ZX:YZ:XY:ZX:YZ) ▂▄▆█
[15:24:36][I][wifi:313]: WiFi Connecting to 'ala-bala'...
[15:24:42][W][wifi_esp8266:510]: Event: Disconnected ssid='ala-bala' reason='Probe Request Unsuccessful'
[15:24:42][W][wifi:653]: Error while connecting to network.
[15:24:42][D][wifi:697]: Retrying with hidden networks...
[15:24:42][I][wifi:313]: WiFi Connecting to 'ala-bala'...
[15:24:48][W][wifi_esp8266:510]: Event: Disconnected ssid='ala-bala' reason='Probe Request Unsuccessful'
[15:24:48][W][wifi:653]: Error while connecting to network.
[15:24:48][D][wifi:697]: Retrying with hidden networks...
[15:24:53][I][wifi:313]: WiFi Connecting to 'ala-bala'...
[15:24:59][W][wifi_esp8266:510]: Event: Disconnected ssid='ala-bala' reason='Probe Request Unsuccessful'
[15:24:59][W][wifi:653]: Error while connecting to network.
[15:24:59][D][wifi:697]: Retrying with hidden networks...
[15:24:59][I][wifi:313]: WiFi Connecting to 'ala-bala'...
[15:25:05][W][wifi_esp8266:510]: Event: Disconnected ssid='ala-bala' reason='Probe Request Unsuccessful'
[15:25:05][W][wifi:653]: Error while connecting to network.
[15:25:05][D][wifi:697]: Retrying with hidden networks...
[15:25:10][D][dallas.temp.sensor:054]: 'Bedroom temperature': Got Temperature=18.9°C
[15:25:10][D][sensor:093]: 'Bedroom temperature': Sending state 18.87500 °C with 1 decimals of accuracy
[15:25:10][I][wifi:313]: WiFi Connecting to 'ala-bala'...
[15:25:16][W][wifi_esp8266:510]: Event: Disconnected ssid='ala-bala' reason='Probe Request Unsuccessful'
[15:25:16][W][wifi:653]: Error while connecting to network.
[15:25:16][D][wifi:697]: Retrying with hidden networks...
[15:25:16][I][wifi:313]: WiFi Connecting to 'ala-bala'...
[15:25:22][W][wifi_esp8266:510]: Event: Disconnected ssid='ala-bala' reason='Probe Request Unsuccessful'`

Removing "logger" module solved the problem. I hope this help someone.

ssieb, 2025-02-02:
It's not the logger that was the issue as has been mentioned many times.  It's just a random change.