esphome/issues #3876: MCP9600 only works on initial install

kyleteal, 2022-12-06:
### The problem

MCP9600 sensor works when installed, however upon reboot the log shows the error "Connected device does not match a known MCP9600 or MCP901 sensor". If I do not change anything and simply reinstall esphome on the device it works again until the device loses power and reboots. 

### Which version of ESPHome has the issue?

2022.11.4 / 2022.11.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2022.11.2

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

MCP9600

### Example YAML snippet

```yaml
substitutions:
  node_name: pad-secondary-temp
  device_verbose_name: "Pad Secondary Temp"
  platform: ESP8266
  board: d1_mini_lite

esphome:
  name: ${node_name}
  build_path: ./build/${node_name}
  platform: ${platform}
  board: ${board}

wifi:
  networks:
  - ssid: "IoT"
    password: !secret iot_wifi_pass
  ap:
    ssid: ${device_verbose_name} Hotspot
    password: !secret ap_pass
  fast_connect: true

captive_portal:

logger:

api:
  password: !secret api_pass
  encryption:
    key: !secret encrypt_key

ota:
  password: !secret ota_pass

i2c:
  sda: D2
  scl: D1
  scan: true
  frequency: 150kHz

dallas:
  - pin: D7
    update_interval: 10s

sensor:
   - platform: dallas
     address: 0x0801193cfef5ed28
     name: ${device_verbose_name}
     accuracy_decimals: 0
   - platform: mcp9600
     hot_junction:
       name: "Oven Temp"
     cold_junction:
       name: "Ambient Temperature"
     update_interval: 10s

binary_sensor:
  - platform: status
    name: ${device_verbose_name} Sensors Status
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/pad-secondary-temp.yaml...
INFO Starting log output from pad-secondary-temp.local using esphome API
INFO Successfully connected to pad-secondary-temp.local
[21:24:40][I][app:102]: ESPHome version 2022.11.5 compiled on Dec  5 2022, 21:23:05
[21:24:40][C][wifi:504]: WiFi:
[21:24:40][C][wifi:362]:   Local MAC: 
[21:24:40][C][wifi:363]:   SSID: 'IoT'[redacted]
[21:24:40][C][wifi:364]:   IP Address:
[21:24:40][C][wifi:365]:   BSSID: 
[21:24:40][C][wifi:367]:   Hostname: 'pad-secondary-temp'
[21:24:40][C][wifi:369]:   Signal strength: -47 dB ▂▄▆█
[21:24:40][C][wifi:373]:   Channel: 1
[21:24:40][C][wifi:374]:   Subnet: 255.255.255.0
[21:24:40][C][wifi:375]:   Gateway:
[21:24:40][C][wifi:376]:   DNS1: 1.1.1.1
[21:24:40][C][wifi:377]:   DNS2: 8.8.8.8
[21:24:40][C][logger:293]: Logger:
[21:24:40][C][logger:294]:   Level: DEBUG
[21:24:40][C][logger:295]:   Log Baud Rate: 115200
[21:24:40][C][logger:296]:   Hardware UART: UART0
[21:24:40][C][i2c.arduino:052]: I2C Bus:
[21:24:40][C][i2c.arduino:053]:   SDA Pin: GPIO4
[21:24:40][C][i2c.arduino:054]:   SCL Pin: GPIO5
[21:24:40][C][i2c.arduino:055]:   Frequency: 150000 Hz
[21:24:40][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[21:24:40][I][i2c.arduino:068]: Results from i2c bus scan:
[21:24:40][I][i2c.arduino:074]: Found i2c device at address 0x67
[21:24:40][C][dallas.sensor:075]: DallasComponent:
[21:24:40][C][dallas.sensor:076]:   Pin: GPIO13
[21:24:40][C][dallas.sensor:077]:   Update Interval: 10.0s
[21:24:40][D][dallas.sensor:082]:   Found sensors:
[21:24:40][D][dallas.sensor:084]:     0x0801193cfef5ed28
[21:24:40][C][dallas.sensor:089]:   Device 'Pad Secondary Temp'
[21:24:40][C][dallas.sensor:089]:     Device Class: 'temperature'
[21:24:40][C][dallas.sensor:089]:     State Class: 'measurement'
[21:24:40][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[21:24:40][C][dallas.sensor:089]:     Accuracy Decimals: 0
[21:24:41][C][dallas.sensor:097]:     Address: 0x0801193cfef5ed28
[21:24:41][C][dallas.sensor:098]:     Resolution: 12
[21:24:41][C][mcp9600:068]: MCP9600:
[21:24:41][C][mcp9600:069]:   Address: 0x67
[21:24:41][C][mcp9600:070]:   Update Interval: 10.0s
[21:24:41][C][mcp9600:072]:   Device ID: 0x0
[21:24:41][C][mcp9600:074]:   Hot Junction Temperature 'Oven Temp'
[21:24:41][C][mcp9600:074]:     Device Class: 'temperature'
[21:24:41][C][mcp9600:074]:     State Class: 'measurement'
[21:24:41][C][mcp9600:074]:     Unit of Measurement: '°C'
[21:24:41][C][mcp9600:074]:     Accuracy Decimals: 1
[21:24:41][C][mcp9600:075]:   Cold Junction Temperature 'Ambient Temperature'
[21:24:41][C][mcp9600:075]:     Device Class: 'temperature'
[21:24:41][C][mcp9600:075]:     State Class: 'measurement'
[21:24:41][C][mcp9600:075]:     Unit of Measurement: '°C'
[21:24:41][C][mcp9600:075]:     Accuracy Decimals: 1

[21:24:41][E][mcp9600:079]: Connected device does not match a known MCP9600 or MCP901 sensor

[21:24:41][C][status:034]: Status Binary Sensor 'Pad Secondary Temp Sensors Status'
[21:24:41][C][status:034]:   Device Class: 'connectivity'
[21:24:41][C][captive_portal:088]: Captive Portal:
[21:24:41][C][mdns:103]: mDNS:
[21:24:41][C][mdns:104]:   Hostname: pad-secondary-temp
[21:24:41][C][ota:093]: Over-The-Air Updates:
[21:24:41][C][ota:094]:   Address: pad-secondary-temp.local:8266
[21:24:41][C][ota:097]:   Using Password.
[21:24:41][C][api:138]: API Server:
[21:24:41][C][api:139]:   Address: pad-secondary-temp.local:6053
[21:24:41][C][api:141]:   Using noise encryption: YES
[21:24:44][D][dallas.sensor:143]: 'Pad Secondary Temp': Got Temperature=18.9°C
[21:24:44][D][sensor:126]: 'Pad Secondary Temp': Sending state 18.87500 °C with 0 decimals of accuracy
```

### Additional information

I have tried changing the i2c frequency to 100, 150, 200, 500 and experience the same issue every time.

kyleteal, 2022-12-07:
Additional insight, the sensor came back after leaving it running for a day. As soon as the esp reboots however it is unavailable again. 

[image]

zdeden15, 2022-12-07:
I'm having the same exact issue as well except I'm using an ESP32.

jowgn, 2022-12-26:
I don’t get it to work at all

oustagnagna, 2022-12-27:
Same problem with ESPHome v2022.12.3 and  platform: ESP8266 board: nodemcuv2 
on 
Home Assistant 2022.12.8
Supervisor 2022.11.2
Operating System 9.4

jowgn, 2023-01-08:
Isn't the maximum frequency 100kHz and the minimum 10kHz (https://cdn-learn.adafruit.com/assets/assets/000/105/040/original/MCP960X-Data-Sheet-20005426%281%29.pdf?1633364221 page 9), so shouldn't the correct check be a minimum frequency of 10kHz? Or am I wrong?

jowgn, 2023-01-09:
Solution for me is to turn down the i2c frequency to 10kHz.

closed 2023-01-10 by jesserockz

chrisckc, 2023-01-13:
I am seeing the same issue on an ESP32 DevKit V1, the MCP9600 should work fine at 100KHz, and indeed it does if a use the Adafruit MCP9600 library with the example it includes. The library does not change the default I2C frequency which I believe is 100KHz for ESP32.

If I then upload ESPHome to the same setup, I see the error described here. (using 100KHz in the i2c configuration setting as suggested in the docs). I can see there has been a commit to set the minimum to 10KHz, using this speed should be be required as I have found it works at 100KHz using the Adafruit MCP9600 library.

I tried using 4.7k pull-ups on both SDA and SDL instead of relying on internal ones, no difference.
I am actually using the Seeed Studio MCP9600 but chose the Adafruit library because the Seeed library does not compile.

Something interesting is that if I run the usual I2C scanner code such that it does a bus scan every 5 seconds, I keep seeing nothing detected on every other scan.

Found 1 device(s)
Found 0 device(s)
Found, 1 device(s)
.........
etc.